robotology/mesh-support #167: Problem of firware and Board

gregoire-pointeau, 2016-01-21:
Hello.
I have a problem when upgrading the whole system here in Lyon.

The iCub is on the iKart.
We (me and @Tobias-Fischer ) updated the system (yarp, firmware, and icub-main) but when we launch `robotInterface`, we have some errors, and the robot doesn't launch.

The errors seem to be:
```
Left hand: MAIS board is not detected
Right hand: One of the motor boards is not detected -> robotInterface is crashing
CAN Net 5 for the left hand and 6 for the  right hand
```

We don't know from where these errors can come?

Also @randaz81 , our robot config files are apparently not up to date.

Thanks

Tobias-Fischer, 2016-01-21:
Hi @gregoire-pointeau,
can you please attach the log when you start `robotInterface`? That will make the life for @super-ste and @julijenv easier for debugging.

I think one of the boards in the hand must have a loose connection.

Best,
Tobi

gregoire-pointeau, 2016-01-21:
Hi, here is the log:
[log21-01-16_with_arms.txt](https://github.com/robotology/icub-support/files/99183/log21-01-16_with_arms.txt)

gregoire-pointeau, 2016-01-21:
And this is without arms. Note that without arm, the head seems to start normally, but the torso doesn't move (and in the end, it still fails).

[log21-01-16-noarms.txt](https://github.com/robotology/icub-support/files/99185/log21-01-16-noarms.txt)

Tobias-Fischer, 2016-01-21:
This log is different from the log I had seen earlier today. Where is the error about the firmware in that log? And in the log I had seen there were segfaults.

gregoire-pointeau, 2016-01-21:
I don't know why, but the log seem a bit strange, the segfault was present in both case, but not in the log (I did: `robotInterface -> log.txt`)

Tobias-Fischer, 2016-01-21:
Can you run `robotInterface &> log.txt` instead?

gregoire-pointeau, 2016-01-21:
Ok, I updated the 2 files.

Tobias-Fischer, 2016-01-25:
Any news @gregoire-pointeau @robotology/icub-support-team @randaz81?

randaz81, 2016-01-25:
For the log with the arms: you should check from the canloader if the MAIS board is correctly working or if it is broken. Check the power supply voltage given by the iKart. Check the led colors on the MAIS board. Update the MAIS firmware.
For the log without the arms: nothing can be said about this, since the torso was already in hardware fault status before launching robotInterface. Please re-post a clean log starting robotInterface 10 seconds after turning on the motors. 
Contact icub-support on skype.

gregoire-pointeau, 2016-01-25:
Hi @randaz81 , for some reason, now the torso is working fine (maybe the icub was tired at the end of the day) and without arms, the robot is starting "normally".

I still have some question:
> check from the canloader if the MAIS board is correctly working or if it is broken.

*   How should I check this ?

> Check the power supply voltage given by the iKart.

*   What is supposed to be the good value?

> Check the led colors on the MAIS board.

*   Which led are these and what are the correct led colors?

> Update the MAIS firmware.

*   Correct me @Tobias-Fischer  but I think we already did it last thursday.

Tobias-Fischer, 2016-01-25:
We didn't update the MAIS, because the boards were not reachable by canLoader. It seems like the voltage is too low: http://wiki.icub.org/wiki/IKart#External_power_supply_.28read_carefully.21.29 (better check with IIT though). What do you mean by upper/lower voltage?!?

Did you contact icub-support via Skype?

gregoire-pointeau, 2016-01-25:
Hum, we have 2 block of alimentation (that need to be both active).
I didn't contact icub-support by skype yet, I want to be sure that we cannot fix it simply.

Tobias-Fischer, 2016-01-25:
Hmm .. no idea how it works with two power supplies. I guess there is no easy fix, and you need to contact icub-support ..

julijenv, 2016-01-25:
Hi @gregoire-pointeau , come on skype, we could check faster, I guess I had already added you on our contact list

gregoire-pointeau, 2016-01-25:
skype: gregoire_pointeau

gregoire-pointeau, 2016-01-25:
Solved.
Problem was coming from the flashing of the firmware to the boards that was somehow unclean, this is for right hand.
For left hand, problem is hardware: problem of connection of two (I don't know the real name @julijenv ) card of the fingers. We can have the robot without the 2 last fingers of the left hand (skin and motors) and without the skin of the left thumb.

Thanks you julien and Tobias !

closed 2016-01-25 by gregoire-pointeau