robotology/mesh-support #168: Camera could not be opened

marijnfs, 2016-01-22:
I have a problem that the camera wont 'open' when running moddev:

    icub@pc104:~$ icubmoddev --from camera/dragonfly2_config_left.ini&
    [1] 5444
    icub@pc104:~$ [DEBUG]Format = 1
    [ERROR]GUID=00b09d01009a72ed camera not found
    [ERROR]LINE: 266
    DragonflyDeviceDriver2: can't open camera
    yarpdev: ***ERROR*** driver <dragonfly2> was found but could not open
    yarpdev: ***ERROR*** driver <grabber> was found but could not open
    yarpdev: ***ERROR*** device not available.

yarpdev --list shows dragonfly devices:

    Device "dragonfly2", available on request (found in /usr/local/src/robot/icub-main/build-pc104/lib/iCub/dragonfly2.so library).
    Device "dragonfly2raw", available on request (found in /usr/local/src/robot/icub-main/build-pc104/lib/iCub/dragonfly2.so library).

barbalberto, 2016-01-22:
It looks like the GUID in your config file is not correct, please refer to this guide ```http://wiki.icub.org/brain/group__icub__dragonfly2.html ``` for the instruction on how to get the GUID.
It is specific for each cam (left/right) so it has to be updated when changing cameras.

marijnfs, 2016-01-22:
Hmm strange, we just updated it but maybe it gets reset when rebuilding icub-main or so? I would expect a git pull to refuse when it overwrites some files..

barbalberto, 2016-01-22:
You need to update the source in icub-main/app/robots/<robotname>/... and not the one in  the build.
The config in the build get overridden each time the cmake / make is run.

When the change is made in the source and you are happy about it, you can commit the change in the repo

closed 2016-01-28 by marijnfs