robotology/mesh-support #170: Calibration Fails after firmware update

jypuigbo, 2016-01-26:
Hi, 

I've just updated icub-main and icub-firmware and when starting robotInterface I get the following errors: 

```
Please verify that the target of calibrate action is a controlBoardWrapper2 device.                                                       This doesn't prevent the robotInterface to correctly operate, but no remote calibration and homing will be available 
Torso_Calibrator : detected an hardware fault on joint  0  before starting the calibration procedure! 
Torso_Calibrator : detected an hardware fault on joint  1  before starting the calibration procedure! 
...
```
For some joints in the arms, and torso (not head and hands). I attach the whole ``` log.txt``` . 

There was no change in the calibration, and we just used the files available in icub-main/build/share/robots/icub-barcelona01. 

Any insight will be useful. 

Thanks, 

Jordi

[log_pc104_robotInterface_11611.txt](https://github.com/robotology/icub-support/files/105084/log_pc104_robotInterface_11611.txt)

randaz81, 2016-01-26:
Dear @jypuigbo,
it's not possibile to know the cause of an hardware fault that happened in a previous launch of robotInterface, like in your case ```(Torso_Calibrator : detected an hardware fault on joint  0  before starting the calibration procedure!)```.

Please post a new log relaunching robotInterface after powercycling the motor control boards.

jypuigbo, 2016-01-26:
Thanks, this solved it. My apologies for not trying it first. 

Jordi

closed 2016-01-26 by jypuigbo