robotology/mesh-support #171: Broken wires on left leg

KevinAStein, 2016-01-27:
Hi,
We have a couple of broken wires on the left leg, precisely on top of the knee joint (see picture below).
This seems not to have affected the functionality of the robot (robotInterface starting without problems, all demos still working the same way), but without skin sensors. They have been broken for a while, but we plan to repair them today/this week.

What do these wires connect exactly?

[image]

KevinAStein, 2016-01-27:
Follow-up question, is it ok to remove this piece from the robot?
Is it removable by just unscrewing the screw on top of the part? (which seems to be the one that keeps it there?)

super-ste, 2016-01-27:
Hi @KevinAStein 
the device whose wires got disconnected is hte board used to measure the deflection of the spring in your SEA setup.
From the picture you sent, looks like the spring ("C" shaped steel thingy which came in a box with your robot) is not mounted therefore I'd expect you read no deflection there.

It is OKAY to remove that board as long as you remember what you are doing and are okay not to measure the defleciton of the spring.
And YES you guessed correctly as which one is the screw to remove :smiley: 

If you want to resolder the wires on the NEI (this is the name of the device) plese follow the order below

[image]

KevinAStein, 2016-01-27:
HI @super-ste 
Thanks, we already resoldered the wires as per correct order.

I'm not sure about how to test if it is working again, I mean, if we now mount the spring, then how do we read the deflection using this board? 
We have mounted springs before but we used to read encoders motor side and joint side to see the difference between the two, and probably never used this "hte board"?

Thanks!

super-ste, 2016-01-27:
@randaz81 @maggia80 
this goes far beyond my reach... any hint?

hu-yue, 2016-01-29:
@randaz81 @maggia80 should I open anew issue for this in QA since it is not related to "broken wire"?

closed 2018-01-16 by julijenv