robotology/mesh-support #172: Skin board 10 and 11 unreachable

hu-yue, 2016-01-27:
Hi,
After a long time, today we mounted the skin boards back and updated all the firmware from `canLoader` as well. However, if we switch the robot on with skin sensors only, it says that the board 10 is unreachable, if we comment the left leg/board 10, then we can see that also 11 is not reachable.

The config file we're using is simply

```xml
<?xml version="1.0" encoding="UTF-8" ?>
<!DOCTYPE robot PUBLIC "-//YARP//DTD robotInterface 1.0//EN" "http://www.icub.org/DTD/robotInterfaceV1.0.dtd">

<robot name="iCubHeidelberg01" build="1" portprefix="icub">
    <params file="hardware/electronics/pc104.xml" />

    <!--  SKINS -->
<!--    <devices file="wrappers/skin/left_leg-skin_wrapper.xml" /> -->
<!--    <devices file="hardware/skin/left_leg-ems10-skin.xml" />-->
    <devices file="wrappers/skin/right_leg-skin_wrapper.xml" />
    <devices file="hardware/skin/right_leg-ems11-skin.xml" />
	
    <!--  CALIBRATORS -->
</robot> 
```

The log is [here](http://orb.iwr.uni-heidelberg.de/ftp/icub_logs/skin_right_ypmrprunlog_27_01_2016_02_58_36.log).

Thanks!

p.s. we have not tried to run the robot with the other parts yet.

julijenv, 2016-01-27:
Hi @hu-yue ,
could you make some picture of the boards 10 and 11 before putting back the cover in order to make sure that the Ethernet cable is correctly coonected.

KevinAStein, 2016-01-27:
Hi @julijenv here are the pictures:
EMS 10
[image]

EMS 11
[image]

But since I could see the boards and all the connections of the CAN1 and CAN2 (10B1-13, 11B1-13), doesn't that mean that the connection is fine? or that not necessary mean that the communication is working?

julijenv, 2016-01-27:
Hi @hu-yue ,
you are perfectly right if you can see every can line of the boards EMS10 and 11 it means that the board are speaking and so i'll tend to think that something is going wrong at the level of firmware/config files because from the log i can see this sentence:
<#STRING_START#>in EthReceiver::config() the config socket has queue size =  2097152 ; you request ETHRECEIVER_BUFFER_SIZE=  
<#STRING_END#>16
1.012191
4
ANyone to help @randaz81 @marcoaccame ?

hu-yue, 2016-02-01:
@randaz81 @marcoaccame 

I tried to run `robotInterface` with everything (FT,motors,calibrators etc)including the skins and I don't get the message that the skin boards are not reachable anymore, and the skin sensors also work fine (tried with skin gui).

But if I run `robotInterface` with skin only, as stated in the first comment, then it says that boards 10 and 11 are not reachable. Is there any dependency that I do not remember for which if I want to test skin only I have to include something else other than the skin wrappers and ems configurations?

Thanks!

closed 2018-01-16 by julijenv