robotology/mesh-support #174: iCub pc104 problem - robotInterface hangs

AlexAntn, 2016-01-27:
Hello All, 

We have recently upgraded the Plymouth iCub (now it has skin), and we have been facing some problems in the general configurations.

Initially, the iCub had issues with the arms calibration, upon starting the robotInterface (the arms were slightly off the right position).

We tried to run our project (POETICON++), but actionsRenderingEngine was not working, returning a "[FATAL] attempted to read from empty bottle".

When trying to solve this problem, we updated Yarp and other applications (icub-main and icub-contrib). We faced some problems stated in this issue [https://github.com/robotology/QA/issues/86](https://github.com/robotology/QA/issues/86), and we updated all yarp folders, including the one in pc104 (using instructions provided here: [http://wiki.icub.org/wiki/Compilation_on_the_pc104](https://github.com/robotology/QA/issues/86) ).

We had some problems with "clock skewed", but it compiled and generated all necessary binaries.

When we tried to run robotInterface, It did not run anymore. I further tried to update the firmware, using instructions provided here: [http://wiki.icub.org/wiki/Firmware](http://wiki.icub.org/wiki/Firmware), running the icub-firmware-build/updateRobot.sh script. All elements on the list showed an "ALL OK" tag, but robotInterface still did not work.

In order to try and debug it, i ran the robotInterface binary in a terminal (ssh icub@pc104). It did not give any errors, however it hanged after opening the ports to the different motors (though it never opened the /icub/inertial port). It did, however, return a huge list of warnings (about 3-4 for each motor), saying that the xml files were not what it expected. 

Some of the warnings were as follows:

"[Warning] Found  while expecting "robot""
"[Warning] Found "iCubPlymouth01" while expecting "icub""

I double checked the yarp. icub-main and icub-firmware-shared installation, and everytime a ROBOT_NAME was required, iCubPlymouth01 was used.

Could you provide any help in this issue? It is quite urgent, as we cannot use iCub as it is.

Thanks in advance!

barbalberto, 2016-01-27:
Can you post the output of robotInterface, please?

AlexAntn, 2016-01-27:
Here is the output:

[output.txt](https://github.com/robotology/icub-support/files/106890/output.txt)

Thanks!

barbalberto, 2016-02-01:
A **question**: if you start the robotinterface, after it finishes to print all strat-up informationm if you wait let's say 5 minutes or so, do you see  a print like "robotInterface is running happily" or similar?

It looks that some of the required modules are simply not loaded at all. 
I think the first thing to do would be fixing all the warnings related to config files, maybe the problem is hidden under all this flooding of warnings.

AlexAntn, 2016-02-01:
RobotInterface doesn't print any other messages after all the start-up.

lorejam, 2016-02-01:
Hi Alex,
indeed, you seem to have problems in the config files, starting from the
main one (icub_all.xml).

First , are you sure that the folder you are loading the file to is the
correct one?
Then, the beginning of the icub_all.xml file that you load shuold look like:

<?xml version="1.0" encoding="UTF-8" ?>
<!DOCTYPE robot PUBLIC "-//YARP//DTD robotInterface 1.0//EN" "
http://www.icub.org/DTD/robotInterfaceV1.0.dtd">
<robot name="iCubPlymouth01" build="1" portprefix="icub">
....
....

Can you verify this first?

-- 
===================================================================

Lorenzo Jamone - Research Associate
http://lorejam.blogspot.pt/
ljamone@isr.tecnico.ulisboa.pt <Tecnicojasv@isr.tecnico.ulisboa.pt>

VisLab - http://vislab.isr.ist.utl.pt
Instituto Superior Técnico - http://tecnico.ulisboa.pt/

Institute for Systems and Robotics, Torre Norte (7th floor).
Av. Rovisco Pais, 1
1049-001 Lisbon, Portugal

Office (room 7.25): +351218418299
Laboratory (room 7.26): +351218418050

====================================================================

AlexAntn, 2016-02-01:
Hello @lorejam 

The file is loaded from /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubPlymouth01/./icub_all.xml, which i assume is correct (since it is loading from the iCubPlymouth configuration files)

however, the file itself is different from what you described:

<?xml version="1.0" encoding="UTF-8" ?>
<robot name="icub">

    
...
...

barbalberto, 2016-02-01:
yes, there are a lot of thing to be fixed in the xml files related to the warnings you see, but they does not prevent the robotInterface to start.

All the config files seems correct, the only thing I can say it that maybe there is something wrong in the setup or something went wrong while updating the firmware.

Can you try the following commands and post the output?
```
which robotInterface
sudo updatedb 
locate canmotioncontrol.so
```

AlexAntn, 2016-02-01:
which robotInterface: 
/usr/local/src/robot/icub-main/build-pc104/bin/robotInterface

sudo updatedb: 
sudo: updatedb: command not found

locate canmotioncontrol.so: 
-bash: locate: command not found

apparently i need to install the package mlocate, should i go ahead? 
(haven't done it yet as it might be an indication something is wrong with this installation)

barbalberto, 2016-02-01:
You can try with the 'find' alternative
```
sudo find /usr/ -iname canmotioncontrol.so
sudo find /lib* -iname canmotioncontrol.so
```

barbalberto, 2016-02-01:
Another test is to run is ``` yarp plugin --all```
and then give the output, thanks.

AlexAntn, 2016-02-01:
sudo find /usr/ -iname canmotioncontrol.so:
/usr/local/src/robot/icub-main/build-pc104/lib/iCub/canmotioncontrol.so

sudo find /lib* -iname canmotioncontrol.so
empty

yarp plugin --all
- it prints a lot of messages "Cannot find or load shared library", then it hangs. These messages are not present on the output file, which is here: [output.txt](https://github.com/robotology/icub-support/files/112626/output.txt)

Thanks!

barbalberto, 2016-02-02:
So, are you telling that also ```yarp plugin``` command hangs at a certain point?

If this is the case, my only guess is that something went wrong during compilation. Maybe CMakeCache got some wrong path for whatever reason.

First thing I'll try is to completely remove the build-pc104 of all projects -yarp, icub-main - icub-firmware-shared, redo a git pull (just in case) and recompile everything. Be sure to be logged into the pc104 when you run ```ccmake ..``` command.

AlexAntn, 2016-02-02:
hello @barbalberto ,

I recompiled yarp, icub-main and icub-firmware-shared, after running "git pull" on each. 

Is the iCubPlymouth01 an ETH icub?

anyway, i ran all the other commands again, and this was the output:
`sudo find /usr/ -iname canmotioncontrol.so` - /usr/local/src/robot/icub-main/build-pc104/lib/iCub/canmotioncontrol.so

`sudo find /lib* -iname canmotioncontrol.so` - empty

`yarp plugin --all` - [output.txt](https://github.com/robotology/icub-support/files/113894/output1.txt)

Then i ran robotInterface again, and this was the output: [output.txt](https://github.com/robotology/icub-support/files/113895/output.txt)
It crashes with a segfault.

barbalberto, 2016-02-02:
Ok, now it looks much better!!!  :+1: 
There was definitely something corrupted with the previous compilation.

> Is the iCubPlymouth01 an ETH icub? 

No, it is a CAN based, so you can skip compilation of all embObj* devices (does not harm but speed up compilation time)

Now to address the current situation, the error is the following

**********************************
**iCubInterface CAN protocol: 1.6**
Firmware report:
sharedcan [0] joint: 0 can_address:  1 board type: 4 (BLL) version: 1.62 build: 85 **CAN_protocol:1.4**
sharedcan [0] joint: 1 can_address:  1 board type: 4 (BLL) version: 1.62 build: 85 CAN_protocol:1.4
sharedcan [0] joint: 2 can_address:  2 board type: 3 (4DC) version: 2.15 build: 85 CAN_protocol:1.4
sharedcan [0] joint: 3 can_address:  2 board type: 3 (4DC) version: 2.15 build: 85 CAN_protocol:1.4
sharedcan [0] joint: 4 can_address:  3 board type: 3 (4DC) version: 2.15 build: 85 CAN_protocol:1.4
sharedcan [0] joint: 5 can_address:  3 board type: 3 (4DC) version: 2.15 build: 85 CAN_protocol:1.4
**********************************

##################################################################################

**It has been detected that your control boards are not using the same**
**CAN protocol used by iCubinterface. iCubInterface cannot continue.**
**Please update your system (iCubInterface and/or your control board firmware.**
**For further information please visit: http://wiki.icub.org/wiki/Firmware.**

#################################################################################

checkFirmwareVersions() failed. CanBusMotionControl::open returning false, 

So the next step is to update the firmware on CAN boards. Maybe the last update of repos required an update of firmware.

julijenv, 2016-02-02:
Hi iCubPlymouth01 users,
could you please send me the entire log file, i could be helpful at that point of the process

AlexAntn, 2016-02-02:
hello @barbalberto,

I ran the firmware update script "updateRobot.sh", and the result was:
`cfw2can 0 1 2BLLDC.1.62.out.S --> ALL OK
cfw2can 0 2 4DC.2.15.out.S --> ALL OK
cfw2can 0 5 2BLL.1.54.out.S --> ALL OK
cfw2can 0 6 2BLL.1.52.out.S --> ALL OK
cfw2can 1 1 2BLL.1.50.out.S --> ALL OK
cfw2can 1 2 2BLL.1.57.out.S --> ALL OK
cfw2can 1 3 4DC.2.19.out.S --> ALL OK
cfw2can 5 5 4DC.2.28.out.S --> ALL OK
cfw2can 5 7 4DC.2.30.out.S --> ALL OK
cfw2can 2 1 2BLL.1.50.out.S --> ALL OK
cfw2can 2 2 2BLL.1.57.out.S --> ALL OK
cfw2can 2 3 4DC.2.19.out.S --> ALL OK
cfw2can 6 5 4DC.2.28.out.S --> ALL OK
cfw2can 6 7 4DC.2.30.out.S --> ALL OK
cfw2can 3 5 2BLL.1.51.out.S --> ALL OK
cfw2can 3 6 2BLL.1.51.out.S --> ALL OK
cfw2can 3 7 2BLL.1.51.out.S --> ALL OK
cfw2can 4 8 2BLL.1.51.out.S --> ALL OK
cfw2can 4 9 2BLL.1.51.out.S --> ALL OK
cfw2can 4 10 2BLL.1.51.out.S --> ALL OK`

I then tried to run the robotInterface again, but the error persisted: [output.txt](https://github.com/robotology/icub-support/files/114068/output.txt)

@julijenv : is this the log file you asked for (the output of robotInterface)? Or do you need any other logs, and if so which?

Thanks!!

barbalberto, 2016-02-02:
@randaz81: do you have any idea about this?

randaz81, 2016-02-03:
Dear @AlexAntn,
your last log shows that you are still running firmware build 85, but the current fw release is build 87.
So please git pull the repo icub-firmware-build and re-run the script updateRobot.sh

AlexAntn, 2016-02-03:
Hello @randaz81 

I updated the icub-firmware-build, ran the script, started robotInterface and now the robot moves!

However, it is not initializing properly. Both the torso and the upper arms are not moving when i start robotInterface (only the hands, fingers, lower arms and head move).

RobotInterface tries repeatedly to interact with the arms, but keeps failing. i have printed the output of the program: 
[output.txt](https://github.com/robotology/icub-support/files/115554/output.txt)

Thanks!

randaz81, 2016-02-03:
mmmh, have you power-cycled the motor boards after the firmware update?

AlexAntn, 2016-02-03:
@randaz81: I had not, so i did now, and tried again.

Now it moves also the torso on startup, but the upper left and upper right arms are still not responding. Here is the output of RobotInterface: 
[output.txt](https://github.com/robotology/icub-support/files/115829/output.txt)

Also, i tried to run the yarpmotorgui, but it presented the following popup error messages:
Error while opening device for part /icub/head
Error while opening device for part /icub/torso
Error while opening device for part /icub/left_arm
Error while opening device for part /icub/right_arm

(i didn't select any of the legs)

and then is closes.

Thanks!!

randaz81, 2016-02-03:
uhm, it's not clear: the log says that the upper left/right arm calibration was successful. Is it the right log?
Regarding the yarpmotorgui issue, I suppose that it is running on another machine. Can you double check the yarp is updated and comppiled correctly?

AlexAntn, 2016-02-03:
Hello @randaz81 

I checked yarp, but it was the current version. i remade it anyway. I tried again to run yarpmotorgui, but it gave me the same error. output: 
[output_motorgui.txt](https://github.com/robotology/icub-support/files/116031/output_motorgui.txt)

As for the robotInterface and the upper arms: it keeps repeating the message:
`[INFO]icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet`
and the same for the right arm. Output: 

Also, when i terminate the robotInterface, the following message appears, before "aborting":
[RobotInterfaceError.txt](https://github.com/robotology/icub-support/files/116039/RobotInterfaceError.txt)

I also tried running the other modules present in the iCubStartup XML. All started, except for gravity compensator, but eventually they all crash. I ran gravityCompensator in a terminal, and it keeps getting this error:
`[ERROR]expecting protocol 1 4 1, but remotecontrolboard returned protocol version 1 5 1`

barbalberto, 2016-02-04:
Side note. 
```[INFO]icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet```
This message is perfectly fine. It in an info message, not error. It means the cartesian server is running on the pc104 and it is waiting for the solver to be launched.

AlexAntn, 2016-02-04:
Information update: All the modules in the iCubStartup XML (iKinCartesianSolver, iKinGazeCtrl, wholeBodyDynamics, gravityCompensator and fingersTuner), they all return the same Error message, and keep retrying until reaching timeout:
'[ERROR]expecting protocol 1 4 1, but remotecontrolboard returned protocol version 1 5 1'
and...
'yarpdev: ***ERROR*** driver <remote_controlboard> was found but could not open'

yarpmotorgui still does not run.

I updated and rebuilt yarp, icub-main and icub-contrib-comon on this computer (where we run these modules).

Thanks!

barbalberto, 2016-02-04:
This means that the machines you are running the modules and the pc104 have a different version of yarp. From the message it looks like pc104 is the most updated. 
Check if there is no older version of yarp shadowing the latest build (use ```which yarpmotorgui``` or any other executable and verify it points to the correct build folder.
Check also the ```node``` filed of iCubStartup xml to look where the modules are supposed to run and verify that the version of yarp on those machines is updated (and no shadow installation are present as well)

AlexAntn, 2016-02-04:
Ok, i checked that someone had made an installation using sudo, resulting in two different installations of yarp, icub-main and icub-contrib-comon.

I fixed that, and now all modules on startup and yarpmotorgui are working!

Thanks!!

closed 2016-02-04 by AlexAntn