robotology/mesh-support #175: Working with faceexpressions

marijnfs, 2016-01-28:
Hi, I'm trying to use the face expressions, but they don't seem to work. I have the serial and serialport built, and run the device like this:

    yarpdev --name /icub/face/raw --device serial --subdevice serialport --context faceExpressions --from serialport.ini

The port is created but sending commands to the /icub/face/raw/in port like

    set raw L03

doesn't do anything; no 'ok' or something like that. yarpdev does seem to see the command and says:

    The [DEBUG]Sending string: set

Any suggestions on how to fix this?

julijenv, 2016-01-28:
Hi @marijnfs ,
First of everything could you make sure that both power cord and /USB cable are connected?
Launch robotInterface and launch the module facexpression from (g)yarpmanager.
go inside $ICUB_ROOT/app/faceexpression
launch cycle.sh
let us know

marijnfs, 2016-01-28:
The yarpmanager doesn't show me much. There is a templates directory with a faceexpressions xml file but nothing to run it with.

marijnfs, 2016-01-28:
The USB seems connected, don't know which is the power cable?
[image]

julijenv, 2016-01-28:
Plese come on skype and open as well a teamviewer session, thx

julijenv, 2016-01-29:
we succeed to understand that the hardware of the face Expression is alright (we unlight the face expressioon with the old style command line) but it appears that the overall system files and compilation,  isnt as good as it could be even if we check step by step the procedure on the wiki for server and pc104 side.
yarpmanager does not launch corrrectly or the cluster is not correctly set.
Is there anyone @barbalberto , @drdanz , @apaikan able to fix an appointment for a teamviewer session to check that the compilation/installation is correct in order to give a perfect understanding of the system.
thx in advance .

marijnfs, 2016-02-08:
Hello julijenv,
The face expressions are again not working. Do you remember how we fixed it last time? I tried connecting the usb cable again but it doesn't help.

julijenv, 2016-02-08:
Hi @marijnfs ,
first try unconnect and reconnect the usb port and "dmesg tail" to make sure the usb is detected by the pc104.
if yes, proceed by launching from yarpmanager faceexpression module
and later gooes to the repository : $ICUB_MAIN/app/faceExpresion/scripti
and launch cycle.sh until yuo get the expression you want.\
let me know, im still behind skype so keep in touch

marijnfs, 2016-02-08:
So the device connects: 

     [ 9469.984980] cdc_acm 2-1:1.0: ttyACM0: USB ACM device

and then I start it with:

    yarpdev --name /icub/face/raw --device serial --subdevice serialport --context faceExpressions --from serialport.ini

but there doesn't seem to be any working device when setting raw commands, or connecting emotionInterface

julijenv, 2016-02-08:
So the device connects:

 [ 9469.984980] cdc_acm 2-1:1.0: ttyACM0: USB ACM device
and then I start it with:

yarpdev --name /icub/face/raw --device serial --subdevice serialport --context faceExpressions --from serialport.ini

AFAIK you just miss to launch from $ICUB_ROOT/app/faceExpression/script
cycle.sh
Let us know

marijnfs, 2016-02-08:
Yes the cycle just sends commands to the emotion interface and doesn't work.

marijnfs, 2016-02-08:
The yarpdev seems to work normally:
```
icub@pc104:~$ yarpdev --name /icub/face/raw --device serial --subdevice serialport --context faceExpressions --from serialport.ini
[DEBUG]Subdevice serialport
[INFO]Starting Serial Port in /dev/ttyACM0 
[INFO]created device <serialport>. See C++ class yarp::dev::SerialDeviceDriver for documentation.
yarp: Port /icub/face/raw/in active at tcp://10.0.0.2:10045
yarp: Port /icub/face/raw/out active at tcp://10.0.0.2:10046
[INFO]created wrapper <serial>. See C++ class yarp::dev::ServerSerial for documentation.
[INFO]Server Serial starting
yarp: Port /icub/face/raw/quit active at tcp://10.0.0.2:10047
[INFO]device active in background...
[INFO]device active in background...
[INFO]device active in background...
```

but when i ctr+x it, it doesn't close:

```
^C[INFO][try 1 of 3] Trying to shut down /icub/face/raw/quit
^C[INFO][try 2 of 3] Trying to shut down /icub/face/raw/quit
Terminator port not found
^C[INFO][try 3 of 3] Trying to shut down /icub/face/raw/quit
Terminator port not found
```

marijnfs, 2016-02-08:
The file is I think uses is: /usr/local/src/robot/icub-main/app/faceExpressions/conf/serialport.ini
```
/// A initialization file for configuring chico facial expressions interface
comport /dev/ttyACM0
baudrate 9600
xonlim 0
xofflim 0
readmincharacters 1
readtimeoutmsec 0
parityenb 0
paritymode EVEN
ctsenb 0
rtsenb 0
xinenb 0
xoutenb 0
modem 0
rcvenb 0
dsrenb 0
dtrdisable 0
databits 7

```

alecive, 2016-02-08:
For that matters, the fact that it does not close properly is an expected behavior (see https://github.com/robotology/yarp/issues/469 ). I would call it a feature at this point :) @randaz81 @francesco-romano

apaikan, 2016-02-09:
Hi,
I am informed by @julijenv  that there is another issues regarding `yarpmanager`. It is a different issues but @julijenv guess that it could be related to this. 
anyway, can you show me the environmental variables of the machine (pc104 and the other hosts which you run `yarpmanager`). Run the following commands and post the answer, please: 
 
-  `$ which yarp`
-  `$ which robotInterface`
-  `$ env`  or alternatively your `~/.bashrc` 

my guess is that your `YARP_DATA_DIRS` is not pointing to the folder where icub applications are installed. Or you still have the applications (.xml) in the template format (.xml.template).

francesco-romano, 2016-02-09:
@alecive we are just leaving the PR in the decanter. The older the better :wine_glass:

marijnfs, 2016-02-09:
For the PC104:

```
icub@pc104:~$ which yarp
/usr/local/bin/yarp
icub@pc104:~$ which robotInterface
/usr/local/src/robot/icub-main/build-pc104/bin/robotInterface

icub@pc104:~$ env
LC_PAPER=de_CH.UTF-8
CFW2CANAPI_DIR=/lib/modules/3.2.0-4+icub.1-rt-amd64/iCubDrivers//cfw002/
LC_ADDRESS=de_CH.UTF-8
LC_MONETARY=de_CH.UTF-8
TERM=xterm
SHELL=/bin/bash
XDG_SESSION_COOKIE=971a85bdbfefcb6a5bed739b47cb3fae-1455027847.989567-543447301
YARP_TRACE_ENABLE=0
YARP_DIR=/usr/local/src/robot/yarp/build-pc104
SSH_CLIENT=10.0.0.1 45043 22
YARP_FORWARD_LOG_ENABLE=0
ICUB_ROBOTNAME=iCubLugano01
YARP_ROOT=/usr/local/src/robot/yarp
LC_NUMERIC=de_CH.UTF-8
SSH_TTY=/dev/pts/2
YARP_VERBOSE_OUTPUT=0
USER=icub
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arj=01;31:*.taz=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lz=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.axv=01;35:*.anx=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.axa=00;36:*.oga=00;36:*.spx=00;36:*.xspf=00;36:
LC_TELEPHONE=de_CH.UTF-8
YARP_COLORED_OUTPUT=1
ROBOT_CODE=/usr/local/src/robot
ICUB_DIR=/usr/local/src/robot/icub-main/build-pc104
MAIL=/var/mail/icub
PATH=/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/usr/local/src/robot/icub-main/build-pc104/bin:/usr/local/src/robot/yarp/build-pc104/bin
LC_IDENTIFICATION=de_CH.UTF-8
PWD=/home/icub
CMAKE_BUILD_TYPE=Release
LANG=en_US.UTF-8
YARP_ROBOT_NAME=iCubLugano01
FIRMWARE_BIN=/usr/local/src/robot/icub-firmware/build
LC_MEASUREMENT=de_CH.UTF-8
ICUBcontrib_DIR=/usr/local/src/robot/iCubContrib
YARP_DATA_DIRS=/usr/local/src/robot/yarp/build-pc104/share/yarp:/usr/local/src/robot/icub-main/build-pc104/share/iCub:/usr/local/src/robot/iCubContrib/share/ICUBcontrib
SHLVL=1
HOME=/home/icub
LOGNAME=icub
icub_firmware_shared_DIR=/usr/local/src/robot/icub-firmware-shared/build-pc104
SSH_CONNECTION=10.0.0.1 45043 10.0.0.2 22
ICUB_ROOT=/usr/local/src/robot/icub-main
LC_NAME=de_CH.UTF-8
_=/usr/bin/env

```

marijnfs, 2016-02-09:
for the server:```

icub@icubsrv ~ $ which yarp
/usr/local/bin/yarp
icub@icubsrv ~ $ which robotInterface
/usr/local/bin/robotInterface
icub@icubsrv ~ $ env
LC_PAPER=de_CH.UTF-8
XDG_VTNR=8
LC_ADDRESS=de_CH.UTF-8
SSH_AGENT_PID=2274
XDG_SESSION_ID=c2
LC_MONETARY=de_CH.UTF-8
GPG_AGENT_INFO=/run/user/1000/keyring-7ZWPnt/gpg:0:1
VTE_VERSION=3409
TERM=xterm
SHELL=/bin/bash
XDG_SESSION_COOKIE=4f593997c38795b66637800f561b8c60-1453293624.956967-2031041312
YARP_TRACE_ENABLE=0
YARP_DIR=/usr/local/src/robot/yarp/build
YARP_FORWARD_LOG_ENABLE=0
GJS_DEBUG_OUTPUT=stderr
YARP_ROOT=/usr/local/src/robot/yarp
LC_NUMERIC=de_CH.UTF-8
WINDOWID=42960265
GNOME_KEYRING_CONTROL=/run/user/1000/keyring-7ZWPnt
GJS_DEBUG_TOPICS=JS ERROR;JS LOG
ICUBRC_FILE_LOADED=true
YARP_VERBOSE_OUTPUT=0
USER=icub
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arj=01;31:*.taz=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lz=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.axv=01;35:*.anx=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.axa=00;36:*.oga=00;36:*.spx=00;36:*.xspf=00;36:
CLUTTER_BACKEND=x11
LC_TELEPHONE=de_CH.UTF-8
YARP_COLORED_OUTPUT=1
SSH_AUTH_SOCK=/run/user/1000/keyring-7ZWPnt/ssh
SESSION_MANAGER=local/icubsrv:@/tmp/.ICE-unix/1760,unix/icubsrv:/tmp/.ICE-unix/1760
USERNAME=icub
DEFAULTS_PATH=/usr/share/gconf/cinnamon.default.path
XDG_CONFIG_DIRS=/etc/xdg/xdg-cinnamon:/etc/xdg
ROBOT_CODE=/usr/local/src/robot
ICUB_DIR=/usr/local/src/robot/icub-main/build
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/local/src/robot/icub-main/build/bin:/usr/local/src/robot/yarp/build/bin
DESKTOP_SESSION=cinnamon
GDM_XSERVER_LOCATION=local
LC_IDENTIFICATION=de_CH.UTF-8
PWD=/home/icub
CMAKE_BUILD_TYPE=Release
LANG=en_US.UTF-8
YARP_ROBOT_NAME=iCubLugano01
FIRMWARE_BIN=/usr/local/src/robot/icub-firmware/build
MANDATORY_PATH=/usr/share/gconf/cinnamon.mandatory.path
LC_MEASUREMENT=de_CH.UTF-8
ICUBcontrib_DIR=/usr/local/src/robot/iCubContrib
YARP_DATA_DIRS=/usr/local/src/robot/yarp/build/share/yarp:/usr/local/src/robot/icub-main/build/share/iCub:/usr/local/src/robot/iCubContrib/share/ICUBcontrib
MDM_XSERVER_LOCATION=local
GDMSESSION=cinnamon
CINNAMON_VERSION=2.6.11
SHLVL=1
XDG_SEAT=seat0
HOME=/home/icub
GNOME_DESKTOP_SESSION_ID=this-is-deprecated
XDG_SESSION_DESKTOP=cinnamon
LOGNAME=icub
icub_firmware_shared_DIR=/usr/local/src/robot/icub-firmware-shared/build
XDG_DATA_DIRS=/usr/share/cinnamon:/usr/share/gnome:/usr/local/share/:/usr/share/:/usr/share/mdm/
DBUS_SESSION_BUS_ADDRESS=unix:abstract=/tmp/dbus-U4mxbdXLu0,guid=6ce221b6fb0cb8e6c24c6d16569f803b
LESSOPEN=| /usr/bin/lesspipe %s
ICUB_ROOT=/usr/local/src/robot/icub-main
MDMSESSION=cinnamon
TEXTDOMAIN=im-config
WINDOWPATH=8
XDG_RUNTIME_DIR=/run/user/1000
DISPLAY=:0
MDM_LANG=en_US.UTF-8
XDG_CURRENT_DESKTOP=X-Cinnamon
LESSCLOSE=/usr/bin/lesspipe %s %s
TEXTDOMAINDIR=/usr/share/locale/
COLORTERM=gnome-terminal
XAUTHORITY=/home/icub/.Xauthority
LC_NAME=de_CH.UTF-8
_=/usr/bin/env

```

marijnfs, 2016-02-09:
Yes it could be that its a problem of the yarp datadir, where should it point to exactly?

closed 2019-10-12 by pattacini