robotology/mesh-support #177: Encoder has spikes warnings on board 8

hu-yue, 2016-02-01:
Hi,
We have encountered the warning on encoder spikes on board 8 adr 0 (left ankle pitch I guess?)
```
[WARNING] from BOARD 10.0.1.8 (rightLeg), src LOCAL, adr 0, time 861s 69m 539u: (code 0x0200000a, par16 0x0000 par64 0x0000000000000002) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = ID of joint. + .
```
I remember we had similar issues before, and back then I had to send the robot back to IIT for repair (but back then the `robotInterface` was flushed with warnings like this).
Is there anything we can do to fix the problem?

Thanks!

julijenv, 2016-02-02:
Hi @hu-yue ,
Could you please send the enitre logfile which could be useful to get more datas  on this regard.
thx in advance

hu-yue, 2016-02-02:
Hi @julijenv, I somehow have not seen this warning anymore today, I'll close this issue for the time being and reopen in case it shows up again.

closed 2016-02-02 by hu-yue