robotology/mesh-support #180: Robot failing to calibrate

vtikha, 2016-02-03:
We seem to have a random issue where the robot does not calibrate the torso. 
The only debug info is the following: 
Any hints are appreciated 
```
[ERROR] from BOARD 10.0.1.5 (Torso), src LOCAL, adr 0, time 390s 479m 135u: (code 0x05000012, par16 0x0003 par64 0x0000000000000001) -> CFG: EOtheEncodeerReader cannot be configured. verification fails. in p16 the number of joints (lsb) and encoders (msb), in p64 the error status of verification reading + .
[ERROR] from BOARD 10.0.1.5 (Torso), src LOCAL, adr 0, time 390s 479m 220u: (code 0x0500000d, par16 0x0000 par64 0x0000000000000026) -> CFG: EOtheMotionController cannot be configured. verification of encoders fails. see other messages for more details + .
```

randaz81, 2016-02-03:
You started the robot too early after turning on the motors @marcoaccame

vtikha, 2016-02-03:
Eaiti g for more than a minute is too early?

julijenv, 2016-02-05:
hopefully this time it will make the diffrence."
I change the connector.
think alright now, lets see

closed 2016-10-04 by julijenv