robotology/mesh-support #182: recommended iCub power supplies

frank-foerster, 2016-02-08:
Hi all,
We are considering getting quieter power supplies for the iCub (we're still having the old Xantrex XFR 60-46 + Xantrex 35-35 ones).  So I was wondering if there are any official recommendations as the ones mentioned on the iCub Wiki (http://wiki.icub.org/wiki/Power_supply) seem to be outdated since quite a while. 
The listed line of Xantrex bench power supplies were bought by Ametek in 2008 (http://www.programmablepower.com/brands/sorensen.htm) and go now under the name 'Sorensen'. The XMP 2600 model, also still linked on the wiki, appears to have been discontinued (http://www.programmablepower.com/products/XMP/downloads/XMP_Datasheet.pdf). 

The Osaka group recommended Elektro-Automatik's EA PS 8080-60 2U, which is apparently much quieter than the old Xantrex ones. 
So I was wondering if these are also recommend by the official iCub support or what the official recommendation is. 
I couldn't find any noise measurements for any of the above mentioned power supplies, so I guess I have to rely on subjective opinions.
For our purposes, generally, the quieter the power supply, the better.

Many thanks

Frank

gsaponaro, 2016-02-08:
We in Lisbon would also like to get quieter power supplies and possibly a noise reduction box (https://github.com/robotology/QA/issues/94). Any official suggestion from IIT would be greatly appreciated.

/cc @lorejam

julijenv, 2016-02-09:
Hi all, ( @frank-foerster , @gsaponaro )
It would be in the day add the official power supply (less rumourous than the older ones) on the wiki with the pdf file description but I'm gonna give you a little taste of it.
This power supply is now paired with another same one in order to have the same structure as before , meaning 2 power suplly one for 12V and the other for 48V, the reference is:
Elektro-automatik	EA-PS 8080-60 2U, estimated price following your supplier 1200-1500E.

julijenv, 2016-02-10:
wiki page updated
http://wiki.icub.org/wiki/Power_supply

closed 2016-02-10 by julijenv