robotology/mesh-support #183: PC104 unreachable

gregoire-pointeau, 2016-02-09:
Hello.
Since a few weeks, our PC104 was very unstable.
We loose it sometimes (unreachable) and until now the solution was to unmount the usb key, mount it on a computer, then remount it on the iCub to get it back.
We don't know if it solved the problem  because of the cable in the head that moved (very likely) or because we mounted it on another computer to "unfreeze" it somehow.

But until today we lost the PC104 totally.
This morning we started it without problem, made a few experiment with it, stopped robotInterface, then when I called the `poweroff` the console didn't gave me back the hand and was stuck, and once again we had lost the robot. This already arrived once or twice, but we always managed to restart the PC104 just after.
Since this morning we cannot restart it anymore.

Now our symptoms are:
- We CAN ping the ikart, and it is working fine.
- We CANNOT ping the PC104
- The light on the USB stick is enlighten the first second then keep shut.
- PC104 fans are running
- PC104 is correctly powered.
- USB image of the PC104 works perfectly on another computer.
- We created a new USB key with instructions here: (http://wiki.icub.org/wiki/The_Linux_on_the_pc104#How_to_put_the_iso_on_a_USB_memory) but still the same: light the first second, then black, and no ping.

Our guess is that some cables are down in the PC104.

Attached: picture of the back of the head of the robot.

[image]
[image]

julijenv, 2016-02-09:
Hi @gregoire-pointeau ,
Please try to check the continuity in between the 4 pins of the usb ports and the connector plug into the pc104.
Let us know

gregoire-pointeau, 2016-02-10:
How and where exactly should we proceed ?
The fact is that the cables are not directly accessible.
Do you confirm we should check the continuity between these 2 sets of points:
[image]

How can we uncover the adapters to access the cables ?

julijenv, 2016-02-10:
Hi @gregoire-pointeau ,
So indeed it is exactely this set of wires you need to check the continuity but to get an insight of the state of the wires at the base of the USB host port (where you circled) you may first need to check the continuity at the level of the port itself I mean where the pc104 key is plugged..
you may have to untangle the wires in order to know which 4 wires are coming from this pc104 USB host port.
Is it clear enough?
Let us know

gregoire-pointeau, 2016-02-10:
Hi @julijenv, we checked the connectivity inside the usb port (where the key should be).
To sum up:
once from inside the usb port on the right, with the base on the pc104 (in red)
once from inside the usb port on the left (not circled on the picture) with the base of the pc104. (in green)

The pc104 has only 7 pins, one is the 4th pin of both usb port.
All is fine

[image]

julijenv, 2016-02-10:
Hi @gregoire-pointeau ,
could you try to boot on another usbkey? (not the one we gave you)
or in the worst case scenario why not trying to get eventually the display of the pc104 on a screen via a VGA cable.
https://svn.robotology.eu/repos/iCubHardware-pub/trunk/mechanics/public/doc/howTo/Connect_VGA_to_PC104.pdf
Let us know

gregoire-pointeau, 2016-02-10:
Hi, we already try with another key with the same result.
Concerning the VGA to PC104 we tried to craft an adaptor but it wasn't successful...
The adaptor seems to work, but we get nothing on the screen.

julijenv, 2016-02-11:
Hi @gregoire-pointeau ,
last but not the least test to do before proceeding to send you a replacement pc104,
Do you you have a ram at least 2Go for laptiop that you can momently borrow and try to replace on the pc104? (just behind the board with the heatsink and fans there is a slot for RAM, could you try to swap with a working ram and tell us if the pc104 restart?
Let us know,
from your reply we'll decide the next step (send pc104 or RAM)

gregoire-pointeau, 2016-02-11:
I can't find a tutorial to change the RAM on the PC104, do you have one ? I don't know how to do it and I don't want to break it.

julijenv, 2016-02-11:
come on skype

julijenv, 2016-02-11:
Just remove the 4 screws at the 4 corner of the pc104 (standoff for the cover basically) from now on you can unplug the pc104. truning it you will be able to see the slot for the RAM. change it, replace the pc104 on his seat and you are good to test.

gregoire-pointeau, 2016-02-11:
Changing the RAM didn't solved the problem.
The solution is to replace the PC104.

gregoire-pointeau, 2016-02-12:
@julijenv Can you confirm us the date of shipment of the new PC104 please ?
Thanks

julijenv, 2016-02-12:
Hi @gregoire-pointeau ,
should be shipped today, still waiting for the guys to pick up the package.
I'll you know all the details of the shipment ASAP 'll get them

julijenv, 2016-02-12:
The shipment will be picked up this afternoon.
your tracking number information:
http://www.dhl.com/cgi-bin/tracking.pl?AWB=5859981573,
let us know when it arrives and thast you wil have replaced it.
Of course close the issue if ever the change of the pc104 make everything works again.

gregoire-pointeau, 2016-02-15:
Hi @julijenv , we just mounted the new PC104 and started it.
This time the key is normally lit (which is new).

However we still have no access to the PC104 (the ping gives us a host unreachable since more than 10 minutes).
Is there a special configuration to do with the new PC104 ?

gregoire-pointeau, 2016-02-15:
Hi, false alarm.
We went in the log files of the PC104 and `eth0` and `eth1` had been changed in `eth3` and `eth4`.
We reset them to 0 and 1, and the iCub is alive !

Thanks !
Hi, false alarm.
We went in the log files of the PC104 and `eth0` and `eth1` had been changed in `eth3` and `eth4`.
We reset them to 0 and 1, and the iCub is alive !

Thanks !
[image]

closed 2016-02-15 by gregoire-pointeau