robotology/mesh-support #185: [iCubParis02] sharedcan [1] board 2 ABS ENCODER ERROR AXIS 0

rlober, 2016-02-11:
@iron76 
@vpadois 
@julijenv 
@super-ste

I am submitting this issue as a general support ticket for `iCubParis02`. We have noticed problems since CoDyCo Y2 demos but they have been sporadic. Now we have an encoder error on launching `robotInterface` which prevents startup. I have opened up the back and left arm to check for connection errors but I can't see anything. 

Here is a log of the error:
[log_pc104_robotInterface_3144.txt](https://github.com/robotology/icub-support/files/126514/log_pc104_robotInterface_3144.txt)

I think many of our problems originate from swapping control/motor boards during the Y2 demos and I think it would be good to do a diagnostic on all of them - but before I get ahead of myself I just wanted to post this general issue to do a first round of trouble shooting before jumping into an overhaul.

julijenv, 2016-02-11:
Hi @rlober ,
it seems to be a problem with the ABS ENCODER of the joint 2 of the left arm.
So hopefully, it is a the level of the connector plug into the board 1B1 (upper board on the left side) in the back of iCub or else it means you will need to check the sensor wire in between the sensor and the connector (the one from the pictures) - it may have an extension actually so check if  there are no wires broken at this level when you closed it last time you may have pinch some wires.

[image]

[image]

[image]

[image]

In the worst case scenario you may have to check the little AEA chipset to see if the wires are still solder on it.
Please take a look along the cable from the last picture in order toi detect where it is broken - hopefully some place easily repairable.
let us know

rlober, 2016-02-12:
Hey @julijenv, thanks for the quick response!

I just looked at th cable `L1S0` and the connectors/wires are fine as far as I can tell. What board does this cable plug into?

julijenv, 2016-02-12:
Hi @rlober ,
as mentionned on the reply of yesterday ( :P) you need to check on the upper board on the left side in the back of iCub (when you will have removed the back cover).
LEt us know
I'm on skype if ever needed

rlober, 2016-02-12:
Ok so looking at the board I can not see which wires are plugged into which sockets. The cables are all hidden behind the board and I cannot see their labels. Can you show me which of the connectors it should be plugged into?

julijenv, 2016-02-12:
ok you need to take a look at the L1S0 sensor which is located into the header pin - P13 - at the right side of the upper board on the left side when watch from the back of iCub (see different picture below and schematics).
[image]
[image]
[image]

I fear now that i consider the fact that the connector is inside the robot (so kind of difficult  to move and so to get broken) you may have some wires most probably broken at the level of the sensor. But try to see anyway the connector side at the level of the board, we never know.
Could you try to make some test (diode test) to compare the other AEA sensor on the same board but plug into the header P14 to understand which wires are broken.

Good Luck, Let us know

rlober, 2016-02-12:
Ok I will give this a try on Monday! Thanks for all the info!

rlober, 2016-02-17:
@julijenv 

Sorry for the delay. I check the connector on the board. It was hard to see but the connector was properly inserted and the wires still attached. 

After discussing with @iron76 and @vpadois we have decided to do a full overhaul of the iCubParis02 on the software side then a thorough debugging hardware side. The end result of this may be a visit to Paris or us sending the robot to Genoa.

julijenv, 2016-02-18:
Ok @rlober , let us know, I'm always ready to visit you in Paris

julijenv, 2016-03-16:
Any news with the replacement sensor?

vpadois, 2016-03-16:
Hello Julien,

If I may, we have discussed in Paris our various iCub issues and the best
way to solve them on the 12th of February with Francesco (in CC). Francesco
wrote a list of actions to perform in order to solve our problems with
iCubParis02. He transmitted this list to icub-support or whoever is in
charge of the maintenance. He confirmed to me less than a week ago that
this list was is the hands of Lorenzo (in CC).

So let me reverse the question and ask : any news with the replacement of
the sensor? 😉

Best
Vincent

rlober, 2016-06-09:
Ok so after firmware, yarp and icub-main updates we still have the same issues with the left arm board 2 axis 0 & 1.

The connectors between the elbow encoder and the board are good - checked with continuity meter.

We tried to check the encoder readings but we are unable to launch `yarprobotinterface` with the fault button pressed.

We were able to start the robot with the `left_arm` device commented out of `icub-all.xml` but we can not get any state info to check the encoder.

**New Issue**
With the left arm deactivated we we able to run `yarprobotinterface` but... for no (visually) obvious reason we got a HW fault error and the icub leaned back forcefully. The torso pitch dof rotated backwards very quickly. We had to emergency fault to prevent the robot from doing self damage. 

This is now a persistent issue as well...

Here is a log with both errors:
[log_pc104_yarprobotinterface_6481.txt](https://github.com/robotology/icub-support/files/306480/log_pc104_yarprobotinterface_6481.txt)

Help me Obi Wan Kenobi, you are my only hope. :disappointed:

jeljaik, 2016-06-09:
Let me just add that this new issue may be also related to something we're familiar with in Genova. At some point the robot's torso bent forward and the cable to one of the torso encoders on the front may have gotten pinched against the head of the supporting pole. I don't know if there's an easy way to test this. This happened a couple of times in the past with iCubGenova01 as far as I remember.

rlober, 2016-06-09:
just to add to @jeljaik's comment the cable is that of the torso roll abs encoder.

julijenv, 2016-10-04:
can be closed I guess, @jeljaik @rlober

jeljaik, 2016-10-04:
Finally yes :) thanks to you @julijenv

closed 2016-10-04 by jeljaik