robotology/mesh-support #187: Power switch/power supply issue

kt10aan, 2016-02-17:
Hi,

today we turned on the power switch for the CPU. The green light didn't come up and the pc104 was not powered. The motor switch was not working either.

The power supplies (we have the Electro-Automatik ones) were both showing 00.00 for voltage and current. Shouldn't the voltage indication be non-zero?

We opened the switch box and a continuity test shows that the switches are working.
Is it a simple matter of setting up again the power supplies? Something else may be going on?

kt10aan, 2016-02-17:
Further update:

- The Output buttons on both power supplies were at the ``off`` place

- Turning the output button on the top power supply (CPU) to ``on``, it switches to ``off`` after a second.

- Disconnecting the spinal cable from the iCub, the power supplies behave the same

- Disconnecting the serial connector from the back of the bottom power supply (motors), the bottom power supply starts normally, with Output to ``on`.

So the problem is in the top power supply (the one for the CPU). Any ideas of what I should check? I have already checked the connections in the back of the power supply.

kt10aan, 2016-02-17:
Further update:

- When on the top power supply the output button is pushed to ``on`` and after a second switches to ``off``, the middle screen shows ``OV`` (for overvoltage I assume).

maggia80, 2016-02-17:
Check the setup of the top power supply. You should have this values. In order to see this numbers you have to unlock the powersupply by pressing the Lock button.

Undervoltage: 11.50V

Voltage: 13-13.50V

Overvoltage: 15V

here you can find the datasheet:

http://uk.rs-online.com/web/p/bench-power-supplies/6736749/

kt10aan, 2016-02-18:
Thanks,we have power again :)

Just for completeness, and since I cannot find the relevant page on the wiki, could you write all the settings (Undervoltage, Voltage, Overvoltage, Current etc) for both power supplies?

maggia80, 2016-02-18:
Here you can  find the info:

http://wiki.icub.org/wiki/Power_supply

closed 2016-02-18 by maggia80