robotology/mesh-support #189: Accelerometer from upper right leg sending invalid data to YARP port

nunoguedelha, 2016-02-22:
##### Description of the failure
The accelerometer 11B6 from the MTB board on the upper right leg is sending invalid data (`-1`). All the other accelerometers from the same board are sending valid data.

##### Detailed conditions when the failure happened
We were collecting accelerometers data from the right leg while swinging the leg very slowly (right leg pitch joint).

Connected ports: `icub/right_leg/inertialMTB`

Find below a chunk of collected data (one timestamp), with the invalid raw measurements (`-1.0 -1.0 -1.0`). `43` is the identifier for the sensor 11B6 :
```matlab
21068 1452882025.493055 12.0 6.0 38.0 1.0 2730051042.0 15808.0 -2416.0 3904.0 39.0 1.0 2730050044.0 16128.0 816.0 4160.0 40.0 1.0 2730051049.0 16384.0 528.0 3744.0 41.0 1.0 2730050060.0 16064.0 3392.0 960.0 42.0 1.0 2730050067.0 16112.0 400.0 3632.0 43.0 1.0 2730051056.0 '-1.0 -1.0 -1.0' 44.0 1.0 2730050053.0 -16080.0 2080.0 -3248.0 34.0 1.0 2730051064.0 16096.0 560.0 784.0 35.0 1.0 2730050075.0 16480.0 -16.0 -432.0 36.0 1.0 2730050082.0 16528.0 352.0 1632.0 37.0 1.0 2730050089.0 16256.0 -288.0 -1328.0 0.0 0.0 0.0 0.0 0.0 0.0
```

The format of the data can be found here: [codyco-modules wiki](https://github.com/robotology/codyco-modules/wiki/External-Inertial-Sensors-for-iCubGenova02).

nunoguedelha, 2016-02-22:
As I've already mentioned to @julijenv and @marcoaccame, I've observed another issue, somewhat similar, in my last data capture (on Friday 22/02). The accelerometers 11B4 to 11B7 where publishing constant data (not invalid = `-1`). We were measuring the gravity and the measurements made sense, but they were constant, and shouldn't, since the leg orientation was changing. It's like the sensors stopped updating their measurements and publishing the same old data.

I never reproduced the issue since.

julijenv, 2016-02-23:
Hi @nunoguedelha ,
I ve just finshed checking the 4 boards yuo told me not working BUT unfortunately either the skin part of the obard or the accelerometer (inertial sensor) is sending proper values, could you try to check if the config files are correctly configured.
let me know

nunoguedelha, 2016-02-23:
I had already checked the config `.xml` files and they are correct. Well, when I observed the issue on the sensors 11B4, 11B5, 11B6, 11B7, they were publishing measures close to gravity ground truth values so the configuration file is definitely ok. the issue should come from the sensor or the board.

traversaro, 2019-10-03:
If I remember correctly, eventually you were able to get correct data from the accelerometers in the iCubGenova02 robot, right @nunoguedelha ?

nunoguedelha, 2019-10-04:
Honestly I don't remember. This is so old that we should just do a full check of the accelerometers and gyros. So I suggest we close this issue in the meantime.

closed 2019-10-21 by nunoguedelha