robotology/mesh-support #191: Left elbow broken tendon

usigneb, 2016-02-26:
Hi all,
 I'm having some problems with our iCub.
We have broken one of the left arm elbow tendons. The video and photoes of 
the broken part are attached to my mail.

I have some extra tendons here with different lengths 
and end types (but with numbers of them). 
But I'm not sure how to replace the new tendon and which tendon to use.

Any help would be really nice.

Thanks in advance.

Bengisu

[image]
[image]
[image]

julijenv, 2016-02-26:
re: Hi Elis,
From the pictures I guess you broke with not certainty N9 and/or 10 - hard to see if just one is broken or both.
please see this manual "shoulder wiring" on the wiki page.

http://wiki.icub.org/wiki/Tendons_and_replacement

On which you will be able to see either the size of the cable and how to wire it.
I would advise you, to remove the upper arm cover in order to proceed  with more safety the change of the cable.

In the worst case scenario please come on skype  add icub.support on your contact list and i would be here surely to assist you.

As explained on mail last reply to your previous mail please copy paste your failure on a new issue on the github/icub-support.
https://github.com/robotology/icub-support/issues

good luck let us know.

pattacini, 2016-02-26:
@robotology/all-robotology-developers 

Lately, we've introduced a [**template**](https://github.com/robotology/icub-support/blob/master/.github/ISSUE_TEMPLATE.md) for opening up an issue here on `icub-support`.

It's composed of two sections the user is asked to fill in. Let's try to stick to this, then.

Of course @robotology/icub-support-team is free to adapt the template to their needs.

pattacini, 2016-02-26:
The template shows up as soon as the user click on the [[**`New Issue`**]](https://github.com/robotology/icub-support/issues/new) button.

usigneb, 2016-02-26:
Sorry. I'm new and I didn't notice. Next time I'll be careful about that.

pattacini, 2016-02-26:
No worries @usigneb :smile:

julijenv, 2016-03-16:
Any news with the replacement?

julijenv, 2016-03-31:
@usigneb , any news with the replacement?

closed 2016-07-05 by julijenv