robotology/mesh-support #192: Very big gap between hip pitch joint encoder and motor encoder

hu-yue, 2016-02-28:
##### Description of the failure
I don't think this is a failure, anyway in a recent experiment I've seen that the motor encoders and encoders of the hip pitch joint have very big gap, the error between motor encoder readings and encoder readings is up to 40 deg, in both right and left legs.
Here are pictures of the right leg hip pitch:
[image]
And left left pitch:
[image]

Motor encoder readings in the plots are obtained as (m_enc - offset)/gear_ration.

##### Detailed conditions when the failure occurred
The robot was in position direct control mode and commands were sent every 10 ms.
The desired trajectory coincides with the encoders side.
The gap seems to increase with the desired joint angle?

I just would like to know if this is normal, and if yes, why is it like this?

randaz81, 2016-03-02:
Hi, a very

randaz81, 2016-03-02:
Hi, that joint has an internal coupling = 1.5 due to to the ratio of the tendon pulleys.
Test `motorEncoders-consistency` takes that in account (and should be successful if executed on your robot)
So, yes, it's normal and nothing has to be changed on your robot.
In the future we may revise gearbox section adding an additional parameter to store this additional conversion factor.
About your plots: multiplying the blue data by 1.5 the two plots should match.

hu-yue, 2016-03-02:
@randaz81 thanks! =)
Yes now they match, I didn't check the test, I just remember that I had to divide by the gear ratio and take care of the offset.
Now it makes much more sense.

closed 2016-03-02 by hu-yue