robotology/mesh-support #193: Thumb opposition tendon replacement

AGiagkos, 2016-03-01:
Dear list,

Our iCub has a broken tendon thumb opposition tendon which we are trying to replace. We use the uncoated Carl Stahl CG077063 cable but we are unable to pass it through the inner hole of the pulley (hole tagged as "H" in the document). We tried to glue its end so that strands do not become curly as we pass it through. We also tried to shoulder it and fine it a bit to get it thinner. Even with thinner strings we need to be very precise in order to find the correct angle to pass them through.

Yet, even if it comes out from the other side (after putting a lot of effort), strands break or twist making it impossible to pass through the pulley..

Is there any hint or particular technique we could use?

Many thanks in advance,

Alexandros

julijenv, 2016-03-02:
Hi @AGiagkos ,
FIrst thing first, which robot are talking about? Please check the [template](https://github.com/robotology/icub-support/blob/master/.github/ISSUE_TEMPLATE.md) when opening a new issue, thx in advance - need it later to be able to track more efficiently issue by robot name and also for polling.
So to answer to your question, I would suggest you 2 ways to proceed:
1) Try to fit any metallic needle smaller than the hole size inside the hole in order to align the hole inside the shaft and the pulley (your shaft may have probably rotated)by using it as leverage.
2) if you succeed at loeast once to pull out the wire inside the hole (even if gets curly) try to use the cable as a file and do some in and out movement (20 to 30 times) in order to get the 2 pieces to reposition or to make bigger the inner hole.

Let me know if you succeed to make it passthrough the hole and not having anymore the curly cable when it gets out.

AGiagkos, 2016-03-07:
Hello julijenv, 

Thanks for your reply and help. We managed to fix the thumb opposition following your advice; the shaft was indeed slightly rotated so we used a needle to put it in place. We then managed to pass the cable (shouldered end to avoid curly strands) quite easily. 

That was the Aberystwyth iCub - thumbs up again!

Regards, 

Alexandros

julijenv, 2016-03-07:
Nice to hear that.
Can we close this issue?

AGiagkos, 2016-03-09:
Yes please - thanks for your help again.

Alexandros

closed 2016-03-10 by julijenv