robotology/mesh-support #194: Starting Robotinterface shows error icub Viola

traversaro, 2016-03-02:
##### Description of the failure
90029		22701,767589		ERROR		cfw2can [0] thread:6 msg:65 joint:1 timed out

##### Detailed conditions when the failure occurred
When starting yarpmotorgui it was found some hardware fault in the arms.
After proceeding to restarting robotinterface and the motors, the logger showed the previously mentioned error. We would see the head move a little and the eyes too, afterwards it would stop moving. No movement of the arms and hands was observed.

traversaro, 2016-03-02:
it was  @fjandrad who posted the issue, sorry Silvio I didn't notice you had your git account open on the icub23.

julijenv, 2016-03-16:
didnt hear anything since then - should I close it?

traversaro, 2016-03-16:
The day after we were not able to reproduce this issue, closing.

closed 2016-03-16 by traversaro

randaz81, 2016-03-16:
Just as a memo: this error shows up when a board gets disconnected from the pc104, either because it is unable to respond on the CAN bus (wires damaged) or the board power supply is down.