robotology/mesh-support #195: Calibration of thumb opposition encoder

gsaponaro, 2016-03-03:
Please fill in the following two sections:

##### Description of the failure

While doing some manipulation experiments, we broke the metal piece that has the magnet for the right thumb opposition (j8).

##### Detailed conditions when the failure occurred

We are in the process of replacing the metal piece using a spare that you had sent us last year (https://github.com/robotology/icub-support/issues/23). So far so good:

[image]

Our question is: how to re-calibrate the encoder associated to the joint r_thumb_oppose? We are thinking of doing it qualitatively like for the arms http://wiki.icub.org/wiki/ArmFineCalibration, checking ```yarpmotorgui``` as we move the motor, then replacing the value of the parameter ```Zeros``` in the ```GENERAL``` section of [iCubLisboa01/hardware/motorControl/icub_right_hand.xml](https://github.com/robotology/icub-main/blob/master/app/robots/iCubLisboa01/hardware/motorControl/icub_right_hand.xml), first column. Is this the correct procedure and file? Anything else we should be careful about? Thanks.

/cc @lorejam

julijenv, 2016-03-03:
Hi @gsaponaro and @lorejam ,
### PLEASE READ EVERYTHING ONCE BEFORE STARTING
From a totally uncalibrated joint, the fine calibration procedure has not been intended to be used for this purpose. Please follow the following lines with carefulness and don't hesitate to ask if ever you have any doubts.
so to proceed with the calibration of the opposition thumb,
### PLEASE DO THAT ON A WINDOWS MACHINE - macros are not working properly on linux
I take as a prerequisite that you didn't change anything in your main repository (the official path from the icub-main repository that you surely downloaded from github is ..../icub-main/app/robots/iCubLisboa01/.
I would suggest you to backup the repository just as matter of safety.
from that folder you may find a excel file, named : iCub_Calibration_V1_3_1_iCubLisboa01.xls
If you open it  from a windows machine (on linux the macros dont work properly)
[image]
you will have to fill up the right_hand table for the joint 0 - top of the table in the picture above.
How to get the numbers you need to enter inside the table?
You will have to open the file general.xml, for this time from the repository iCubLisboa01 from either .local if you use it or .../icub-main/build-pc104/share/icub/robots/icubLisboa01/ (hint to check where your files are taking from launch a quick robotInterface instance (red button pressed) and ctrl+C directly after the launch.
read the first lines that should indicate you you the path where it read general.xml and all the other files basically.
Edit this file (general.xml)
and select the option set to TRUE for either the skipCalibration and useRawEncoderData.

**RED BUTTON PRESS**

1. launch robotInterface from the computer link to the robot (server)
2. press about 20 times the enter button to skip all the message regarding the fact that the robot is launched with the rawEconderValue TRUE and that could be a problem if the RED butoon is NOT pressed - thats why you have to make sure that the RED BUTTON SHOULD BE PRESSED.
-> when on the screen the robotInterface display everything run ok
3. launch yarpmotorgui
4. get the value of the encoder raw for the thumb opposition in the open palm position
5. put the value inside the Vmin fo the excel file
6. move the tumb to postion it at 90 degree (perpendicular to the palm)
put the value in Vmax insider the table 

## **WARNING**
1. CHANGE the path to point the folders calibrators/ and Hardware/MotorControl/
2. PRESS the button macro "Generate All"

**all the files inside the 2 repositories get changed and hopefully just the ones regarding the calibration of the right_hand are different from before the changes made to the excel files.
make some diff with the new files and the backup files you made a the beginning.**
3. If you can confirm that just the files of the right_hand either the calibrator or the hardware/motorControl/ repository have been modified just for the joint 0 you can proceed

### close the robot and modify general.xml to leave it as u found it B4 doing anything

4. do commit - **only the files label related to the right_hand should be detected for commit**
5. **NOW** update your icub-main repository from your server (linux machine)
6. ccmake .. the build-pc104 repository
7. compile and generate
8. do MAKE
## RED BUTTON PRESS
9. launch robotInterface from /icub-main/build-pc104/share/icub/robots/icubLisboa01/ (the new files has been copied when u did MAKE)
when ready
10. launch yarpmotorgui check the thumb opposition joint and red if it goes from 10 degre (open hand) to 90 degree (thumb perpendicular to the palm)

Let me know

gsaponaro, 2016-03-03:
Thanks a lot, a couple of questions:

* In the open palm position, the raw encoder reading we observe is 2500 (according to the instructions this is the Vmin), while with thumb perpendicular to the palm it is 1900 (Vmax). Question: is it normal that in this case Vmin>Vmax? Or should we swap them?

* In Excel on the Windows machine, after setting the values Vmin and Vmax and confirming that the paths were correct (both hardware/motorControl and calibrators paths were already correct), we click on Generate All. The xml files in the calibrators folder were not overwritten (nothing changed, is it normal?), while the xml files in the hardware/motorControl changed (not only icub_right_hand.xml that we were expecting, but also all of the other files, is it normal? Do we just ignore the other files changed?)

gsaponaro, 2016-03-04:
Regarding my first question above, I guess it is plausible that "Vmin" has a higher value than "Vmax", that depends on the position of the magnet.

gsaponaro, 2016-03-04:
Regarding the fact that, when clicking Generate All, not only the right_hand xmls but also other robot parts xml get changed: some of those changes can be ignored (trivial/aesthetic stuff, https://github.com/robotology/icub-main/issues/305), but other ones like this are more puzzling:

[image]

Not sure why we obtained new values for left_arm, when we were just inputting right_hand Vmin/Vmax  in Excel. I presume we can ignore this, and only focus on the new right_hand parameters.

gsaponaro, 2016-03-04:
Something strange happening:
1. the raw encoder reading changes sign (but keeps the same absolute value), i.e., a few times we read Vmin=2500 Vmax=1900, a few other times Vmin=**-**2500 Vmax=**-**1900. We are always power-cycling the motors (waiting several seconds) between every experiment;
2. it seems that this change of behaviour happens when we toggle the skipCalibration & useRawEncoderData values in general.xml.

randaz81, 2016-03-07:
AFAIK, the sign is assigned only by the `Encoders` parameter, which is computed by the excel file, using the `Vmin`/`Vmax` values. So if you are swapping Vmax/Vmin (that should be NOT done), you are (maybe) changing the sign of the Encoders `parameter` , and the next time you start the robot you'll have a reading with a different sign. Setting `useRawEncoderData` will just set `Encoders`=+1/-1, preserving the original sign of the parameter.

gsaponaro, 2016-03-07:
Hi @julijenv @randaz81,
we re-did the calibration procedure carefully from the beginning. As instructed, we took the values for Vmin (thumb open like the rest of the palm and fingers) and Vmax (thumb perpendicular to palm) as they are: we did not swap them. Conclusion so far: the calibration of the thumb now looks like this:
https://youtu.be/9_uJdoYXaLY

As you can see, the thumb is making too much force against the rest of the hand. When it finishes the calibration, we see the following slider in yarpmotorgui and we cannot move this joint in either direction (however, if we set it to IDLE and we manually move the finger, we read correct calibrated encoder values from 0 to 90 degrees).

[image]

For completeness, below we show the full way that we re-did the calibration.

1. acquiring Vmin (when thumb is open like the rest of the palm) in calibration mode:
[image]
2. acquiring Vmax (when thumb is perpendicular to palm) in calibration mode:
[image]
3. putting the values 2500 and 1900 in the Excel file, on a Windows 7 machine with macros enabled:
[image]
4. resulting "Encoder" and "Zeros" values in ```hardware/motorControl/icub_right_hand.xml``` (first column) which we then put in the .local xml file:
[image]
5. resulting "calibration1" value in ```calibrators/right_hand_calib.xml``` (first column) which we then put in the .local xml file:
[image]

Any further suggestion? Thanks.

julijenv, 2016-03-07:
Hi @gsaponaro ,
Ok if yo take a look at the formula for the cells zeros and encoders for this joint you will understand that the sign is important so I guess that if you want to be precise (not having to change the PID values), 2 ways possible:
1. swap Vmin and Vmax values
2. it would be nicer, try to spin the magnet in order to have higher values when the thumb is perpendicular and lower value when the hand is opened.

@randaz81 tell me if not true

gsaponaro, 2016-03-07:
Basically we have two problems:
1. the encoder raw values of JOINT 8 randomly change sign **when useRawEncoderData=true**, both Vmin and Vmax... they are sometimes 2500 and 1900, and some other times -2500 and -1900. Any idea of what could be the cause of this?
2. During the robot calibration at startup, the thumb opposition (JOINT 8) moves to 0, while it should be supposed to move to 10 (according to calibrators/right_hand_calib.xml). We are assuming that JOINT 8 should move to the "calibration2" value, which is currently set to 10. Anything we are doing any mistake here?

gsaponaro, 2016-03-09:
Update:
1. We still observe the phenomenon that, **when using useRawEncoderData=true**, sometimes certain joints give positive values, sometimes the same joints give negative values. Verified many times after power-cycles. Example screenshot:
[image]
2. At the moment, the Excel calibration macro that computes right hand joint8 "calibration1" at cell Z82 does this: `abs("Min", "Encoders")+"Vmin"`. That formula is valid in the case where Vmin<Vmax, but in our situation with the new magnet (which is glued inside and cannot be spinned), poles are inverted with respect to the "common situation", so we have Vmax<Vmin. The correct formula for our case is: `abs("Min", "Encoders")-"Vmax"`. Ideally, the cell should have a macro that accounts for both cases by using Excel's IF or MIN functions. @randaz81 @maggia80 shall we open a separate issue for this?
3. With reference to problem 1 (apparent encoder raw sign change in ```yarpmotorgui```), for now we are assuming that raw values are to be interpreted as positive (unsigned). We are replacing a broken steel cable and then we'll try again with the new "Encoder", "Zeros" and "calibration1" (<- determined with the correct formula) values.

gsaponaro, 2016-03-09:
Good news: with the steel cable fixed and a new calibration procedure, we are able to control the finger again decently in both directions. In addition to "Encoder", "Zeros" and "calibration1", we also had to invert the signs of "kp", "kd", "ki" and to adjust the value of "kp" to correct overshooting. Preparing a Pull Request with the new values.

randaz81, 2016-03-10:
Can you please tell me more about the story of the magnet _[...] which is glued inside and cannot be spinned [...]_? Did you flipped the magnet? Did you replaced it gluing in a different position? Is it the original magnet and you found that it was glued differently respect to the others?

gsaponaro, 2016-03-10:
[Around Sept. 2015](https://github.com/robotology/icub-support/issues/23), one of the screws that holds the steel cable had broken, so **you sent us a replacement, including the new magnet, already mounted and glued** (see also parts in green in this picture):
[image]
We didn't replace this back then, because Ricardo made a temporary solution (a node, to bypass the screw). Now, a few days ago, because the cable broke again, we decided to replace the part, with the spare that you gave us. And we finally noticed this issue that the magnet was rotated (with respect to the previous one).

randaz81, 2016-03-10:
@julijenv can they rotate the magnet in order to make it consistent with the standard configuration?

gsaponaro, 2016-03-18:
Hi @julijenv, whenever you can (perhaps next Monday?) Ricardo would like to talk to you about the possibility (or not) to rotate the magnet. Thanks.

gsaponaro, 2016-04-08:
Hi @julijenv (or anybody else from @robotology/icub-support-team), is it possible to schedule this Skype meeting we've been talking about? To determine whether we can rotate that magnet or not, and the actions to take as a consequence. Thanks in advance.

gsaponaro, 2016-05-09:
Hi @julijenv,

Following your suggestions on Skype, we have loosened the cables, removed the side screws, and then attempted to rotate this magnet in order to reach the standard configuration (standard for the PID signs, /cc @randaz81). So far we were unsuccessful. Please see this video of our attempt, which shows that the magnet is glued/fixed to the metal piece:

https://youtu.be/lkvp4TdtEis

julijenv, 2016-05-09:
Hi @gsaponaro ,
as I could see the magnet is broken now, so try to remove and clean the place where it was.
try to fetch the old magnet from the old finger, if you cant just let me know, we will send you a new one.
please on one mail address to me give me all the information needed to send you this magnet
(person in charge to retrieve the enveloppe)
full address
phone number of the responsible to get the magnet
let me know

gsaponaro, 2016-05-12:
Hi @julijenv @randaz81, progress on this issue:

- [x] new magnet received and installed inside our old thumb opposition metal piece as instructed
- [x] re-did Excel calibration of the thumb opposition joint, now having standard magnet spin - obtained new calibration1, Encoder, Zeros values which we put in our .local xmls (not committed yet). We didn't touch the PID terms, i.e., they are the ones from the repository file ```iCubLisboa01/hardware/motorControl/icub_right_hand.xml```
- [x] encoder reading when joint is idle and manually moving the finger: OK
- [ ] however, we observe **oscillations** during control: see [video1 (yarprobotinterface startup)](https://youtu.be/D5k32hSKbxk) and [video2 (controlling from yarpmotorgui)](https://youtu.be/B-bGBg-Tals). We verified that the cables are fine. --> It seems necessary to adjust PID control terms. For now we tried lowering the absolute value of the "kp" term, in that case the situation improves in the closing direction (0 -> 90 deg), but problem persists in the opening direction (90 -> 0 deg). Suggestions?

/cc @lorejam

julijenv, 2016-05-16:
Hi guys from Lisbona ( @lorejam and @gsaponaro ),
In my point of view for sure the PID could be twick a bit in order to do not overshoot (are you using the same as the left thumb), but in reality it looks quite as good (or bad - only you could judge according to your demos) as the other robot with arm V1.
Let me know

gsaponaro, 2016-05-16:
About this issue:

The PID gains that we are using for this right thumb opposition joint are: kp=-8000, kd=-32000 ki=-5 as in ```iCubLisboa01/hardware/motorControl/icub_right_hand.xml``` (we confirm that the file on the repository is the same that we have in .local).

Agreed, we should tweak these values to reduce the overshooting. Any empirical suggestion for this would be greatly appreciated.

gsaponaro, 2016-12-22:
Closing this issue for now.

We have to replace the steel cable of this joint anyway, because it recently broke (j8 r_thumb_oppose).

When we get to replace the cable, we'll check again the behaviour of this joint and open a new issue if there is still overshoot.

/cc @vicentepedro

closed 2016-12-22 by gsaponaro