robotology/mesh-support #198: error with encoders on board #6

mbrunettini, 2016-03-04:
##### Description of the failure
the following error appears on robotInterface: 
`[ERROR] from BOARD 10.0.1.6 (leftLeg), src LOCAL, adr 0, time 859s 270m 108u: (code 0x0500000d, par16 0x0000 par64 0x0000000000000055) -> CFG: EOtheMotionController cannot be configured. verification of encoders fails. see other messages for more details + .
[ERROR] from BOARD 10.0.1.6 (leftLeg), src LOCAL, adr 0, time 859s 270m 199u: (code 0x05000014, par16 0x0404 par64 0x0f00000006060606) -> CFG: EOtheEncodeerReader cannot be configured. verification fails. in p16 the number of joints (lsb) and encoders (msb), in p64 the error status of verification reading + .
[ERROR]embObjAnalogSensor::getFullscaleValues(): ETH Analog sensor: request for calibration parameters timed out for board 6
[ERROR]embObjAnalogSensor::sendConfig2Strain() has failed in calling  embObjAnalogSensor::getFullscaleValues()
`

##### Detailed conditions when the failure occurred
Starting robotInterface

**NOTE** : there is currently mounted the new live V6 on the PC104, please contact me

julijenv, 2016-03-04:
@marcoaccame ,
mi potresti dire di quale giunto o sensor si tratta?
grzie in anticipo

julijenv, 2016-03-04:
it may have been fixed, check it out and let me know on moanday morning the sensor LLNEI0 where one of the wire was found broken, i resoldered it.

traversaro, 2016-03-07:
There are still intermittent failures of the encoder. Everything works fine for several minutes, and then the `robotInterface` reports this error: 
~~~
1035		52,387305		ERROR		 from BOARD 10.0.1.6 (leftLeg), src LOCAL, adr 0, time 942s 670m 526u: (code 0x02000008, par16 0x0000 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = ID of joint. + .
1036		52,387357		ERROR		 from BOARD 10.0.1.6 (leftLeg), src LOCAL, adr 0, time 942s 670m 620u: (code 0x02000008, par16 0x0001 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = ID of joint. + .
1037		52,387373		ERROR		 from BOARD 10.0.1.6 (leftLeg), src LOCAL, adr 0, time 942s 670m 711u: (code 0x02000008, par16 0x0002 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = ID of joint. + .
1038		52,387386		ERROR		 from BOARD 10.0.1.6 (leftLeg), src LOCAL, adr 0, time 942s 670m 799u: (code 0x02000008, par16 0x0003 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = ID of joint. + .
1045		54,959605		ERROR		 from BOARD 10.0.1.6 (leftLeg), src LOCAL, adr 0, time 945s 242m 534u: (code 0x02000008, par16 0x0000 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = ID of joint. + .
1046		54,959643		ERROR		 from BOARD 10.0.1.6 (leftLeg), src LOCAL, adr 0, time 945s 242m 629u: (code 0x02000008, par16 0x0001 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = ID of joint. + .
1047		54,959656		ERROR		 from BOARD 10.0.1.6 (leftLeg), src LOCAL, adr 0, time 945s 242m 720u: (code 0x02000008, par16 0x0002 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = ID of joint. + .
1048		54,959668		ERROR		 from BOARD 10.0.1.6 (leftLeg), src LOCAL, adr 0, time 945s 242m 810u: (code 0x02000008, par16 0x0003 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = ID of joint. + .
1057		59,953931		ERROR		 from BOARD 10.0.1.6 (leftLeg), src LOCAL, adr 0, time 950s 242m 534u: (code 0x02000008, par16 0x0000 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = ID of joint. + .
1058		59,953974		ERROR		 from BOARD 10.0.1.6 (leftLeg), src LOCAL, adr 0, time 950s 242m 628u: (code 0x02000008, par16 0x0001 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = ID of joint. + .
1059		59,953987		ERROR		 from BOARD 10.0.1.6 (leftLeg), src LOCAL, adr 0, time 950s 242m 719u: (code 0x02000008, par16 0x0002 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = ID of joint. + .
1060		59,964001		ERROR		 from BOARD 10.0.1.6 (leftLeg), src LOCAL, adr 0, time 950s 242m 809u: (code 0x02000008, par16 0x0003 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = ID of joint. + .
1068		64,958622		ERROR		 from BOARD 10.0.1.6 (leftLeg), src LOCAL, adr 0, time 955s 242m 537u: (code 0x02000008, par16 0x0003 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = ID of joint. + .
2910		406,793809		ERROR		 from BOARD 10.0.1.6 (leftLeg), src LOCAL, adr 0, time 1297s 76m 527u: (code 0x02000008, par16 0x0001 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = ID of joint. + .
2911		406,793864		ERROR		 from BOARD 10.0.1.6 (leftLeg), src LOCAL, adr 0, time 1297s 76m 622u: (code 0x02000008, par16 0x0002 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = ID of joint. + .
2912		406,793881		ERROR		 from BOARD 10.0.1.6 (leftLeg), src LOCAL, adr 0, time 1297s 76m 712u: (code 0x02000008, par16 0x0003 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = ID of joint. + .
2914		406,803959		ERROR		 from BOARD 10.0.1.6 (leftLeg), src LOCAL, adr 0, time 1297s 82m 527u: (code 0x02000008, par16 0x0000 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = ID of joint. + .
[... cut ...]
3110		489,957885		ERROR		 from BOARD 10.0.1.6 (leftLeg), src LOCAL, adr 0, time 1380s 242m 720u: (code 0x02000008, par16 0x0002 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = ID of joint. + .
3111		489,967919		ERROR		 from BOARD 10.0.1.6 (leftLeg), src LOCAL, adr 0, time 1380s 242m 811u: (code 0x02000008, par16 0x0003 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = ID of joint. + .
~~~
And the first four joints of the left leg go in hardware fault.

traversaro, 2016-03-07:
cc @mbrunettini @arrenglover @nunoguedelha

traversaro, 2016-03-07:
cc @jeljaik

traversaro, 2016-03-10:
@jeljaik the issue was solved, right?

jeljaik, 2016-03-10:
As reported by Julien,  yes

closed 2016-03-10 by traversaro