robotology/mesh-support #201: Problems with encoders (Always left leg)

jeljaik, 2016-03-12:
##### Description of the failure
Again yesterday evening `robotInterface` was reporting issues with a magnetic encoder in the left leg. Unfortunately I have forgotten to copy the error message. I will update on Monday with the details. Just in case someone will use it this weekend. 

##### Detailed conditions when the failure occurred
During torque-controlled walking tests, as we faulted the robot, making the left knee bending more than necessary. Although it is worth mentioning, that we're not making it collapse in an abrupt way, and that the knee is not even fully bending backwards just enough before it hangs from the crane.

francesco-romano, 2016-03-14:
I tested this morning (just turning on `robotInterface`) and no errors appeared.
I'm quite scared about this issue, considering the two demos we have to do.

@julijenv

jeljaik, 2016-03-14:
Yeah, I just did the same :open_mouth: and also neither I saw any error this time. Closing for now ... but I may be back :sunglasses:

jeljaik, 2016-03-14:
And I'm back. So, before I turned on the `robotInterface` and after a few minutes, the same error of Friday showed up, i.e.:

```bash
[ERROR] from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 511s 736m 541u: (code 0x02000008, par16 0x0000 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = ID of joint. + .
```

The robot was not even moving, just hanging from the crane.

jeljaik, 2016-03-14:
After about one hour just sitting next to the robot with just the `robotInterface` and waiting to see if the error reappeared as before, it crashed with the following error:

```bash
[ERROR]embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD EB6 IP 10.0.1.6 for nv ID32 = 0x0100020b -> IND = 2, TAG = eoprot_tag_mc_joint_status_target 
[ERROR]embObjMotionControl::getRefTorqueRaw() could not read reference pos for  BOARD EB6 IP 10.0.1.6 joint  2 
[ERROR]embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD EB6 IP 10.0.1.6 for nv ID32 = 0x0100020b -> IND = 2, TAG = eoprot_tag_mc_joint_status_target 
[ERROR]embObjMotionControl::getRefVelocityRaw() could not read reference vel for  BOARD EB6 IP 10.0.1.6 joint  2 
[ERROR]Received an answer message nobody is waiting for (eoThreadFifo::pop) 
[ERROR]Received an answer message nobody is waiting for (MCmutex_post) 
robotInterface: symbol lookup error: /usr/local/src/robot/icub-main/build-pc104/lib/iCub/embObjMotionControl.so: undefined symbol: embObjPrintWarning
```

Not sure if related to what occurred before, but it's always related to board EB6. Before that, the `robotInterface` displayed no error message, just the usual warnings and debug info.

jeljaik, 2016-03-14:
@marcoaccame

marcoaccame, 2016-03-16:
From the collected log file here attached, diagnostics from board EB6 show two different problems which have happened in three consecutive launches of robotInterface.

[failure-14mar16.txt](https://github.com/robotology/icub-support/files/176011/failure-14mar16.txt)

1. the motor of joint number 2 has been switched off to avoid damaging it because there was an overcurrent.
2. all its 4 joints have been forced in idle mode because all the AEA encoder do not read correctly.

Problem number 1 could be explained because the robot was performing an experiment (as @jeljaik told me) which could have put too much effort on its left leg and the motor effectively was absorbing too much current. 

In the robot, the excessive current consumption is detected by an HW circuit inside the 2FOC board which for protection of the motor switches it immediately off (please refer to @ale-git for more details).

Problem number 2 instead has been unclear for some two days, but it is now explained.

The AEA encoders are not working because their supply voltage sometimes drops from 5 V down to 800 mV. The reason is a defective STRAIN can board which seems to drain too much current.

See attached picture with the measured power supply.

[image]

By removing the STRAIN board or by using another one there are not voltage drops.

Needless to say, ... the problem was not systematic but happened now and then.

@julijenv is now working to replace the STRAIN.

marcoaccame, 2016-03-16:
About the error logged by @jeljaik with following log excerpt

```
[ERROR]embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD EB6 IP 10.0.1.6 for nv ID32 = 0x0100020b -> IND = 2, TAG = eoprot_tag_mc_joint_status_target 
[ERROR]embObjMotionControl::getRefTorqueRaw() could not read reference pos for  BOARD EB6 IP 10.0.1.6 joint  2 
[ERROR]embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD EB6 IP 10.0.1.6 for nv ID32 = 0x0100020b -> IND = 2, TAG = eoprot_tag_mc_joint_status_target 
[ERROR]embObjMotionControl::getRefVelocityRaw() could not read reference vel for  BOARD EB6 IP 10.0.1.6 joint  2 
[ERROR]Received an answer message nobody is waiting for (eoThreadFifo::pop) 
[ERROR]Received an answer message nobody is waiting for (MCmutex_post) 
robotInterface: symbol lookup error: /usr/local/src/robot/icub-main/build-pc104/lib/iCub/embObjMotionControl.so: undefined symbol: embObjPrintWarning
```

the problem is  solved by commit robotology/icub-main@9ed2de1f60430036e251f4fe5d6ae22659e935d6.

jeljaik, 2016-03-16:
Awesome job @marcoaccame and @julijenv. Thanks so much! Let us know when it's all been replaced.

marcoaccame, 2016-03-16:
@jeljaik .... wait till tomorrow before to say it is over.

Today we have changed the STRAIN and tested the system. There was the same problem. We suspect that the EMS was damaged as well. Thus, we have just mounted a new EMS and a new STRAIN.

The system is now running fine but still without cover. We want to have it running for some time (1 hour) before tomorrow morning we mount the cover back and test it again for some time.

julijenv, 2016-03-22:
It has been replaced both the EMS board and the FT sensor. evrything was like it workd fine before leaving for CHICAGO.

closed 2016-03-22 by julijenv