robotology/mesh-support #202: Neck tendon broken

Tobias-Fischer, 2016-03-14:
##### Description of the failure
Hi all,
We were just running some experiments and something strange happened. The head of the robot all of the sudden moved very aggressively left and right in an oscillating manner. This caused one of the neck tendons to break (we need the cable urgently - please see shipping address below).

[image]

##### Detailed conditions when the failure occurred
Only pretty standard modules were running:
`robotInterface`, `ARE`, `SFM`, `iKinGazeController`, `iKinCartesianSolver`, `gravityCompensator`, cameras, `IOL`, `wholeBodyDynamics`
The only WYSIWYD related module running was `babbling`. We will check with @pattacini today that this code is safe, which we think it is.

The weird thing is that this happened before on several other robots (`iCubLondon01`, `iCubBarcelona01`) in situations where you really don't expect anything going wrong.
A video which we happened to capture can in one of the situations be found here (by the way, `babbling` wasn't running back then): https://slack-files.com/T053A59T4-F0B1WACGL-e64d5fbece
However, when it happened today it was MUCH more aggressively.

Unfortunately we don't have log files. However, when it happened, the head of the iCub was facing left; however in `iCubGUI` we could see that the encoders detected that the iCub encoders were returning values which suggest that the iCub is facing right. This would suggest the oscillations.

### Shipment of the cable
As we have an integration this week, it would be great if you could ship a tendon or two as soon as possible using express service.
The address is:
Gregoire Pointeau
Inserm U846
Stem Cell and Brain Research Institute
Human and Robot Cognitive Systems
18, avenue du doyen Jean Lépine
69675 Bron cedex
France

##### iKart current supply
@jypuigbo suggests this might be due to too high current / voltage supplied to the iKart. Apparently on `iCubBarcelona01` they changed the voltage to ~39V instead of the 52.5V (see also https://github.com/robotology/icub-support/issues/30 and https://github.com/robotology/icub-main/issues/205).

/cc @robotology/icub-support-team @julijenv @pattacini @gregoire-pointeau @pattacini

randaz81, 2016-03-14:
I just pushed an update for your configuration files increasing the shift factor

Tobias-Fischer, 2016-03-14:
Thanks for sending the tendon so quickly @robotology/icub-support-team!!! That will be helpful for the rest of the week.
Thanks to @randaz81 changes we hopefully won't have this problem again.

Tobias-Fischer, 2016-03-14:
We should see whether @randaz81 changes need to be applied to other robots (https://github.com/robotology/icub-main/commit/3d1e4d1852e2b1c4f993b806eb163c0d5abd9286) - and whether changes from other robots need to be applied on iCubLyon01 (`kp`, `kd` and `ki`).

randaz81, 2016-03-14:
Yes, those changes are safe to apply and recommended to all iCubs equipped with a V2 head.

Tobias-Fischer, 2016-03-14:
Okay thanks! I created a pull request: https://github.com/robotology/icub-main/pull/309

Should we also lower the `kp`, `kd` and `ki` on `iCubLyon01`?

randaz81, 2016-03-14:
Regarding the second point I'd not lower the PID gains unless you notice visibile vibrations on all the robot, especially on the shoulder joints (for the head, previous commit should be sufficient). However, let me think tomorrow if I can add an configuration option to automatically adjust the gains depending on the supply voltage.

Tobias-Fischer, 2016-03-14:
Okay great, thank you :)

randaz81, 2016-03-14:
PS: On CAN robots, boards cannot measure the supply voltage (this feature is available instead on ETH boards). However, I'm thinking to let the user set it with a global configuration parameter e.g. <param name="powerSupplyVoltage">  40 </param>. Changing this parameter should do the job in a much easier way than modifying the gains all across the robot parts. And it should be easy to implement!

Tobias-Fischer, 2016-03-16:
We were able to put the cable pack in place. Thanks a lot for the fast shipment of the cable @julijenv!

Best, Tobi

closed 2016-03-16 by Tobias-Fischer