robotology/mesh-support #203: icubInterface not working after maintenance (follow-up from robotcub-hackers) (iCubLausanne01)

nisommer, 2016-03-15:
@julijenv 

##### Description of the failure
We recently received our iCub (iCubLausanne01) after a maintenance form IIT.

We noticed that the ubuntu and the iCub codes on the laptop are updated, and there is a new procedure to run the robot.
It seem that we need to run "robotInterface" instead of "icubInterface", but it is not working. A log is attached.
​Do you we still need to run the cluster?

##### Detailed conditions when the failure occurred
Running robotInterface: 
LOG: [icubLausanne_log.txt](https://github.com/robotology/icub-support/files/174013/icubLausanne_log.txt)

julijenv, 2016-03-18:
Hi @nisommer ,
Could you please try to launch the robot by part until reaching the point where you can detect which part of the robot prevent the entire robot from calibrating?
create some new icub_all.xml such as icub_left_arm.xml in which you will have just the left _arm parts enabled in it and see if it starts. do the same for all the other parts unitl you find the culprit.
(dont forget to modify the robotInterface.ini and change the file it should launch).
Let us know

barbalberto, 2016-03-18:
From the log it looks to me the error is it cannot set the controlmode position.
`setControlModeRaw delay ( j:7), current mode: idl, requested: pos`
` [ERROR] 100ms Timeout occured in setControlModeRaw ( j:7), current mode: idl, requested: pos`
and it seems to happen on all parts, but I don't know why

randaz81, 2016-03-18:
It happens because the cfw2 communication on the canbus is somehow corrupted:
`run()-pushReadMsg() failed on CAN bus 4`
`recv buffer overrun ( 500 >  500)`
In my experience this error is typically related to a damaged board (*) which is flooding the buffer (which is shared between all the can buses). Hence, @julijenv is right suggesting to individually disable each part of the robot until the faulty board is located. Other possibilities include a faulty connection between the pc104 and the cfw2 board or a critical damage to the cfw2 board itself, but let's not consider this for now.

(*) Looking at the log I suspect a mais board, but I may be wrong. Disabling each part of the robot one by one will help in locating the source of issue.

barbalberto, 2016-03-18:
thanks @randaz81 for explanation!

randaz81, 2016-03-18:
Check also that you are NOT running by chance this yarp version:
https://github.com/robotology/QA/issues/86

julijenv, 2016-03-23:
Hi @nisommer ,
any news? Did you try what I suggest you to do in order to detect one member error?
let us know

nisommer, 2016-03-23:
Well, I just checked again and this time the robot worked much better when I **released the safety stop button** ...
Sorry for your time, we got a bit confused when the iCub came back and iCubInterface was replaced with robotInterface. Is there a documentation where we can find more info about the other changes like this ?
Thanks

closed 2016-03-23 by nisommer