robotology/mesh-support #204: Right arm board 2B2 not found

kt10aan, 2016-03-16:
Please fill in the following two sections:

##### Description of the failure
During startup, board 2B2 (boardId 3) of the right arm cannot be found. 

##### Detailed conditions when the failure occurred
The arm was working fine, until it was grabbed quite hard from the plastic part. Next startup the problem appeared. Likely reason seems to be a cable disconnection somewhere. In January we had made some repairs to the right arm, and putting the covers back, it is possible that the cables didn't end up in their optimal position under the covers. 

Should cables should we test for continuity?

julijenv, 2016-03-16:
Hi @kt10aan ,
please follow this procedure in order to understand what is going on:
screenshot of net 2 and 6 and 8 with canLoader 
open the upper arm cover -remove the cover nad skin part
check the behaviour of the lights for 2B2 compare to 2B3, 2B4 when you turn on the motor switch.
take some pictures of the wire at the level of the power supply (rainbow colors connector).
cant you see any wire disconnected?
Let us know

kt10aan, 2016-03-17:
Here are screenshots of nets 2, 6 and 8
[image]
[image]
[image]

Here is a photo of the rainbow connector

[image]

The cables seem fine and I have pushed the connector all the way in just to make sure.

I also inspected all the connectors of the board. I could see anything being disconnected.

hen starting up the motors, the following behaviour is observed:

All three boards have their side-by-side blue and orange LEDs lit up. But afterwards, boards 2B3 and 2B4 have the blue LED constantly on and the orange one blinking at a slow rate. However, 2B2 after the initial startup whre both blue and orange is lit up, only blue remains lit up and the orange does not light up, nor blinking.

Here is an image, taken when the 2B3 and 2B4 orange LED is lit up during blinking.
The bottom board is the 2B2, and the orange LED is not blinking or lighting up.

[image]

kt10aan, 2016-03-18:
A further update:

Starting the motors and hitting the Connect button to Net2 within a couple of seconds in ``canLoader``, the board shows up. Trying to flash the firmware using
``canLoader --canDeviceType cfw2can --canDeviceNum 2 --boardId 3 --firmware 4DC.1.19.out.S``

fails with 

``\Initializing prompt version of canLoader...``
``Selecting canDeviceType cfw2can canDeviceNum 2, boardId 3, firmware 4DC.1.19.out.S``
``Cfw2Can::open() returning error -2``
``yarpdev: ***ERROR*** driver <cfw2can> was found but could not open``
``[ERROR]no boards found ``
``[ERROR]check canbus cable, power supply connection etc.``

Also, with the above procedure, the blue and the orange LEDs on board 2B2 are staying on (while on the other boards are following the usual pattern "both on at startup, then blue on, orange blinking".

julijenv, 2016-03-18:
Hi @kt10aan ,
Please remove the 4 screws holding the 3 boards stack (the aluminium L shape) and check the connexion on the power side of the board 2B2 - this board is missing from canLoader so by taking a look to the picture the lights are ON on the board so it means that the CAN line "balck filter" has a cable disconnected.
Please check this board. Let me know

traversaro, 2016-03-18:
@kt10aan can you try to flash the board using the canLoader graphical interface connected to the board in the first couple of seconds from giving power to the board? This solved a similar issue for me in a similar issue, see https://github.com/robotology/icub-support/issues/205#issuecomment-197931051 .

julijenv, 2016-03-18:
@kt10aan , try to see if you can get connected to the board when clicking connect on canLoader within 5 seconds after having powering up the boards? this will tell you if he flash of the board could have failed in some point.

randaz81, 2016-03-18:
Cfw2Can::open() returning error -2
means that you tried to use canLoader while another instance of canLoader (or robotInterface) was running!

kt10aan, 2016-03-18:
@randaz81, doh, you are right, I forgot that the GUI canLoader was running!
@julijenv, I could not see any cable being disconnected. Also, see my answer above; I could see the board when connecting in canLoader within 5 seconds of starting the motors.
I reflashed the board using the GUI canLoader and it worked! @traversaro , thanks for the tip, I had seen #205 and I was going to try it as last resort before "find the bad cable"

So all the boards are showing up nicely now. Very happy that I did not have to  fix cables. However, I am mystified by the reason for this. Before  the issue appeared, everything was working fine and no firmware updates took place during that time.
How can we be sure for the reliability of the firmware update process? Apart from flashing each board 3 times just to make sure! :)
In any case, issue solved.

closed 2016-03-22 by julijenv