robotology/mesh-support #205: Board with ID 5 on Can Net 5 not found

traversaro, 2016-03-17:
##### Description of the failure
Today @DanielePucci , @pi-q and @gabrielenava were updating the robot, and after a firmware update we were not able to launch the robot, with the robot interface giving the following error:
~~~ 
9893		2225.685895		ERROR		sharedcan [5] thread:0 msg:91 joint:0 timed out
9895		2225.685980		ERROR		getFirmwareVersion: message timed out
9897		2225.686115		ERROR		Unable to read firmware version 
9899		2226.179320		ERROR		sharedcan [5] thread:0 msg:91 joint:1 timed out
9901		2226.189387		ERROR		getFirmwareVersion: message timed out
9903		2226.189455		ERROR		Unable to read firmware version 
9907		2226.682756		ERROR		sharedcan [5] thread:0 msg:91 joint:2 timed out
9909		2226.682820		ERROR		getFirmwareVersion: message timed out
9911		2226.682854		ERROR		Unable to read firmware version 
9914		2227.186294		ERROR		sharedcan [5] thread:0 msg:91 joint:3 timed out
9915		2227.186316		ERROR		getFirmwareVersion: message timed out
9917		2227.186350		ERROR		Unable to read firmware version 
9927		2227.266963		WARNING		sharedcan [5] joint: 0 can_address:  5 Unable to detect firmware version. Old firmware running?
9929		2227.276846		WARNING		sharedcan [5] joint: 1 can_address:  5 Unable to detect firmware version. Old firmware running?
9931		2227.276915		WARNING		sharedcan [5] joint: 2 can_address:  6 Unable to detect firmware version. Old firmware running?
9933		2227.276950		WARNING		sharedcan [5] joint: 3 can_address:  6 Unable to detect firmware version. Old firmware running?
9947		2227.277185		ERROR		
9949		2227.277216		ERROR		###################################################################################
9951		2227.277253		ERROR		###################################################################################
9953		2227.277287		ERROR		
9955		2227.277319		ERROR		  It has been detected that your control boards are not using the same
9957		2227.277353		ERROR		  CAN protocol used by iCubinterface. iCubInterface cannot continue.
9959		2227.277387		ERROR		  Please update your system (iCubInterface and/or your control board firmware.
9961		2227.277421		ERROR		  For further information please visit: http://wiki.icub.org/wiki/Firmware
9963		2227.277454		ERROR		
9965		2227.277484		ERROR		###################################################################################
9967		2227.277519		ERROR		###################################################################################
9969		2227.277554		ERROR		
9971		2227.277585		ERROR		checkFirmwareVersions() failed. CanBusMotionControl::open returning false, 
~~~

Checking with the CanLoader, we noticed that the board with ID 5 on Net 5 was not appearing on the network. 

##### Detailed conditions when the failure occurred
The robot was last updated on Jan 21, and at that time @julijenv and @barbalberto did some modification to use the right firmware for the hands, that had been upgrade to V2 mechanics.

randaz81, 2016-03-17:
use the canloader to reprogram the board with the emergency procedure (via bootloader)
click on the `connect button with 5 secs after turning on the board`. The board will appear in the list and you'll be able to reprogram it. Then restart the board.

traversaro, 2016-03-17:
Thank @randaz81 , that fixed everything!

closed 2016-03-17 by traversaro