robotology/mesh-support #207: iCubBielefeld03 converting config files failed

lschilli, 2016-03-21:
Kostas and me ran an attempt to clone an iCub Demo (see issue https://github.com/robotology/QA/issues/102).
The process was quite smooth, we updated the firmware, and things seemes to be progressing fine.
However, the roadblock was that the iCubBielefeld03 was using the old format .ini config files.

For safety reasons we limited our attempt only to the head related configurations, so we limited our attempts to hardware/motorcontrol/icub_head.xml and calibrators/head_calib.xml.

We checked the old iCubOsaka01 ini-format config files. There was no difference in the calibration3 line between the old config file version and the new one. Thus, we replicated the same changes for the iCubBielefeld configuration. i.e. we copied the ini file values for calibration3 from the old Bielefeld config to the new xml format. We processed hardware/motorControl/icub_head.xml in a similar way.

After some testing in fault mode we decided to run the robot calibration. During calibration the head turned to the right along the yaw axis. We obviously oversaw some parameters or hardware differences between the Osaka and Bielefeld robots. Despite our efforts trying to convert the head config files, we were not successful. It seems the zero position is completely off. We currently do not have an explanation for this. About 3 weeks ago the robot was used with the old configuration and at least calibration was working fine. 

So it is likely we did not perform the conversion step correctly or oversaw some other adjustments we should have taken care of. 
We are attaching

- XML files: general.xml, icub_all.xml, calibrators/head_calib.xml and hardware/motorcontrol/icub_head.xml . These were based on Osaka01 corresponding files, with only the changes mentioned above from BIelefeld03 (calibration3 line).
- Screenshots of CAN net0 before and after the firmware upgrade
- RobotInterface logs: One is with RawEncoder data enabled and calibration skipped. The other is with calibration enabled and RawEncoder data disabled. In both logs, the robot is faulted.
- Screenshots of yarpmotorgui with RawEncoder data with the head being approximately in the zero position. And A screenshot of yarpmotorgui with normal encoder data, after calibration and the head approximately in the zero position (always with the robot faulted).

Maybe you can help us in converting the Bielefeld03 head configuration files to xml. The motorgui-raw-nocalib screenshot shows the raw encoders data when the head of the robot is at position zero.
Also, it would be interesting to know the mechanicsms behing these conversions :)

[icub-bielefeld03-debug.zip](https://github.com/robotology/icub-support/files/181591/icub-bielefeld03-debug.zip)

randaz81, 2016-03-21:
According to the screenshot, you flashed the wrong FW version on board n.2. On you hardware revision you need to use v.2.15 and not v1.15. If the hardware is not damaged, reporgramming the board will fix the yaw calibration issue.

lschilli, 2016-03-21:
Wow, that was a stupid mistake :/. Despite double checking we oversaw the wrong board id. Thanks for the quick reply, Marco. We just changed the firmware and calibration worked in an initial test. We will continue to report on our cloning experiment.

kt10aan, 2016-04-19:
Just closing it:)

closed 2016-04-19 by kt10aan