robotology/mesh-support #208: Report of on-site visit of 14-18/03/2016 to Lyon

pattacini, 2016-03-21:
##### Description of the failures

During the WYSIWYD integration meeting we spotted the following two problems on **iCubLyon01**:

- If the **MAIS** boards of the **left arm** are all in place and connected, the robot starts with left forearm joints in **HW fualt**. By disconnecting the MAIS board which is accommodated above the pinkie, the robot starts off with the former joints properly configured, but clearly with some fingers not.
- The focus of the eyes, and particularly the one of the left eye, needs to be mechanically adjusted.

##### Detailed conditions when the failure occurred
These two failures were already present on the robot when we kicked off the meeting.

##### Final note
It is quite crucial to have `iCubLyon01` in a good shape before the beginning of the **second annual review**, which will take place **in Lyon on 19/04/2016**.

@julijenv could you please get in touch with @gregoire-pointeau in order to take the necessary actions?

cc @robotology/wysiwyd-developers

julijenv, 2016-03-22:
Hi @gregoire-pointeau ,
ok let's try to explain how to proceed to find the problem with the MAIS (and actually most probably with the pinky finger of the left hand.
Please use a tester to find the cable which make the short circuit.
1. first try to see if you have contiunity between any of the 4 pin on the top of the nail of the pinky finger and the carcasse and or in between them.
2. remove the glue on the top of the board above the MAIS (FC connector) and check if there is some short-circuit. one pin on ground or red wires and the other in any of the other pads to find the continuity (short circuit)where it shouldnt.
[HAND_CABLING_SCHEMA_V1_2.pdf](https://github.com/robotology/icub-support/files/184413/HAND_CABLING_SCHEMA_V1_2.pdf)

Now let's try to explain how to resolve the issue with the focus of the lenses - take a look at the pictures below:
[image]
[image]
[image]

1. try to remove the IRIS and loose the screw preventing the lens from being unscrewed and so adjust the focus
2. if you cant loose the focus screw (picture 3), remove the 2 lateral screws (picture 1) in order to remove the eye from the mechanics
3. remove the 3 screws holding the white bulb in place and tune the camera (picture 2)
4. reput back in place the overall screws previously mentionned in the other way around

pattacini, 2016-03-24:
@gregoire-pointeau any news on this?

gregoire-pointeau, 2016-03-25:
We haven't been 2 person with the robot yet. It's on the schedule for Tuesday morning !

gregoire-pointeau, 2016-04-01:
Hi @julijenv ,
Concerning the part 1 (left hand). I removed the nail of the pinky finger, but I don't find any pins.
Also, I have alimented the motors but should I run the `pc104` ?

julijenv, 2016-04-01:
Hi @gregoire-pointeau,
 Don't turn on the motors!!
You don't need that to test the continuity.
If you can't find any pins just try to find wire and the pin are where the wires are soldered to the PCB.
Is there any wire broken or disoldered?
From that check if any of those 4 pins is connected to carcasse( metallic part of the robot). And after that in between them.
Later check at the level of the fc connector above the MAIS, is there any wire broken.
Let me know

gregoire-pointeau, 2016-04-06:
Hi @julijenv ,
Concerning the hand:
First of all, the status: if we plug the board after starting the motors, we have access to the joints in `yarpmotorgui`. When we move the fingers of the robot us, by touching it, we see the changes in the Guis (either `yarpmotorgui` or `icubGui`).
If we try to close the finger with `yarpmotorgui`, joint `15` goes to `40` then stops for the Guis, while on the robot, we hear the motors but nothing move.

If we stop the motors, and restart them with the board already plugged, we cannot access it at all. 

Concerning the connectivity.
Under the nails, we have not inter connection (between pins of the nails).
With the carcasse we have a connectivity between the bottom pin the closer from the elbow (for both 4th and 5th finger).

Concerning the board, we would rather not remove the glue, because we are not sure we have the skill and the material to repair it again, but last time we did (less than 6 months), we had no problem of connectivity.

Are you on skype these days?

pattacini, 2016-04-08:
Hi @robotology/icub-support-team @spada1978 

Since @julijenv is on vacation these days, is anyone of you willing to follow this up? It'd be great to try to solve this issue before @julijenv will go to Lyon at the end of next week.

gregoire-pointeau, 2016-04-08:
I add @AnneLaureM to the discussion

julijenv, 2016-04-10:
Hi @gregoire-pointeau ,
You do have some problema with a connection with the cacasse it will be good to discussione about it, but i thhink you may have also a problem with cable of the finger 4-5. Coukd yiu try to open the forearm cover and check if the cable is broken? Ill try to read and give yoi other advucea tomorrow.

pattacini, 2016-04-11:
@julijenv it really seems you're having a "good" orangina on your holiday :smirk:

gregoire-pointeau, 2016-04-12:
From what we see, all cables seem to be fine.
We didn't removed the glue however.

pattacini, 2016-04-17:
Hope @julijenv managed to fix things up in Lyon.

Tobias-Fischer, 2016-04-17:
The `MAIS` is working nicely now. So far, we haven't touched the eyes regarding the focus, mainly not to mess up the calibrations before the demo ....

julijenv, 2016-04-18:
Let's try to sum up the different fixes that have been done during preparation and review meeting (17-19/04/2016):

- The left skin/MAIS issue has been solved -> power wires from N7 at the level of upper arm almost broken so less current poured inside it.

- Elbow hanging piece from the forearm get loose -> thread glued and retighten the locking screw with the coupling piece.

- retense the abduction fingers and opposition thumb movement for the left hand

- refocus the left eye

- retighten screws along the hands

pattacini, 2016-04-18:
Thanks @julijenv :exclamation:

closed 2016-04-18 by pattacini