robotology/mesh-support #213: Pronosupination wrong bounds

pattacini, 2016-03-29:
Together with @randaz81 and @julijenv we found out that the pronosupination (i.e. joint 4 of arm) bounds were actually wrong, that is equal to `[-25,80] deg` rather than being correctly `[-60,60] deg`. Those wrong limits were causing the joint to go in IDLE state when commanded towards the hardware stop, and the downstream wrist joints to go in HW-FAULT; ultimately a big mess :open_mouth: 

We fixed these bounds in the system as well as in the repo.
However, the question is: @robotology/icub-support-team why was iCubGenova01 delivered to the lab with those wrong values?

closed 2016-03-29 by pattacini

pattacini, 2016-03-29:
cc @vtikha

maxime-petit, 2016-03-30:
Hi all,

With iCubLondon01, we have **`[-90,90]`** for the limit of the joint 4. Should we change it also to **`[-60,60]`** or the version of our iCub is fine with these bounds?

./Max
cc @Tobias-Fischer

pattacini, 2016-03-30:
Hi @max 
`iCubGenova01` has `v2` forearms, hence I think you're not affected.

traversaro, 2016-03-30:
@pattacini 
Just for reference : the incorrect value found in a lot of configuration files (even `iCubGenova03`) is **`[-50,80] deg`**, not as **`[-25,80] deg`**. 

see https://github.com/robotology/icub-main/commit/2c438c2761c543f273a9a18136063f78e1e6b82f?w=1

randaz81, 2016-03-30:
If executed, test **jointLimits** from https://github.com/robotology/icub-tests/ is able to detect this issue. It should be always executed when a robot leaves the production area. @davidetome @julijenv @valegagge

davidetome, 2016-03-31:
@randaz81 , jointLimits are executed since 24/02/2016 when you noticed us the existing of tests and  you said to production to do that by email. In particular starting from iCubTwente01 robot on which tests were still debugged by @valegagge  cc @Fabrizio69 @pattacini