robotology/mesh-support #214: PC104 not appearing on the network

traversaro, 2016-04-01:
##### Description of the failure
I turn on the PC104, but it does not appear on the network (i.e. `pc104` ---> 10.0.0.2 is not replyng to any ping). 

##### Detailed conditions when the failure occurred
The iCub was away, and the setup was just mounted. I could have done something stupid and just don't realize it. Anyway I tried to connect to the pc104 directly, but I was unable to find a suitable VGA cable.

pattacini, 2016-04-01:
It's very likely that we didn't connect the black ETH cable into the 10.0.0.x LAN.

traversaro, 2016-04-01:
At a first glance all cables seem to be connected, but I would need some more expert eyes to double check them on Monday.

pattacini, 2016-04-04:
The `PC104` was not connected to the LAN, actually. In the lab we don't use the switch located above the server.

closed 2016-04-04 by pattacini