robotology/mesh-support #215: Report of Chicago mission 30/3/2016

francesco-romano, 2016-04-02:
This issue to report on the mission held in Chicago on the 30th of March 2016 with iCubGenova02. 

The main issues we encountered were on the setup. In particular we miss the following functionalities on the two laptops:
- WiFi support 
- Bluetooth support
- Audio through HDMI

francesco-romano, 2016-04-02:
Cc @DanielePucci @mbrunettini @maggia80 @lornat75

mbrunettini, 2016-04-05:
The two laptops have:
1) wifi
2) bluetooth
3) HDMI audio

DanielePucci, 2016-04-05:
The main point here is that they have to be properly configured. It seemed to us that the drivers were not installed for bluetooth and wifi, while we were not able to work with the HDMI audio. It would be desirable to have them work properly once in Genoa. Thanks!

mbrunettini, 2016-04-05:
They were supposed to be working.
No drivers are needed for wifi and bluetooth, they are included in the kernel image.
I will check them once in Genoa.

DanielePucci, 2016-04-05:
We have been struggling a lot to make them work with no success :disappointed: Thanks for giving us feedback on this!

mbrunettini, 2016-04-06:
Further investigations show:
1) wifi module as been removed from kernel and must be enabled using "Additional drivers" in system settings
2) same as above for bluetooh, but to enable it is not so easy (there is no direct support in this Ubuntu release)
3) HDMI audio is not available due to poor support of nVidia Dirvers in dual GPU configuration, I will check if installing proprietary drivers fixes it
The newcoming release of Ubuntu LTS (16.04) will fix at least the first two issues (I just checked).
My idea is, when the new LTS will be available, to upgrade all iCub laptops.

lornat75, 2016-04-07:
If these issues are still unsolved, we could keep the issue open as a reminder until we do the upgrade?

DanielePucci, 2016-04-07:
@lornat75 👍

mbrunettini, 2016-04-22:
Update after testing Ubuntu 16.04 :
1) wifi works
2) bluetooth works
3) HDMI and dual screen does not work properly
I'm trying other ways to fix 3)

alecive, 2016-04-22:
For what matters, I was able to get HDMI audio working on my Ubuntu only if I connect the cable before booting up (i.e. if I connect it with the laptop already booted I get only video). Then you have to manually change the audio source in the settings.
Disclaimer: I have an Intel integrated graphics card, and not an Nvidia dedicated one.

DanielePucci, 2016-04-24:
Thans @alecive, we did not think about it back in Chicago. Nice workaround :)

alecive, 2016-04-25:
@DanielePucci I don't know if it works also on an NVIDIA card, though :)

mbrunettini, 2016-04-26:
Update: HDMI dual screen output works on Ubuntu 16.04 with latest (361.41) nvidia drivers, from ubuntu proprietary drivers repository.

alecive, 2016-05-04:
@mbrunettini yeah, I just tried to give some help! :+1:

mbrunettini, 2016-05-05:
@alecive : yes, I know, thanks. I tried but it didn't work

mbrunettini, 2016-08-03:
the of Ubunutu 16.04 LTS fixed the above issues.

DanielePucci, 2016-10-16:
It seems to me that Ubuntu 14.04 is installed on both laptops of the ~iCubGenova02~ setup. As a consequence, we could not use bluetooth and wifi, thus making internet connection very difficult to establish. Without these internet connection interfaces, the likelihood of failing demos increases dramatically when performing demos not in Genoa (and on national live tv) :( 

@francesco-romano @iron76 @julijenv

mbrunettini, 2017-02-14:
Update to 16.04 has been stopped on icub-purple. All other IIT platforms have latest Ubuntu LTS. 
Ask @ChiaraBartolozzi for further details.

mbrunettini, 2017-03-07:
Laptops have been upgraded to Ubuntu 16.04

closed 2017-03-07 by mbrunettini