robotology/mesh-support #216: Torso only working for limited amount of time

Tobias-Fischer, 2016-04-07:
Hi @robotology/icub-support-team,
We have some problems with one of the boards of the torso. It might be related to https://github.com/robotology/icub-support/issues/61

##### Description of the failure
We can start the iCub okay. After a while (typically 30-60 minutes), we see the `actionsRenderingEngine` crashing. When trying to restart, it fails saying `failed to get weights from cartesian controller`. Other times, the `actionsRenderingEngine` is not crashing, but does not control the head anymore. However, it thinks it is controlling as issuing a command like `home all` finishes just fine (without moving the head, although it should).
You might think that restarting the cartesian solvers fixes the problem, but it does not. Instead, we need to completely shutdown the robot and try to start again. However, when switching back on the motors, one of the controller boards does not start anymore. It is the upper one with the bright lights in the attached pictures. Both, the lights of the upper and the lower board briefly flicker, then switch off, and after a few seconds only the lights of the lower board switch back on.
[image]
If we try to run `robotInterface` when the lights of the upper board are not switched on, the robot does not start.

We assume that the connectivity of some cables is not 100%, and sometimes becomes loose. Typically, we just wait 10 minutes and "play" a bit with the cables then the board switches on okay.

Could you please give us a hint which cables could cause the problem and how to track down the problem further?

Thanks a lot,
Tobi
/cc @maxime-petit @julijenv @martz21 @hjchang

maxime-petit, 2016-04-12:
Log from robotinterface provided.

[yamrprunlog_12_04_2016_10_30_24_iCubLondon01.log.gz](https://github.com/robotology/icub-support/files/214690/yamrprunlog_12_04_2016_10_30_24_iCubLondon01.log.gz)

julijenv, 2016-04-13:
@maxime-petit , please provide me with a log.txt Fi!e I can't read from where I am now, or else I won't be able to open it b4 next Tuesday, thx in advance

Tobias-Fischer, 2016-04-13:
Hi @julijenv, sure :)
[yamrprunlog_12_04_2016_10_30_24_iCubLondon01.txt](https://github.com/robotology/icub-support/files/217946/yamrprunlog_12_04_2016_10_30_24_iCubLondon01.txt)

julijenv, 2016-04-13:
Hi guys,
From the log file i Can read this from CAN line 0 bottom part:
3
<#STRING_START#>sharedcan [0] joint: 0 can_address:  5 board type: 4 (BLL) version: 1.54 build: 87 CAN_protocol:1.6
<#STRING_END#>
757
5.452138
3
<#STRING_START#>sharedcan [0] joint: 1 can_address:  5 board type: 4 (BLL) version: 1.54 build: 87 CAN_protocol:1.6
<#STRING_END#>
759
5.452227
4
<#STRING_START#>sharedcan [0] joint: 2 can_address:  6 Unable to detect firmware version. Old firmware running?
<#STRING_END#>
761
5.452307
4
<#STRING_START#>sharedcan [0] joint: 3 can_address:  6 Unable to detect firmware version. Old firmware running?
<#STRING_END#>

As you can see it is not possible to detect the board addr 6 which handle joint 3 and 2. Please take a look at the CAN line connection for this specific board the filter try to see if there is continuity. Try within 5 s of the the switch ON the motors if you can see board 6 on CN line 0 from canLOader. If yes try to reflash the fir!ware for this specific board.
Let us know.

Tobias-Fischer, 2016-04-17:
Hi,
After talking to @julijenv, he recommended to do the following:

- [ ] Replace the big black capacitor
- [ ] Check the motor end of the cable with the three wires (red, white, black)
- [ ] Check the filter for loose connections, and maybe re-crimp the filters
- [ ] Check for wear and tear of the wires, for both boards
- [ ] Check the flat cable between the two boards (continuity test)

Thanks @julijenv!

kt10aan, 2016-04-19:
@Tobias-Fischer, just a very freak and extreme case, you may want to have a look at #91 , the section regarding the filters continuity checks.

julijenv, 2016-07-05:
hi @Tobias-Fischer ,
still the issue ???

Tobias-Fischer, 2016-07-05:
Hi @julijenv,
Yes, very rarely but we still sometimes get this issue. It's really hard to debug though :(.
I'll keep you updated, thanks for asking.

Tobias-Fischer, 2016-09-22:
Hi @julijenv, could you please send us the part numbers of the flat cables between the boards (the short one as well as the long one)? We'll order new ones and replace the old ones, just in case.
We just replaced the capacitors, let's see whether it improves the situation :).

Thanks!

julijenv, 2016-09-30:
Hi @Tobias-Fischer ,
here are the part number on Data base
the flat cable is : 
Ribbon flat cable Speedblock, 26 poles, p.0.635mm, 150V, AWG30
You can get it from RS components with th code : 424-2012
the black connector is: 
Flying connector, ERNI SMC-B, 26 contacts, 2 rows, female, p.1.27mm, w to w, lock, nokey, 1.2A
ERNI ELECTRONICS GMBH	214346
I takea look at ERNI and it seems that the connectors looks obsolete, I'll see on monday what to do to send you those parts.
Pleasesend me a personnal mail on moanday to remind me this

Tobias-Fischer, 2016-09-30:
Hi @julijenv,
So we replaced the flat cable as well as the capacitors in the meantime, and we still have the reported issue :(. I'll see whether someone is capable of re-crimping the filters, but I fear that no one is willing to touch them as the cables are too tiny ..
I'll drop you an email regarding the connectors on Monday.

Have a nice weekend,
Tobi

julijenv, 2016-10-04:
any news? @Tobias-Fischer

Tobias-Fischer, 2016-10-04:
Hi @julijenv,
So I've asked our soldering+crimping expert in the lab, and he says the cables visually look okay. However, there is some glue missing at one of the connectors, so we definitely played around with it before. Would it be possible for you to send a complete "pack" of the two filters + cables in between them to us? That might be the best way to go forward. The label is "0N" or "N0", I'm not too sure.

Best, Tobi

Tobias-Fischer, 2016-10-04:
Hi @julijenv,
as requested a photo with annotations. It would be great if you send the two filters (blue+red) along with the cable between them (yellow), and possibly also as many other cables as possible attached which go to the boards in the background.
Let me know if you have questions.

Best,
Tobi

julijenv, 2016-10-05:
CAnt you first try to get your crimping boy to recrimp the N0/NO cable as well as the wires on the side of the connector blue and red? Make a continuity test after the crimping has been done and if everything has been crimped corectly and give good feedback sound from the contuity test you can replug everything. if nothing changed so i will send you some spare parts.
let me know

Tobias-Fischer, 2016-10-05:
Hi @julijenv,
I asked our crimping guy, and I am afraid we don't have a crimp tool for such small crimps :(.
I am on vacation tomorrow and Friday, but @maxime-petit might do the continuity tests tomorrow. Maybe this might provide some insight to our problem.
Thanks again!

Best, Tobi

Tobias-Fischer, 2016-10-25:
Hi @julijenv,
So the past few weeks we haven't seen this issue again. We've seen it only once after replacing the capacitors. We'll keep you updated.
Best, Tobi

julijenv, 2016-10-25:
Hi @Tobias-Fischer , 
good to hear ;)

closed 2018-01-16 by julijenv