robotology/mesh-support #218: Left hand thumb proximal encoder problem

gsaponaro, 2016-04-11:
##### Description of the failure

After using the robot hand for grasping experiments, the encoder of joint 9 (l_thumb_proximal) stopped providing updated readings.

##### Detailed conditions when the failure occurred

After a clean start of the robot and ```robotInterface```, some joints in the left arm start as idle when viewed with the motor gui: 8, 9, 11, 13. For 8, 11 and 13 we have correct encoder readings and we are able to manually activate the control in the gui. However, joint 9 seems to have a more serious problem because the encoder reading is stuck at the value of 108 deg. If we manually move the finger, the value either stays at 108, or sometimes jumps to 10, then going back to 108 abruptly.

[image]

We thought we should measure the continuity, but we are not sure where this encoder is connected in the first place. Is it to the boards in the hand wrist, or to the ones in the upper arm? Any further suggestion? Thanks.

/cc @atabakd @lorejam

julijenv, 2016-04-13:
Hi @gsaponaro,
When you have some problems with the calibration or encoder values of the hands it is a good habit to check the position of the FC boards right above the MAIS board.
Are they all parallels and aligned one to the others? (Remove the dorso cover to have a look to those boards)
If ever you can't see anything please check the wires in between base of the proximl joint and the first FC board to detect broken wire.
Let us know

gsaponaro, 2016-04-15:
The alignment of the five FC boards was correct, but some wires were loose. Problem identified and fixed, thanks. :)

gsaponaro, 2016-05-13:
Unfortunately this problem with the thumb proximal encoder reappeared.

We confirmed that the FC boards position and alignment are correct, and that the wires extremities are connected, both at the boards and at the encoder.

**However, a continuity test showed that on the red wire, current does not flow**. We think the problem might be in the middle of the coloured wires bundle, for example in the part circled in red below, where the wires are protected by the black rubber, which shows some signs of wear.

[image]

[image]

Suggestions?

Thanks in advance.

julijenv, 2016-05-16:
Hi @gsaponaro ,
I unfortunately cant see much on your picture about wear, but AFAIK if one of the 3 wires of this sensor make contact on the frame this why you have a jump on the values and most of the time it got blocked. (could you try to check along the finger if one of the wire touch the frame, if one of the wire is loose or about to disconnect I would presume close to the sensor the red circle you mention earlier).
If you think the red wire is the culprit, please check it try to pull it a bit to see if it is already almost unsoldered and if it is the case try to resolder it (It wont be easy - I prefer telling you).
Let me know your findings

gsaponaro, 2016-05-16:
Thanks @julijenv, and sorry for the bad pictures earlier - here are better ones below. In the finger, none of those wires touches the frame: they are well soldered in that area, as shown. Because of this and because of the continuity test failing on the red wire, we believe that the problem is somewhere "in the middle" of the wire bundle (perhaps at the black rubber part, perhaps elsewhere).

About the suggestion of pulling the red wire slightly: we just tried doing that in the part between black rubber and frame, with no success (we didn't make more force or we would have broken the other wires in the bundle too).

[image]

[image]

julijenv, 2016-05-16:
Hi @gsaponaro and @lorejam ,
the pictures show some interesting features, could you try to see what is exactly this metallic bit of wire at the level of the sensor on the red wire, it looks like it touches the frame?
If ever it is just a flying bit of wire, so it means you will need to check along the 3 wires to know which on has a flaky connection. from the top of the FC connector (the one you circled in green on a previous mail),remove the glue and do continuity test in between the sensor 3 wires (sensor side ) and the FC connector. If it is inside the protective plastic sleeve it would be hard.
Let me know

gsaponaro, 2016-05-16:
Hi @julijenv, thanks for the help on Skype.

Here is a summary for @lorejam and for ourselves as a reminder:

We verified that (1) the artifact in one of the pictures was some hair / junk, not a metal piece, (2) the problem is indeed on the red wire (the one that is attached to the 2 other red wires of the sensors of the finger). Fixing the wire, which is a bit complicated and long, will be required.

Ideally it would be better to use a new wire (AWG38) instead of re-soldering the existing one. We don't have a spare, but if IIT sends us a new spring related to issue #241, then perhaps ~20 cm of this wire could be included in the package.

gsaponaro, 2016-05-30:
Fixed with a new red wire that we disentangled from elsewhere. Thanks.

closed 2016-05-30 by gsaponaro