robotology/mesh-support #220: AEA encoder has spikes warning

traversaro, 2016-04-12:
##### Description of the failure 
In the robotInterface log I noticed this (quite scary) error:
~~~
38244		5616.371848		WARNING		 from BOARD 10.0.1.3 (EB3), src LOCAL, adr 0, time 100s 200m 532u: (code 0x0200000a, par16 0x0403 par64 0x0000000000000002) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
38358		5662.381048		WARNING		 from BOARD 10.0.1.3 (EB3), src LOCAL, adr 0, time 146s 200m 531u: (code 0x0200000a, par16 0x0403 par64 0x0000000000000001) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
38335		5646.380049		WARNING		 from BOARD 10.0.1.3 (EB3), src LOCAL, adr 0, time 130s 200m 531u: (code 0x0200000a, par16 0x0403 par64 0x0000000000000001) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
~~~

##### Detailed conditions when the failure occurred

I launched the robotInterface, and the robot is still hanging on the crane, not moving at all.
Perhaps it is related to https://github.com/robotology/icub-support/issues/177 ? 

Complete log: 
~~~
38237		5614.367968		INFO		icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
38238		5614.649908		INFO		icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
38241		5614.861327		DEBUG		cfw2can [0] thread ran 595 times, req.dT:10[ms], av.dT:10.07[ms] av.loopT :0.02[ms]
38242		5614.861350		DEBUG		     Device has no ICanBusErr interface
38243		5614.861356		DEBUG		cfw2can [0] printing boards infos:
None
38244		5616.371848		WARNING		 from BOARD 10.0.1.3 (EB3), src LOCAL, adr 0, time 100s 200m 532u: (code 0x0200000a, par16 0x0403 par64 0x0000000000000002) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
38245		5616.532979		DEBUG		 from BOARD 10.0.1.2 (EB2), src LOCAL, adr 0, time 100s 363m 184u: (code 0x0500000e, par16 0x0000 par64 0x0000000000000009) -> CFG: EOtheMotionController can correctly configure mc4-based motion. more info will follow + .
38246		5616.533012		DEBUG		 from BOARD 10.0.1.2 (EB2), src LOCAL, adr 0, time 100s 366m 160u: (code 0x05000015, par16 0x0000 par64 0x7f00000000000000) -> CFG: EOtheInertial can be correctly configured. tbd + .
38247		5616.533024		DEBUG		 from BOARD 10.0.1.2 (EB2), src LOCAL, adr 0, time 100s 366m 245u: (code 0x05000004, par16 0x0001 par64 0x7f00000000000000) -> CFG: EOtheSKIN can be correctly configured. num of patches in p16, can mapping in p64 upper 32 bits, prot and vers in p64 lower 32 bits + .
38248		5616.533035		DEBUG		 from BOARD 10.0.1.2 (EB2), src LOCAL, adr 0, time 100s 366m 329u: (code 0x05000009, par16 0x000e par64 0x0000000001000000) -> CFG: EOtheMAIS can be correctly configured. can address in p16, prot and vers in p64 lower 32 bits + .
38249		5616.553101		DEBUG		 from BOARD 10.0.1.1 (EB1), src LOCAL, adr 0, time 100s 378m 179u: (code 0x0500000b, par16 0x0000 par64 0x0000000000000009) -> CFG: EOtheMotionController can correctly configure 2foc-based motion. more info will follow + .
38250		5616.553127		DEBUG		 from BOARD 10.0.1.1 (EB1), src LOCAL, adr 0, time 100s 379m 169u: (code 0x05000007, par16 0x010d par64 0x0000000001000000) -> CFG: EOtheSTRAIN can be correctly configured. can address in p16, prot and vers in p64 lower 32 bits + .
38251		5616.573246		DEBUG		 from BOARD 10.0.1.3 (EB3), src LOCAL, adr 0, time 100s 398m 179u: (code 0x0500000b, par16 0x0000 par64 0x0000000000000009) -> CFG: EOtheMotionController can correctly configure 2foc-based motion. more info will follow + .
38252		5616.573278		DEBUG		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 100s 398m 176u: (code 0x0500000b, par16 0x0000 par64 0x0000000000000009) -> CFG: EOtheMotionController can correctly configure 2foc-based motion. more info will follow + .
38253		5616.573290		DEBUG		 from BOARD 10.0.1.7 (EB7), src LOCAL, adr 0, time 100s 398m 163u: (code 0x05000007, par16 0x0101 par64 0x0000000001000000) -> CFG: EOtheSTRAIN can be correctly configured. can address in p16, prot and vers in p64 lower 32 bits + .
38254		5616.573300		DEBUG		 from BOARD 10.0.1.7 (EB7), src LOCAL, adr 0, time 100s 398m 253u: (code 0x0500000b, par16 0x0000 par64 0x0000000000000009) -> CFG: EOtheMotionController can correctly configure 2foc-based motion. more info will follow + .
38255		5616.573309		DEBUG		 from BOARD 10.0.1.8 (EB8), src LOCAL, adr 0, time 100s 398m 177u: (code 0x0500000b, par16 0x0000 par64 0x0000000000000009) -> CFG: EOtheMotionController can correctly configure 2foc-based motion. more info will follow + .
38256		5616.573321		DEBUG		 from BOARD 10.0.1.3 (EB3), src LOCAL, adr 0, time 100s 399m 172u: (code 0x05000007, par16 0x010d par64 0x0000000001000000) -> CFG: EOtheSTRAIN can be correctly configured. can address in p16, prot and vers in p64 lower 32 bits + .
38257		5616.573330		DEBUG		 from BOARD 10.0.1.9 (EB9), src LOCAL, adr 0, time 100s 397m 166u: (code 0x05000007, par16 0x0101 par64 0x0000000001000000) -> CFG: EOtheSTRAIN can be correctly configured. can address in p16, prot and vers in p64 lower 32 bits + .
38258		5616.573340		DEBUG		 from BOARD 10.0.1.9 (EB9), src LOCAL, adr 0, time 100s 397m 258u: (code 0x0500000b, par16 0x0000 par64 0x0000000000000009) -> CFG: EOtheMotionController can correctly configure 2foc-based motion. more info will follow + .
38259		5616.573349		DEBUG		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 100s 399m 172u: (code 0x05000007, par16 0x010d par64 0x0000000001000000) -> CFG: EOtheSTRAIN can be correctly configured. can address in p16, prot and vers in p64 lower 32 bits + .
38260		5616.573358		DEBUG		 from BOARD 10.0.1.8 (EB8), src LOCAL, adr 0, time 100s 399m 205u: (code 0x05000007, par16 0x010d par64 0x0000000001000000) -> CFG: EOtheSTRAIN can be correctly configured. can address in p16, prot and vers in p64 lower 32 bits + .
38261		5616.593386		DEBUG		 from BOARD 10.0.1.5 (EB5), src LOCAL, adr 0, time 100s 417m 164u: (code 0x0500000b, par16 0x0000 par64 0x0000000000000009) -> CFG: EOtheMotionController can correctly configure 2foc-based motion. more info will follow + .
38262		5616.653798		DEBUG		 from BOARD 10.0.1.4 (EB4), src LOCAL, adr 0, time 100s 483m 137u: (code 0x0500000e, par16 0x0000 par64 0x0000000000000009) -> CFG: EOtheMotionController can correctly configure mc4-based motion. more info will follow + .
38263		5616.653828		DEBUG		 from BOARD 10.0.1.4 (EB4), src LOCAL, adr 0, time 100s 486m 165u: (code 0x05000004, par16 0x0001 par64 0x7f00000000000000) -> CFG: EOtheSKIN can be correctly configured. num of patches in p16, can mapping in p64 upper 32 bits, prot and vers in p64 lower 32 bits + .
38264		5616.653840		DEBUG		 from BOARD 10.0.1.4 (EB4), src LOCAL, adr 0, time 100s 486m 278u: (code 0x05000009, par16 0x000e par64 0x0000000001000000) -> CFG: EOtheMAIS can be correctly configured. can address in p16, prot and vers in p64 lower 32 bits + .
38265		5616.653850		DEBUG		 from BOARD 10.0.1.4 (EB4), src LOCAL, adr 0, time 100s 487m 320u: (code 0x05000015, par16 0x0000 par64 0x7f00000000000000) -> CFG: EOtheInertial can be correctly configured. tbd + .
38268		5619.412946		INFO		icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
38269		5619.694898		INFO		icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
38272		5620.863001		DEBUG		cfw2can [0] thread ran 595 times, req.dT:10[ms], av.dT:10.07[ms] av.loopT :0.02[ms]
38273		5620.863031		DEBUG		     Device has no ICanBusErr interface
38274		5620.863041		DEBUG		cfw2can [0] printing boards infos:
None
38277		5624.458105		INFO		icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
38278		5624.740064		INFO		icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
38279		5626.532461		DEBUG		 from BOARD 10.0.1.2 (EB2), src LOCAL, adr 0, time 110s 363m 336u: (code 0x0500000e, par16 0x0000 par64 0x000000000000000a) -> CFG: EOtheMotionController can correctly configure mc4-based motion. more inf [... cut ...]
38509		5755.610925		INFO		icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
38512		5755.892815		INFO		icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
38515		5758.923389		DEBUG		cfw2can [0] thread ran 595 times, req.dT:10[ms], av.dT:10.07[ms] av.loopT :0.02[ms]
38516		5758.923423		DEBUG		     Device has no ICanBusErr interface
38517		5758.923433		DEBUG		cfw2can [0] printing boards infos:
None
~~~

traversaro, 2016-04-13:
While working on `YOGA++`, this warning occurs roughly every minute.

paraschos, 2016-10-21:
What was the resolution of this one? We still have it! 

[ypmrprunlog_20_10_2016_08_28_09_pushing_test.log.txt](https://github.com/robotology/icub-support/files/544119/ypmrprunlog_20_10_2016_08_28_09_pushing_test.log.txt)

closed 2018-01-16 by julijenv