robotology/mesh-support #221: Broken AEA Encoder on l_hip_yaw (EMS 6 Axis 2)

traversaro, 2016-04-12:
##### Description of the failure
The AEA on l_hip_yaw is reporting crazy data, this is an illustrative output of the `/icub/left_leg/state:o`: 
~~~
-0.181275 -0.021973 -2464.722814 0.0 0.087891 0.010986
-0.181275 -0.021973 -2464.722814 0.0 0.087891 0.010986
-0.181275 -0.021973 -2464.722814 0.0 0.087891 0.010986
-0.181275 -0.021973 -2464.722814 0.0 0.087891 0.010986
-0.181275 -0.021973 -2464.722814 0.0 0.087891 0.010986
-0.181275 -0.021973 -2464.722814 0.0 0.087891 0.010986
-0.181275 -0.021973 -2464.722814 0.0 0.087891 0.010986
-0.181275 -0.021973 -2306.079849 0.0 0.087891 0.010986
-0.181275 -0.021973 -2446.793083 0.0 0.087891 -0.076904
-0.181275 -0.021973 -2327.085759 0.0 0.087891 0.010986
-0.181275 -0.021973 -2327.085759 0.0 0.087891 0.010986
-0.181275 -0.021973 -2327.085759 0.0 0.087891 0.010986
-0.181275 -0.021973 -2327.085759 0.0 0.087891 0.010986
-0.093384 -0.021973 -2327.085759 0.0 0.087891 0.010986
-0.181275 -0.021973 -2152.27088 0.0 0.087891 0.010986
-0.181275 -0.021973 -2152.27088 0.0 0.087891 0.010986
-0.181275 -0.021973 -2152.27088 0.0 0.087891 0.010986
-0.181275 -0.021973 -2152.27088 0.0 0.087891 0.010986
-0.181275 -0.021973 -2152.27088 0.0 0.087891 0.010986
-0.181275 -0.021973 -2152.27088 0.0 0.087891 0.010986
-0.181275 -0.021973 -2152.27088 0.0 0.087891 0.010986
-0.181275 -0.021973 -2152.27088 0.0 0.087891 0.010986
-0.181275 -0.021973 -1996.791984 0.0 0.087891 0.010986
-0.181275 -0.021973 -1996.791984 0.0 0.087891 0.010986
-0.181275 -0.021973 -1996.791984 0.0 0.087891 0.010986
-0.181275 -0.021973 -1996.791984 0.0 0.087891 0.010986
-0.181275 -0.021973 -1996.791984 0.0 0.087891 0.010986
-0.181275 -0.021973 -1996.791984 0.0 0.0 0.010986
-0.181275 -0.021973 -1996.791984 0.0 0.087891 0.010986
-0.181275 -0.021973 -1996.791984 0.0 0.087891 0.010986
-0.181275 -0.021973 -1996.791984 0.0 0.087891 0.010986
-0.181275 -0.021973 -1996.791984 0.0 0.087891 0.010986
-0.181275 -0.021973 -1990.112281 0.0 0.087891 0.010986
-0.181275 -0.021973 -1990.112281 0.0 0.087891 0.010986
-0.181275 -0.021973 -1990.112281 0.0 0.087891 0.010986
-0.181275 -0.021973 -1990.112281 0.0 0.087891 0.010986
-0.181275 -0.021973 -1990.112281 0.0 0.087891 0.010986
-0.181275 0.065918 -2003.911142 0.0 0.0 0.010986
-0.181275 -0.021973 -2018.149458 0.0 0.087891 0.010986
-0.181275 -0.021973 -2018.149458 0.0 0.087891 0.010986
-0.181275 -0.021973 -2018.149458 0.0 0.087891 0.010986
-0.181275 -0.021973 -2018.149458 0.0 0.0 0.010986
-0.181275 -0.021973 -2003.911142 0.0 0.087891 0.010986
-0.181275 -0.021973 -2003.911142 0.0 0.087891 0.010986
-0.093384 -0.021973 -1971.479423 0.0 0.087891 0.010986
-0.093384 0.065918 -1834.721276 0.0 0.087891 0.010986
-0.181275 -0.021973 -1834.721276 0.0 0.087891 0.010986
-0.181275 -0.021973 -1834.721276 0.0 0.087891 0.010986
-0.181275 -0.021973 -1834.721276 0.0 0.087891 0.010986
-0.181275 -0.021973 -1970.600514 0.0 0.087891 0.010986
-0.181275 -0.021973 -1970.600514 0.0 0.087891 0.010986
-0.181275 -0.021973 -1970.600514 0.0 0.087891 0.010986
-0.181275 -0.021973 -1970.600514 0.0 0.087891 0.010986
-0.181275 -0.021973 -1970.600514 0.0 0.087891 0.010986
-0.181275 -0.021973 -1970.600514 0.0 0.087891 0.010986
-0.181275 -0.021973 -1970.600514 0.0 0.087891 0.010986
-0.181275 -0.021973 -1970.600514 0.0 0.087891 0.010986
-0.181275 -0.021973 -1970.600514 0.0 0.087891 0.010986
-0.181275 -0.021973 -1970.600514 0.0 0.087891 0.010986
-0.181275 -0.021973 -1792.270001 0.0 0.087891 0.010986
-0.181275 -0.021973 -1766.07853 0.0 0.087891 0.010986
-0.181275 -0.021973 -1766.07853 0.0 0.087891 0.010986
-0.181275 0.065918 -1766.07853 0.0 0.087891 0.010986
-0.181275 -0.021973 -1766.07853 0.0 0.087891 0.010986
-0.181275 -0.021973 -1766.07853 0.0 0.087891 0.010986
-0.181275 -0.021973 -1766.07853 0.0 0.087891 0.010986
-0.181275 -0.021973 -1766.07853 0.0 0.087891 0.010986
[... 363 lines cut ...]
-0.181275 -0.021973 -190.107886 0.0 0.087891 0.010986
-0.181275 -0.021973 -190.107886 0.0 0.087891 0.010986
-0.181275 -0.021973 -355.870009 0.0 0.087891 0.010986
-0.181275 -0.021973 -355.870009 0.0 0.087891 0.010986
-0.181275 -0.021973 -355.870009 0.0 0.087891 0.010986
-0.181275 -0.021973 -218.145064 0.0 0.087891 0.010986
-0.181275 -0.021973 -218.145064 0.0 0.087891 0.010986
~~~

##### Detailed conditions when the failure occurred
We were testing the YOGA++ demo (the robot is switching from balancing on two feet to balancing on one foot) and while the robot was balancing on the right leg (so the left leg was not supporting any weight) we started receiving the following error from the robotInterface: 
~~~
49633		9341.292438		WARNING		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 2364s 513m 531u: (code 0x0200000a, par16 0x0102 par64 0x00000000000003a0) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
49643		9342.299227		WARNING		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 2365s 513m 533u: (code 0x0200000a, par16 0x0102 par64 0x00000000000003a0) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
49647		9343.295823		WARNING		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 2366s 513m 535u: (code 0x0200000a, par16 0x0102 par64 0x0000000000000399) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
49648		9344.302529		WARNING		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 2367s 513m 532u: (code 0x0200000a, par16 0x0102 par64 0x000000000000037b) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
49658		9345.299316		WARNING		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 2368s 513m 530u: (code 0x0200000a, par16 0x0102 par64 0x0000000000000386) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
49677		9346.295902		WARNING		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 2369s 513m 534u: (code 0x0200000a, par16 0x0102 par64 0x0000000000000376) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
49686		9347.292061		WARNING		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 2370s 513m 531u: (code 0x0200000a, par16 0x0102 par64 0x000000000000038e) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
49689		9348.298344		WARNING		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 2371s 513m 530u: (code 0x0200000a, par16 0x0102 par64 0x0000000000000374) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
49693		9349.294939		WARNING		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 2372s 513m 531u: (code 0x0200000a, par16 0x0102 par64 0x0000000000000370) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
49701		9350.291907		WARNING		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 2373s 513m 532u: (code 0x0200000a, par16 0x0102 par64 0x000000000000037e) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
49722		9351.298870		WARNING		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 2374s 513m 531u: (code 0x0200000a, par16 0x0102 par64 0x0000000000000373) -> MC: AEA encoder has spikes. There is impulsive noise in the measures of the magnetic position sensor of the joint. par16 = AEA port (msb) and ID of joint (lsb). + .
~~~
and clearly then the controller went crazy, and the joint went in HARDWARE_FAULT mode.

traversaro, 2016-04-12:
Rebooting motors and pc104 solved (at least for the time being) the problem.

traversaro, 2016-04-13:
Today the error occured while the robot was not moving at all, and it was controlled simply in position. Clearly there is some problem in the encoder (or in its readings), even if rebooting the problem disapperars. 

@robotology/icub-support-team

traversaro, 2016-04-14:
The problem appeared also today. The frequency is more or less once a day.

traversaro, 2016-05-05:
And the error occured again today! Fortunatly we have @julijenv with us this time. Log: 
~~~
49334		23625.464221		ERROR		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 63s 947m 545u: (code 0x02000008, par16 0x0102 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
49535		23630.468486		ERROR		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 68s 947m 545u: (code 0x02000008, par16 0x0102 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
49776		23635.462544		ERROR		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 73s 947m 542u: (code 0x02000008, par16 0x0102 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
49832		23640.466797		ERROR		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 78s 947m 544u: (code 0x02000008, par16 0x0102 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
49849		23645.460485		ERROR		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 83s 947m 543u: (code 0x02000008, par16 0x0102 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
49863		23650.464356		ERROR		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 88s 947m 543u: (code 0x02000008, par16 0x0102 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
49876		23655.458197		ERROR		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 93s 947m 542u: (code 0x02000008, par16 0x0102 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
49882		23660.462156		ERROR		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 98s 947m 540u: (code 0x02000008, par16 0x0102 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
49897		23665.466274		ERROR		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 103s 947m 542u: (code 0x02000008, par16 0x0102 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
49912		23670.460139		ERROR		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 108s 947m 543u: (code 0x02000008, par16 0x0102 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
49923		23675.464221		ERROR		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 113s 947m 549u: (code 0x02000008, par16 0x0102 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
49936		23680.467873		ERROR		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 118s 947m 543u: (code 0x02000008, par16 0x0102 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
49949		23685.472257		ERROR		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 123s 947m 543u: (code 0x02000008, par16 0x0102 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
49955		23690.465716		ERROR		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 128s 947m 542u: (code 0x02000008, par16 0x0102 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
49972		23695.469726		ERROR		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 133s 947m 542u: (code 0x02000008, par16 0x0102 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
49985		23700.472902		ERROR		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 138s 947m 546u: (code 0x02000008, par16 0x0102 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
[... cut ...]
50897		23785.473709		ERROR		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 223s 947m 808u: (code 0x02000008, par16 0x0102 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
50972		23790.477426		ERROR		 from BOARD 10.0.1.6 (EB6), src LOCAL, adr 0, time 228s 947m 808u: (code 0x02000008, par16 0x0102 par64 0x0000000000000000) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
~~~

DanielePucci, 2016-05-07:
This issue simmer to be related to #201

DanielePucci, 2016-05-07:
We think that the problem is definitely similar to that of #201. Unfortunately, we do not have any oscilloscope to deep into the problem, so we cannot be sure that there are drops in the tension. For the time being, we proceed by substituting the EMS 6 @julijenv @traversaro @francesco-romano @nunoguedelha

DanielePucci, 2016-05-07:
Not having available EMS in the trolley toolbox, we switched the skin upper leg EMS 10 with the EMS 6.

DanielePucci, 2016-05-08:
Switching the EMS did not solve the issue. Just occurred again.

traversaro, 2016-05-13:
We substituted the sensor 4 days ago, and since then the issue never occured again. Closing the issue.

closed 2016-05-13 by traversaro