robotology/mesh-support #225: Replacement tendons needed

kt10aan, 2016-04-20:
I am not describing the failure and conditions as this is not a question of how to fix a damage, but rather a request for tendons :)

Cable 1 in the left shoulder broke, and while usually its the best kind of damage as it can be replaced quite easily and fast, we are out of No1 tendons!

In fact, our tendon supply is getting low, and considering the time it takes for spares to reach Osaka, we would be grateful if you could bump our stock a bit.

Here is a list of the tendons we have:

-No1: none
-No2: x1
-No3: x1
-No4: x2
-No5: x2
-No6: x2
-No7: x2
-No8: none
-No9: x2
-No10: x2
-No11: x1
-No12: x1
-No13: x1
-No14: x1

Also, we have no tendons for fingers and yesterday we have a tendon snap at the right thumb:

[image]

Could we also have a length of finger wire?

Please email me if any procedure is needed from our side.

julijenv, 2016-04-22:
Hi @kt10aan ,
How many spare do you need?
Considering that you dont walk (you could reduce the N14 and N16 spare request to 2).
I would advise you to have at least 3 for each cable and extend to 5 spares cable for number 7 and 8 
Make a proper request to the number you want, I can make you a list of price for each piece (price may vary from suppliers we used - so when you put the list of cable you want just keep in mind that the order may be more expensive.
here is the indicative price list for each cable
cable 1 : (IITCODE 4758.1) : 10,13E
cable 1 : (IITCODE 4758.1) : 10,13E
cable 1 : (IITCODE 4758.1) : 10,13E
cable 1 : (IITCODE 4758.1) : 10,13E
cable 1 : (IITCODE 4758.1) : 10,13E
cable 1 : (IITCODE 4758.1) : 10,13E
cable 1 : (IITCODE 4758.1) : 10,13E
cable 1 : (IITCODE 4758.1) : 10,13E
cable 1 : (IITCODE 4758.1) : 10,13E

kt10aan, 2016-04-22:
Just to make it clear, there are 16 cables and not 14?

I will make a list and email it to icus-support and we can proceed from there.

kt10aan, 2016-04-22:
Here is the list of the cables we need then to restock.

- No1: x3
- No2: x2
- No3: x2
- No4: x1
- No5: x1
- No6: x1
- No7: x3
- No8: x5
- No9: x1
- No10: x1
- No11: x2
- No12: x2
- No13: x2
- No14: x1
- No16: x2

We do have the two type of cables for the fingers. But I will have to check on Monday for how much length is left and if we don't have enough  I will add them to the order.

The wrists seem to work fine. Is there some test I should perform on them?

julijenv, 2016-04-22:
Hi @kt10aan ,
Ok I'll wait for you to give me the definitive list in the meanwhile just check if you have some spare part to replace the wrist cables, make some pictures of the wrist to see if the tendon are with the crimp.
thx in advance

kt10aan, 2016-04-25:
Ok, I found a "secret" stash of a few more tendons :) Tendon no1 was among them.

So now the list of cables needed is:

- No1: x2
- No2: none
- No3: none
- No4: x1
- No5: x1
- No6: x1
- No7: x3
- No8: x2
- No9: x1
- No10: x1
- No11: x2
- No12: x2
- No13: x2
- No14: x1
- No16: x2

- 6.5m of uncoated cable for the fingert (its ok if it comes only in a 10m pack, I am just calculating how much we need to reach the 10m stock) 
- 6m of the coated cable for the fingers (the same)

All of the above we definitely need, please put them in the order.

We also have 2x 80cm plastic tubes for the finger tendons and 3x50cm steel tubes for the finger tendons. Do we need more of them?

Here are the photos of the wrists:

Left
[image]

Right
[image]

The only other (tendon/metal) cables we have are these, which look like mini tendons with mini crimps and named only "Cavo A" and "Cavo B" (and I do not kow what they are for)

[image]

Do you think we need any of that special wrist-related cable you mentioned?

julijenv, 2016-04-26:
Hi @kt10aan ,
cavo "A" and "B" are for the neck joints. so yes i guess it would be useful for you to know where they go.
You have the V2 wirst and if you didnt find any other cable with some crimp on it, I would suggest you to order once or 2 for each, i could add them to the list if you wish.
what kind of spring tube do u have?
the smooth one or the harder one?
Let me know that we could start to make an offer for the total wires?

kt10aan, 2016-06-23:
Updated list of tendons we need:

- No1: x2
- No2: x2
- No3: x3
- No4: x1
- No5: x1
- No6: x1
- No7: x4
- No8: x4
- No9: x1
- No10: x1
- No11: x2
- No12: x2
- No13: x2
- No14: x1
- No16: x1

We also want:
- 10m of uncoated cable for the fingers
- 10m of uncoated cable for the fingers
- Three of each plastic cap for the shoulder tendonds

We have: 
- 2 x 50cm spring metal tubes for the hand tendons (the more flexible ones)
- 1 x 50cm spring metal tube for the hand tendons (the not so flexible one)
- 2 x 80cm plastic tubes for the hand tendons

Considering that we want to replace the tendons of one hand soon, how many of the above tubes you think we should have in stock?  Once we have the answer, we add it to the total list and you could give us an exact quote to proceed with the order.

julijenv, 2016-06-23:
Hi @kt10aan ,
replacing the overall tendons is quite a work that should be done just in case of one of the cable snapped or the extension or entire movement is not anymore functionnal but there is always the way to retense them in the V2 version where the end of the proximal joint have a screw knot holder which basically allow you to pull the cable to retense it (at the level of the first phalange the closer from the palm).
this being said, replacing the tendon does not imply also to replace the tubes as well, I mean you should do it only if the tubes got bent or when you rewire the tendons and you find yourself in a position in which the tendons seems no more being able to be got pulled in inside the tube ( it got stuck for some reasons, by retracing the path of the cable you may find out that the tubes (outside one - rigid one or the one a bit less - or the inside one - plastic tube) got bent and of course prevent the cable from being pulled in.
So to conclude, I would suggest you first to check the tension of every single tendon and see if you can improve them by the trick of the retension. Afterwards if it got snapped or seems not working as desired - first, remove the old one and proceed to recable it, if you dont run into trouble by pulling the cable inside the tube you dont need to change them, if you got stuck you may have to change them.

In the set up of cable we will send you the invoice , i will take care of adding the sufficicent amount of tubes for you to rewire an entire hand in the worst case scenario that all of them got bent and of no use anymore ;).

Could you please give the name, address, phone number of the person in charge of proceeding with the order and receiving the set up of cables.

Let us know

kt10aan, 2016-06-23:
@julijenv, thank you for the clarifications!

I know about the tensioners in V2 fore arms. We do however have a completely broken tendon in the right thumb and a badly damaged one in the last two digits. 

Usually, I avoid the wiring of the fingers unless the cables are broken, as it is time-consuming :) 

It's mostly about the peace of mind of having enough spare parts in the event of a major accident as we are quite far away from Italy and all orders take longer for us, especially if you account for the Japanese customs procedures :)

I will send you the details of the person in an email. Thanks again!

julijenv, 2016-06-23:
Hi @kt10aan ,
I made a draft of the list of cables (and tubes) you will need following your request and adding what I suppose you may need.
[order_osaka_list_of_cables.xlsx](https://github.com/robotology/icub-support/files/329991/order_osaka_list_of_cables.xlsx)

take a look at it, and if you agree with that and you will have give me all the necessary informations that i requested to you previously fro the shipment, I will proceed to send the list to the office dedicated to create the invoice, afterwards you will be free to proceed.
One more things please give me the mail address as well of all the people that may need to be informed of the invoice (you, the buyer, your supervisor, you tell me)
thx in advance

kt10aan, 2016-06-23:
Thank you! Due to time differences, I will email you the information you asked tomorrow morning. By the way, could you send me an (approximate) quote for the list?

julijenv, 2016-06-23:
unfortunately I cant, check the one they will send to u, and tell us what you think about it.

julijenv, 2016-10-04:
did you receive the tendons?
if yes could you close the issue thx

closed 2017-05-02 by kt10aan