robotology/mesh-support #226: left_leg F/T sensor constantly overflowing when the robot is attached to the pole

traversaro, 2016-04-21:
##### Description of the failure
The robot is on the pole, turned on by more then an hour.
The left_leg sensor is then constantly overflowing, flooding the robotInterface of this error : 
~~~
212723		2433,690877		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212733		2433,791698		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212744		2433,892473		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212763		2434,093837		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212773		2434,194461		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212796		2434,385725		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212797		2434,395803		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212806		2434,486531		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212816		2434,587314		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212826		2434,688147		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212842		2434,788881		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212852		2434,889711		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212862		2434,990387		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212872		2435,091178		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212886		2435,191882		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212896		2435,292594		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212906		2435,393380		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212916		2435,494125		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212930		2435,594888		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212943		2435,685683		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212953		2435,786455		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212963		2435,887318		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212977		2435,988081		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212987		2436,088870		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
212997		2436,189715		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213011		2436,290547		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213026		2436,391451		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213027		2436,401468		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213036		2436,492291		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213046		2436,593101		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213056		2436,693883		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213067		2436,794661		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213086		2436,986192		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213096		2437,086940		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213106		2437,187687		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213116		2437,288421		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213126		2437,389197		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213136		2437,490037		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213146		2437,590763		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213157		2437,691604		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213178		2437,893125		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213188		2437,994055		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213197		2438,094895		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213198		2438,094928		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213217		2438,286373		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213218		2438,296448		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213227		2438,387228		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213237		2438,487973		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213247		2438,588729		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213257		2438,689425		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213267		2438,790151		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213277		2438,890895		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213287		2438,991672		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213297		2439,092413		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213307		2439,193119		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213317		2439,294065		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213336		2439,485552		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213337		2439,495621		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213346		2439,586290		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213356		2439,687176		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213365		2439,788072		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213366		2439,788111		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213376		2439,888896		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213386		2439,989711		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213395		2440,090472		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
213396		2440,090500		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
[... 242 lines cut ...]
216434		2464,893622		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
216444		2464,994391		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
216463		2465,185812		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
216477		2465,286512		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
216487		2465,387350		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
216497		2465,488170		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
216507		2465,589073		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
216525		2465,689876		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
216535		2465,790714		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
216545		2465,891459		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
216559		2465,992227		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
216569		2466,092943		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
216579		2466,193847		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
216589		2466,294652		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
216606		2466,385428		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
216616		2466,486216		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
216617		2466,496295		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
216626		2466,587120		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
216647		2466,788781		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
216657		2466,889648		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
216667		2466,990494		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
216678		2467,091456		ERROR		AnalogWrapper: left_leg: Sensor returned overflow error (code 2).
~~~

I plan to try to address this by modifying the low-level ADC offsets, but it is probably just pulling a blanket that is too short.

traversaro, 2016-04-21:
cc @DanielePucci @francesco-romano @nunoguedelha @fjandrad

DanielePucci, 2016-04-21:
I think that we changed the AD offset so as they do not sturate when balancing, but they may when the robot is attached to the pole.

julijenv, 2016-10-04:
news from this issue has been solved i guess since then, or not?
@traversaro

traversaro, 2016-10-04:
I do not remember, but I think we can close and eventually open it again if there are again problems.

closed 2016-10-04 by traversaro