robotology/mesh-support #230: unable to start robotinterface in iCubNancy01

serena-ivaldi, 2016-04-28:
Hi,
first message from iCubNancy01 (team includes @dogoepp @jbmouret @misaki43 and olivier rochel).

We just received iCubNancy01 and we are starting it for the first time from the laptop. 
We didn't update the code of the laptop, nor changed anything in the configuration of the robot/laptop.
We plugged the robot's ethernet cable to the IIT laptop. We launched yarpserver on the laptop, the two yarprun clients on icubsrv and pc104, then yarplogger. When we launch robotinterface, either manually from the terminal of the pc104 or from the yarpmanager, we get several errors. 

The robotinterface is unable to start properly and finally it crashes.
In attachment the log of the robotinterface.
[log_0428_2.txt](https://github.com/robotology/icub-support/files/240408/log_0428_2.txt)

The main warnings/errors that I find weird are:

1) at the beginning, we have a lot of warnings related to invalid syntax in the configuration files:

[[01;33mWARNING[00m]Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNancy01/./hardware/electronics/pc104.xml at line 1 . No DTD found. Assuming version robotInterfaceV1.0 
[[01;33mWARNING[00m]Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNancy01/./cartesian/left_arm-cartesian.xml at line 1 . "devices" element should contain the "robot" attribute 
[[01;33mWARNING[00m]Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNancy01/./cartesian/left_arm-cartesian.xml at line 1 . Trying to import a file for the wrong robot. Found  instead of iCubNancy01 
[[01;33mWARNING[00m]Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNancy01/./cartesian/left_arm-cartesian.xml at line 1 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 

It seems that these warnings cannot be ignored as then they seem to create problems in the robot startup. It also worries me that it's saying " No DTD found. Assuming version robotInterfaceV1.0 " 

Later on, it cannot configure properly the cartesian controller (mixing " old format"  and skipping parameters).
[[01;34mINFO[00m]created device <cartesiancontrollerserver>. See C++ class ServerCartesianController for documentation.
[[01;33mWARNING[00m]Rotor: Using default value = 1
[[01;33mWARNING[00m]Position Pids section found, old format
[[01;33mWARNING[00m]Torque control parameters not found for part , skipping...
[[01;32mDEBUG[00m]DEBUG parameters section NOT found, skipping...
[[01;32mDEBUG[00m]IMPEDANCE parameters section NOT found, skipping...
[[01;33mWARNING[00m]jntVelMax: Using default jntVelMax=100 deg/s
[[01;33mWARNING[00m]maxPosStep: Using default MaxPosStep=10 degs
[[01;33mWARNING[00m]Rotor: Using default value = 1
[[01;33mWARNING[00m]Position Pids section found, old format
[[01;33mWARNING[00m]Torque control parameters not found for part , skipping...
[[01;32mDEBUG[00m]DEBUG parameters section NOT found, skipping...
[[01;32mDEBUG[00m]IMPEDANCE parameters section NOT found, skipping...

2) The first error we get is a firmware error:

[[01;34mINFO[00m]CanBusResources::initialized correctly
Registering new thread 0 out of 500
[[01;31mERROR[00m]cfw2can [0] thread:0 msg:91 joint:0 timed out
[[01;31mERROR[00m]getFirmwareVersion: message timed out
[[01;31mERROR[00m]Unable to read firmware version 
<...cut..>
[[01;34mINFO[00m]**********************************
[[01;34mINFO[00m]iCubInterface CAN protocol: 1.6
[[01;34mINFO[00m]Firmware report:
[[01;33mWARNING[00m]cfw2can [0] joint: 0 can_address:  1 Unable to detect firmware version. Old firmware running?
<...cut..>
[[01;33mWARNING[00m]cfw2can [0] joint: 5 can_address:  3 Unable to detect firmware version. Old firmware running?
[[01;31mERROR[00m]###################################################################################
[[01;31mERROR[00m]###################################################################################
[[01;31mERROR[00m]
[[01;31mERROR[00m]  It has been detected that your control boards are not using the same
[[01;31mERROR[00m]  CAN protocol used by iCubinterface. iCubInterface cannot continue.
[[01;31mERROR[00m]  Please update your system (iCubInterface and/or your control board firmware.
[[01;31mERROR[00m]  For further information please visit: http://wiki.icub.org/wiki/Firmware
[[01;31mERROR[00m]
[[01;31mERROR[00m]###################################################################################
[[01;31mERROR[00m]###################################################################################

Seemingly the problem is the head:
yarpdev: ***ERROR*** driver <canmotioncontrol> was found but could not open
[[01;33mWARNING[00m]Cannot open device head_mc 
[[01;33mWARNING[00m]Cannot open device head_mc 

What should we do ? Thanks in advance for any help.

serena-ivaldi, 2016-04-28:
In the meantime, I launched canLoader to check the boards of the head.
With Net0 + cfw2can it gives me the error "Communication error, no answers received, no boards found".
Is maybe a board disconnected?

serena-ivaldi, 2016-04-28:
I opened the head and found a broken wire and a small connector that is not attached to anything.
It comes from a cable with the label N0 so I guess it is the one giving the error.
See attached images.

[image]
[image]
[image]

I have two questions:

1) I guess that we have to put the flying white wire back into the connector 
=> can we do it just by soldering it a little bit or do you advise to put a new pin into the connector?

2) There is a small connector that is "flying" and seemingly not connected to anything. There is however a small black cable around, suggesting that it was connected to something else probably. Is it normal?
In any case, to which board (where) should we connect it? 
I have no clue about that (I tried to look at the blue wiring schemes but I am not sure to understand) and there is not a lot of space between the boards to have a clear look.
=> what should I do?

julijenv, 2016-04-29:
Hi @serena-ivaldi ,
Sorry to hear that, it is really undortunate, that this can of thing appear just by packing the robot.
It is just a little unplug wire (and visible by the way, so quite easy to detect the failure, as you did, nice job) You just need to recrimp the wire now.
and let us know if every thing alright.
Please go on the wiki page of icub to get the essential tool list of daily repair to fetch. Soldering on the pin wont last and it is not appropriate.

jbmouret, 2016-04-29:
OK, we will recrimp the wire. 

What about the second issue (the disconnected connector) Where should we plug it?

julijenv, 2016-04-29:
make a picture of face of the robot withour the face cover at the level f the boards.
thx in advance

dogoepp, 2016-04-29:
Here is the picture.

[image]

julijenv, 2016-04-29:
It is a spare board filter for the spare MC4 board.
try to launch the robot and send me the logfile if ever it doesnt calibrate.
thx c u soon

jbmouret, 2016-04-29:
thanks. We will need to fix the crimp first and will keep you up to date.

dogoepp, 2016-05-09:
Fixing the crimp solved this issue for us. We now face other problems but will open a separate issue for these.
Thanks !

closed 2016-05-09 by dogoepp