robotology/mesh-support #231: Correct limits for arm joint #4

frank-foerster, 2016-04-28:
When testing the joints of our iCub (iCubHertfordshire01) via the yarpmotorgui, I noticed that the maximum values for joint #4 of both arms appeared to be too large.
When manually setting the target values above a certain threshold (between 71 and 75, depending
on the arm) the joint would go into hardware fault. The previous upper limit as specified in the files below was set to 85.

I have changed now the upper limits for this joint in both 
  .../iCubHertfordshire01/hardware/motorControl/icub_left_arm.xml, section LIMITS
and 
 .../iCubHertfordshire01/hardware/motorControl/icub_right_arm.xml (same section and joint)
to 70 as upper limit and the HW FAULT is subsequently not triggered any more.

So I was wondering if these values, which I determined via trial and error, are the correct values or if there might be some underlying problem that I'm masking now by disallowing this range of values (71-85). Pragmatically I can currently not envision an application which would require the arm to go into these extreme positions (joint #4 is the 'rotational ellbow joint', called l_wrist_prosup and r_wrist_prosup in the yarpmotorgui).

Thanks

randaz81, 2016-04-29:
The range of joint 4 (wrist pronosupination) depends on the mechanical revision number. On ICubGenova01 (forearm reversion V2) limits are set to -60 60. @julijenv

kt10aan, 2016-04-29:
I think that iCubHertfordshire has the forearm reversion v2. In that case, the config files shipped with the robot originally (after the upgrade) were using (and still are) the limits for the older version. It may had to do with the fact that the redesign of the forearm V2 happened while the robot was already in the process of being upgraded in IIT.

frank-foerster, 2016-10-17:
After the repair of the motors (issue #232) I just checked the current limits and they are [-58.8, 60]. As this is close enough to the [-60, 60] I'm assuming this is ok.

closed 2016-10-17 by frank-foerster