robotology/mesh-support #232: Joints no. 5 and 6 broken / non-functional

frank-foerster, 2016-04-28:
(robot: iCubHertfordshire01)

Hi,

this issue is probably linked and/or identical to issues robotology/QA#7 and robotology/QA#19 originally raised by @kt10aan 

When doing the 'joint' test already mentioned in issue #231 (which is most likely a separate issue) I noticed the following issues: HW faults and irrecoverable IDLE modes with joints no. 5 and possibly no. 6 on both arms.

##### Detailed conditions when the failure occurred
I replicated the above issues now a dozen times or so, each time after starting up yarpmotorgui on pc104, and by then subsequently manipulating the joints.

Upon starting up yarpmotorgui I observe the following:

On the right arm, joint no. 5 on is immediately in HW fault, and joint no. 6 is irrecoverably IDLE, meaning, it won't let me put it into POSITION mode via the gui.

On the left arm joints no. 4-7 are initially IDLE upon startup of the yarprobotinterface.
After manually putting all these joints into Position mode (via yarpmotorgui), joint no. 6 goes into HW fault as soon as being manipulated, and joint no. 5 jumps simultaneously into IDLE mode (irrecoverably). The same effect is achieved by just operating joint no. 5, i.e. joint 6 goes into HW fault and joint no. 5 into IDLE.

Joint no. 7, after being put into POSITION mode, is fine and contiues to work.
The same holds true for joint no. 4, apart from the issues mentioned in issue #231.

Could you please advise how to diagnose the precise cause of this problem and how to fix it?

P.S.: Attached is the command line output from the yarprobotinterface. You can see several [ERROR]
messages in there.

[yarprobotinterface280416.txt](https://github.com/robotology/icub-support/files/241119/yarprobotinterface280416.txt)

randaz81, 2016-04-29:
This issue should be handled after closing issue:
https://github.com/robotology/icub-support/issues/232
In fact joint 4, 5 and 6 are controlled by the same MC4 board. If joint 4 raises an overcurrent fault (because it is pushing against the hardware limit), joint 5 and 6 may also experience malfunctioning. 

Overcurrent fault may be also raised by a damaged power amplifier on the mc4 board, a damaged motor winding or a short-circuit somewhere in the arm wiring when a joint is moved. @julijenv

julijenv, 2016-05-08:
Hi @frank-foerster ,
Could you please try to get in touch on skype ( our iD : icub.support) from monday?
please fetch a webcam in order to show me with precision the behaviour of the robot when calibrating.
Fetch some scewrdrivers we may have to open forearm and/or upper arm covers.
I saw that they are quite a bunch of the issues opened for your robot, we will try to close them all because they all look like correlated.

frank-foerster, 2016-05-08:
Yes will do. Thanks.

julijenv, 2016-05-10:
after a long debug on the robot, we agreed to the following checks to be done to detect the failure piece : crimp, M4C board or motor burned

- try to give from 3 to 5V to 2M1 and 2M2 motors (max 1A) to understand if the problem is from the motor (crimp or wires or the winding (core of motor = burned motor)  
- if crimp suspected, try to recrimp (if the resistance in between the 2 pole is not something close to 100ohm)
- if wires suspected try to make a continuity test from end of wire and somewhere in the middle of cable (may need a needle to have an inside of the core of the wire)
- the resistance of the motor should give you an hint (by looking a the R in between of the 2 phase of the motor if ever the value is close to zero (about some ohms) or very high (1M or more) -> motor burned most probably)
- if 2 phases motor pass (motor spin when apply some voltage),now u will need to check crimp of the encoder (the other wires of the motor connector), first -> recrimp if not sure they are alright. Afterwards, make a continuity test in between one of remaining wires and the others (you should find one wire taken as ground from which you should get everytime you take the tension with the other wires something close to 0.6 to 0.7 V.
if everything alright till now, possible explanation for not working proper, is maybe one of the output of the MC4 board is down: 
- try to swap all the connectors from 1 board with another one (2B2 and 2B3) and change their adresses and firmware and make a test to see if the hardware error jump to somwhere else in the forearm not anymore ion the wrist. -> in this case the board is the culprit or the connector R3 or R4.
to test R3 and R4, WATCH OUT you will need to turn on the motor switch and do a voltage test in between one pin of the connector and the others (note the voltage in between every single wires and check that it is the same on the other connector for the same wires tested).

Let me know, your findings

frank-foerster, 2016-05-13:
Here are some pictures that I took this morning of the lower arm.
The first one is just to clarify what I mean with 'left', 'right', and 'top'
[image]

Top 1:

[image]

frank-foerster, 2016-05-13:
Here are some pictures that I took this morning of the lower arm.
The first one is just to clarify what I mean with 'left', 'right', and 'top'
[image]

Github is giving me really hard time adding more than one image to a comment, so I'll have to do it via a series of comments. Sorry for that already in advance.

frank-foerster, 2016-07-01:
Hi Julien (@julijenv)
I finally got around to re-crimp the cable on the control board for the left wrist whose connectivity was broken. (Reminder: on the left wrist the motor was working ok, it's the right one for which we should get the replacement soon).
Yet this didn't fix the issue: joint nr. 5 still goes into HW fault when being moved via GUI and as well when joint nr. 4 is being actuated. So I re-measured connectivity between both connectors 2M1 and 2M2 and their respective endings on the motor end and connectivity seemed to be not the issue. 

I also measured the resistance of the motors, just to be sure. I measured between the 'left-most' pin on the connector (the red cable) and its neighbour (cable nr. 2, if the red one is nr. 1), which are the cables where I had applied the external power supply before in order to test whether the motors still worked or not. The resistance between these two cables for both connector 2M1 and connector 2M2 is roundabout 13 Ohm, if I'm reading the multimeter correctly. 

My next step would now be to try your suggestions from above, that is, to check the encoders (5th point from above in your list of suggestions).
So, just to clarify, here some stupid questions ;):
1. Do the motors / boards need to be powered for me to read the encoder values (= voltage  between certain pins?)
2. How do I determine which one is the ground wire?
3. What do you mean with 'taking the tension'? Do you mean the voltage between two cables / pins? Do I have to manipulate the wrist physically while reading with the multimeter? Or should there be a 'still'/static reading? And again, question 1: I guess the arm needs to be powered for me to be able to get these readings, correct?

Thanks already.

julijenv, 2016-07-05:
@frank-foerster ,
please come on skype it will be simpler for me,

frank-foerster, 2016-07-05:
Hi Julien,
I fixed now the connector for the power supply for board 1B3 (left arm) and it works as before (with the above faults).
When running the yarpmotorinterface I noticed now something I haven't noticed before: This is, presumably, a separate problem from the one with the wrist and may have been caused by somebody in my absence or by me damaging something on the board?
On the **right arm** one of the thumb joints is not working any more (joint nr. 8, 'r_thumb_oppose'). The joint is displayed as working/functional in the yarpmotorgui, but when trying to move it via the slider nothing happens. 
Here is the current position of the thumb after calibration (pointing more 'downwards' than usual):
[image]

For comparison: normally (as is the case on the left hand) the thumb is roughly in the following position:
https://goo.gl/photos/YZhRMWzm5FmjWX8q6

frank-foerster, 2016-07-07:
Hi,

here are the images of the connector of board 1B2 (originally connected to 1B3) which appears to be making trouble:
https://goo.gl/photos/C3dAFuwFTtZ7C3kRA

frank-foerster, 2016-07-18:
Hi, 
as agreed I re-crimped one of the newly crimped cables, where the crimp appeared to be slightly pushed out of 'the socket' once the connector was plugged into the receiving/female part on the board. Evidently the uninsulated metal part which enters the crimp was slightly
too long and was subsequently pushed out of the crimp by the metal pin of the female part of the connector (the 'socket').
After re-crimpiong I re-ran the yarpmotorinterface several times, with switching the cables L3 and L4 between boards 1B2 and 1B3. The results are as follows: 

When L3 is connected 1B2, and L4 to 1B3 only joints 4 to 7 are initially displayed as IDLE (as was the case initially, before one of power supply cables was ripped off and me fixing it), and this seems to be stable. That is, I restarted the yarpmotorinterface alltogether 5 times in this setup and it always ended up being like this.
Note: Prior to re-crimping the mentioned cable, the 'behaviour' was not stable, and joints other than the mentioned ones (identical to ones described below) where shown as IDLE.

When L3 is connected to 1B3 and L4 to 1B2, 50% of the time the behaviour is as above, that is, 
joints joints 5 to 7 are initially displayed as IDLE. But in 50% of cases, when restarting the yarpmotorinterface, additionally to joints 4 to 7, also joints 10, 12, 14, and 15 are shown as IDLE.

Interestingly all of them, apart from joints 5+6, once put into POSITION mode, can be operated. 
This includes the previously problematic joint 4 (l_wrist_prosup), which prior to fixing the cable was sometimes displayed greyed out with 'calibrating' next to it. This did not occur during this round of testing any more in none of the settings, that is, no matter whether L3 was connected to 1B2 and L4 to 1B3 or vice versa.

Please let me know how to proceed - also per Skype if that's better. I'm using this tickets as some form of documentation to keep track of the various changes and debugging attempts.

frank-foerster, 2016-08-26:
Hi @julijenv ,
here are my findings when switching boards 1B2 and 2B2 (and back) and looking whether anything changed in terms of the encoder readings of the wrist joints (joints 5,6,7).

The following table is for the right arm only, I will post after this a table for the same joints on the left arm for comparison. On the right arm I could only move the wrist `diagonally', that is, with the hand being oriented roughly in the vertical plane (i.e. orthogonal to how it would be when you type), I could only move it roughly along the vertical plane but also slightly to the left and right. This is due to the top motor being `stuck' or immovable during these tests. 

I restarted the yarprobotinterface three times (called `Attempts' in the table) to see whether my findings where repeatable / had structure. 

**Right wrist ** (joints 5,6,7) with board _1B2_, i.e. after swapping _1B2_ with _2B2_ (all test movements _diagonal_, _all initial joint  values for the three joints are 0 in all attempt upon restarting the robotinterface)
| Attempt # | joint # | min. value | max. value | Notes |
| :---: | :---: | :---: | :---: | --- |
| 1A | 5 | 0 | 0 | encoder readings are constantly 0 |
| 1A | 6 | 0 | 33.6 | once encoder values reach high values they hardly drop any more (only in single digits) |
| 2A | 4 | (-363) - (-36) | n/a | upon reaching _-36_ values stay pretty static despite moving wrist up and down |
| 2A | 5 | 0 | 0 | encoder values don't change at all during entire attempt | 
| 2A | 6 | (-123) - (-34) | n/a | first movement yields _-34_, further movements decrease value to _-134_ |
| 3A | 4 | n/a | 1,700 | 1,700 is reached during first up-down movement, then remain at this high level and only change minimally in response to further movement |
| 3A | 5 | 0 | 0 | constant encoder value despite any movement |
| 3A | 6 | -3,000 | n/a | -3000 is reached during first movement and only changes minimally during subsequent movements |

frank-foerster, 2016-08-26:
Hi @julijenv ,
here are my findings when switching boards 1B2 and 2B2 (and back) and looking whether anything changed in terms of the encoder readings of the wrist joints (joints 5,6,7).

The following table is for the right arm only, I will post after this a table for the same joints on the left arm for comparison. On the right arm I could only move the wrist 'diagonally', that is, with the hand being oriented roughly in the vertical plane (i.e. orthogonal to how it would be when you type), I could only move it roughly along the vertical plane but also slightly to the left and right. This is due to the top motor being 'stuck' or immovable during these tests. 

I restarted the yarprobotinterface three times (called 'Attempts' in the table) to see whether my findings where repeatable / had structure. 

**Right wrist ** (joints 5,6,7) with board _1B2_, i.e. after swapping _1B2_ with _2B2_ (all test movements _diagonal_, _all initial joint  values for the three joints are 0 in all attempt upon restarting the robotinterface)

| Attempt # | joint # | min. value | max. value | Notes |
| :---: | :---: | :---: | :---: | --- |
| 1A | 5 | 0 | 0 | encoder readings are constantly 0 |
| 1A | 6 | 0 | 33.6 | once encoder values reach high values they hardly drop any more (only in single digits) |
| 2A | 4 | (-363) - (-36) | n/a | upon reaching _-36_ values stay pretty static despite moving wrist up and down |
| 2A | 5 | 0 | 0 | encoder values don't change at all during entire attempt | 
| 2A | 6 | (-123) - (-34) | n/a | first movement yields _-34_, further movements decrease value to _-134_ |
| 3A | 4 | n/a | 1,700 | 1,700 is reached during first up-down movement, then remain at this high level and only change minimally in response to further movement |
| 3A | 5 | 0 | 0 | constant encoder value despite any movement |
| 3A | 6 | -3,000 | n/a | -3000 is reached during first movement and only changes minimally during subsequent movements |

frank-foerster, 2016-08-26:
Hi @julijenv ,
here are my findings when switching boards 1B2 and 2B2 (and back) and looking whether anything changed in terms of the encoder readings of the wrist joints (joints 5,6,7).

The following table is for the right arm only, I will post after this a table for the same joints on the left arm for comparison. On the right arm I could only move the wrist ‘diagonally', that is, with the hand being oriented roughly in the vertical plane (i.e. orthogonal to how it would be when you type), I could only move it roughly along the vertical plane but also slightly to the left and right. This is due to the top motor being `stuck' or immovable during these tests. 

I restarted the yarprobotinterface three times (called `Attempts' in the table) to see whether my findings where repeatable / had structure. 

**Right wrist** (joints 5,6,7) with board _1B2_, i.e. after swapping _1B2_ with _2B2_ (all test movements _diagonal_, _all initial joint  values for the three joints are 0 in all attempt upon restarting the robotinterface)
| Attempt # | joint # | min. value | max. value | Notes |
| :---: | :---: | :---: | :---: | --- |
| 1A | 5 | 0 | 0 | encoder readings are constantly 0 |
| 1A | 6 | 0 | 33.6 | once encoder values reach high values they hardly drop any more (only in single digits) |
| 2A | 4 | (-363) - (-36) | n/a | upon reaching _-36_ values stay pretty static despite moving wrist up and down |
| 2A | 5 | 0 | 0 | encoder values don't change at all during entire attempt | 
| 2A | 6 | (-123) - (-34) | n/a | first movement yields _-34_, further movements decrease value to _-134_ |
| 3A | 4 | n/a | 1,700 | 1,700 is reached during first up-down movement, then remain at this high level and only change minimally in response to further movement |
| 3A | 5 | 0 | 0 | constant encoder value despite any movement |
| 3A | 6 | -3,000 | n/a | -3000 is reached during first movement and only changes minimally during subsequent movements |

frank-foerster, 2016-08-26:
Hi @julijenv ,
here are my findings when switching boards 1B2 and 2B2 (and back) and looking whether anything changed in terms of the encoder readings of the wrist joints (joints 5,6,7).

The following table is for the right arm only, I will post after this a table for the same joints on the left arm for comparison. On the right arm I could only move the wrist ‘diagonally', that is, with the hand being oriented roughly in the vertical plane (i.e. orthogonal to how it would be when you type), I could only move it roughly along the vertical plane but also slightly to the left and right. This is due to the top motor being `stuck' or immovable during these tests. 

I restarted the yarprobotinterface three times (called `Attempts' in the table) to see whether my findings where repeatable / had structure. 

**Right wrist** (joints 5,6,7) with board _1B2_, i.e. after swapping _1B2_ with _2B2_ (all test movements _diagonal_, all initial joint  values for the three joints are 0 in all attempt upon restarting the robotinterface)

| Attempt # | joint # | min. value | max. value | Notes |
| :---: | :---: | :---: | :---: | --- |
| 1A | 5 | 0 | 0 | encoder readings are constantly 0 |
| 1A | 6 | 0 | 33.6 | once encoder values reach high values they hardly drop any more (only in single digits) |
| 2A | 4 | (-363) - (-36) | n/a | upon reaching _-36_ values stay pretty static despite moving wrist up and down |
| 2A | 5 | 0 | 0 | encoder values don't change at all during entire attempt | 
| 2A | 6 | (-123) - (-34) | n/a | first movement yields _-34_, further movements decrease value to _-134_ |
| 3A | 4 | n/a | 1,700 | 1,700 is reached during first up-down movement, then remain at this high level and only change minimally in response to further movement |
| 3A | 5 | 0 | 0 | constant encoder value despite any movement |
| 3A | 6 | -3,000 | n/a | -3000 is reached during first movement and only changes minimally during subsequent movements |

                  _Now swapping 1B2 for 2B2_ (i.e. back to original setting)
| Attempt# | joint# | min. value | max. value | Notes |
| :---: | :---: | :---: | :---: | --- |
| 1B | 4 | (-1,790) - (-2,100) | n/a | first movement upwards leads to _-1,790_, down movment does _NOT_ lead to any decrease of the value. 2nd move up pushes the value to _-2,100_ where it remains during subsequent movements |
| 1B | 5 | 0 | 3 | during first movement value stays _0_ and only during later movements reaches _3_ |
| 1B | 6 | -1 | 3 | hardly any modulation during movements | 
| 2B | 4 | -2,900 | 5,354 | First attempt of moving wrist up (wrist stuck) leads to _-2,900_. After wrist becoming losse / moveable value goes up to 5,354 during _DOWN_ movement. Subsequent movements hardly impact upon value |
| 2B | 5 | 0 | 0 | value const. 0 |
| 2B | 6 | 43 | 106 | _43_ is reached during 1st movement, _106_ during 2nd movement. Subsequent movement cause hardly any change (only in single digits). |

_I will post table for left wrist soon .._

frank-foerster, 2016-08-26:
**Left Wrist** with board _2B2_ (joints 4,5,6)
The major difference to the right wrist is that joints 5 + 6 appear to be strongly coupled, often times values are identical. Also on this wrist diagonal movement was possible most times which pushed the values up into the dozen of thousands. Note that what I call here diagonal movement seemed to involve only the bottom motor, whereas up-down movements appeared to involve only the top wrist motor. 

| Attempt# | mov. type | joint# | min. | max. | Notes |
| :---: | :---: | :---: | :---: | :---: | --- |
| 1A | ? | 4 | 2 | 2 | const. value |
| 1A | ? | 5+6 | n/a | 2,000s - 11,000s | jump to 2,000s happens with 1st move, subsequent movements drive values up to the 11,000s. Enc. values hardly ever decrease. As mentioned above enc. values for joints 5 + 6 appear to be strongly coupled, here even identical |
| 2A | up-down | 4 | 0 | 0 | const. value |
| 2A | up-down | 5+6 | -109 | 260 | 1st move yields _24, 2nd move --> 260, then down to -104, 3rd move --> -109, 3rd move up --> -54 |
| 2A | diag | 5+6 | n/a | 20,000s | Diag. movements drive value into the _20,000s_ |
| 3A | up-down | 4 | -1 | 0 | ~ const. value | 
| 3A | up-down | 5+6 | -2 | 360 | 1st move up --> _-2_, 1st move down --> _+19_, 2nd move up --> _15_, 2nd move down --> _116_, 3rd up --> _107_, 3rd down --> _360_ |
| 3A | diag | 4 | 3,869 | 3,930 | 1st move --> _3,930_, down to _3,869_, constant thereafter |
| 3A | diag | 5+6 | -37,000 | 9,413 | down-move --> _9,413_, up --> _-37,000_, subsquent movements modulate enc. values roughly within this range |

                     Now swapping boards 2B2 <--> 1B2 back to original setup

| Attempt# | mov. type | joint# | min. | max. | Notes |
| :---: | :---: | :---: | :---: | :---: | --- |
| 1B | up-down | 4 | 0 | 0 | const. value |
| 1B | up-down | 5+6 | 4 | 341 | initial enc. values are 4, 1st move up --> _no change_, 2st move down --> _87_, 2nd move up --> _no change_, 2nd move down --> _341_|
| 1B | diag | 4 | -318 | -308 | 3rd move down --> _-318_, 3rd move up --> _-308_, 4th move down --> _-336_, 4th move up --> _no change_ | 
| 1B | diag | 5+6 | 12,108 | 38,477 | 3rd move down --> _12,108_, 3rd move up --> _36,720_, 4th move down --> _12,290_, 4th move up --> _38,477_ |
| 2B | up-down | 4 | -1 | 0 | ~ const. value |
| 2B | up-down | 5+6 | 4 | 813 | initial enc. values = _86_, 1st move up --> _4_, 1st move down --> _726_, 2nd move up --> _460_, 2nd move down _813_|
| 2B | diag | 4 | -6,529 | n/a | 3rd move up --> -6,529, enc. values const. thereafter |
| 2B | diag | 5+6 | 2,834 | -47,962 | 3rd move up --> _-47,962_, 3rd move down --> 2,834 |

frank-foerster, 2016-08-26:
A little update from the debugging front   (@julijenv)

Joint 4 on both arms, which you pointed out is not really a wrist joint but, what I would call, rotational joint for the lower arm, works fine on both arms. Now that I know, which joint it is, I moved both lower arms around their own axis and the encoder values change in an apparently meaningful manner.

Regarding the strange coupling of joints 5 + 6 on the left arm: I made small video, so you can see for yourself. If I move the wrist diagonally, which moves the bottom motor only, I get very large encoder readings for both joints. If I move the wrist strictly up-and-down, which moves the top-motor only, I get smaller but nevertheless identical readings for both joints. 

[video of wrist movement and encoder readings](https://vimeo.com/180314707)

If this is not supposed to be the case, as you indicated, what kind of error does that imply?

(But as said, for the moment I'd like to concentrate on the right wrist, so better to treat this as a side note).

Continuing to debug ...

frank-foerster, 2016-10-15:
After the replacement of two wrist/lower arm motors by @julijenv all wrist joints ork again. There's just some smaller issue with joint 10 on the left arm (thumb flex), but I'll describe this in a separate issue.

closed 2016-10-15 by frank-foerster