robotology/mesh-support #233: icub24 laptop has frequent window manager problems

traversaro, 2016-04-29:
##### Description of the failure
We have continuous problems on the iCubGenova02 setup due to the Window Manager (Unity?), see for example how I found the robot laptop today: 
[image]

Given that by chatting with other users this is not the first Window Manager-related problem, can we switch to use by default some more stable and usable window manager (XFCE?)? 

@mbrunettini @drdanz

mbrunettini, 2016-04-29:
I suspect it's not related to the Window Manager, but to the nVidia video drivers. There are other issue, that will be fixed upgrading to Ubuntu 16.04. 
I will switch the default vm to gnome, anyway.

DanielePucci, 2016-04-29:
As @lornat75 pointed out in https://github.com/robotology/icub-support/issues/215#issuecomment-207087477 it is better to leave this kind of issues open till the upgrade.

traversaro, 2016-04-29:
I think that this was fixed by switching the default WM to Metacity.

closed 2016-04-29 by traversaro

DanielePucci, 2016-04-29:
👍

drdanz, 2016-04-29:
You know my answer.

traversaro, 2016-05-03:
Note: to correctly use Qt5 YARP GUIs menus on Metacity on Trusty, it is necessary to remove the appmenu-qt5 package, see https://bugs.launchpad.net/ubuntu/+source/appmenu-qt5/+bug/1307619 for more info.

gsaponaro, 2016-05-03:
/cc https://github.com/robotology/QA/issues/41