robotology/mesh-support #234: Right camera doesn't work!

towardthesea, 2016-04-29:
Please fill in the following two sections:

##### Description of the failure
- yarpdev can connect properly to the right camera and feed something to the yarpview, i.e. the port feature of yarpview shows ~30 fps,  the display feature of yarpview shows ~ 10 fps but no any (static) image is displayed

##### Detailed conditions when the failure occurred
- Firstly, we ran the 'Calib_Camera' application on yarpmanager properly on the icub laptop, it worked fine with both camera, i.e there are video from both camera shown on both yarpviews. Then, the same thing happened with the 'iCubStartup' application. 
- Then we ran 'WYSIWYD_IIT' application on yarpmanager, every connections showed that they seemed to work fine but they could not recognized objects well, so we decided to turn off this application and 'Calib_Camera' and run them again (This was not the first time, but other times everything had come back and worked fine after restart)
- After restart the 'Calib_Camera' application, we found that we lost video from the right camera as description above.

**Remark**: if you are not a daily icub-support user (e.g. newbie), please read carefully the guidelines for contributing linked at the top of this page. In particular, the name of your robot must be always indicated: either using labels if you're already a contributor, as per standard, or by specifying it while detailing the issue.

pattacini, 2016-05-02:
@julijenv @spada1978 

It's very likely that the problem is caused by the flexible firewire connector. Given that in the recent past @vtikha already attempted to fix it up but the issue showed up again, it'd be maybe worth inspecting it more carefully.

Unfortunately, tomorrow we'll have a demo with Blacky, so can you check it today?

vtikha, 2016-05-02:
@julijenv @spada1978 @pattacini 
The issue here is the flat cable coming out of the flip lock located behind the eye. As the right eye moves a lot during calibration after a while it comes out of its housing. The quickest test is to replace the flat cable as its end might have been slightly damaged over time. This would indicate if the problem is with the flat or the flip lock.

vtikha, 2016-05-02:
Flat cable replaced. It seems to be much tighter in the flip lock. Let's leave this issue opened in case it happens again and take it further.

vtikha, 2016-05-04:
The replacement of the flat cable has solved the problem.

closed 2016-05-04 by vtikha