robotology/mesh-support #235: RobotInterface not running

jypuigbo, 2016-04-29:
##### Description of the failure

After updating and compiling `pc104` (see details below), `robotInterface` fails with the errors attached in the [log.txt](https://github.com/robotology/icub-support/files/242793/log.txt) file. 
We have followed all the steps in [here](https://github.com/robotology/yarp/issues/498) and we already requested the changes in [here](https://github.com/robotology/icub-main/issues/273) (although the last shouldn't impede running `robotInterface`, as far as we know. 

Detail of the error: 
```
459		2676,016369		INFO		created wrapper <controlboardwrapper2>. See C++ class yarp::dev::ControlBoardWrapper2 for documentation.
460		2676,046544		ERROR		AxisName not found
461		2676,056646		ERROR		AxisType not found
462		2676,056746		INFO		Position Pids section found, new format
463		2676,056797		ERROR		maxOutput not found
464		2676,056852		ERROR		Position Pids section: error detected in parameters syntax 
465		2676,056893		ERROR		One or more errors found while parsing device configuration 
466		2676,056933				yarpdev: ***ERROR*** driver <canmotioncontrol> was found but could not open
467		2676,056976		WARNING		Cannot open device left_arm_mc 
468		2676,057019		WARNING		Cannot open device left_arm_mc 
469		2676,057062		ERROR		AxisName not found
470		2676,057103		ERROR		AxisType not found
471		2676,057145		INFO		Position Pids section found, new format
472		2676,057182		ERROR		maxOutput not found
473		2676,057220		ERROR		Position Pids section: error detected in parameters syntax 
474		2676,057258		ERROR		One or more errors found while parsing device configuration 
475		2676,057296				yarpdev: ***ERROR*** driver <canmotioncontrol> was found but could not open
476		2676,057336		WARNING		Cannot open device left_hand_mc 
477		2676,066751		WARNING		Cannot open device left_hand_mc 
478		2676,066846		ERROR		AxisName not found
479		2676,066894		ERROR		AxisType not found
480		2676,066934		INFO		Position Pids section found, new format
481		2676,066973		ERROR		maxOutput not found
482		2676,067011		ERROR		Position Pids section: error detected in parameters syntax 
483		2676,067051		ERROR		One or more errors found while parsing device configuration 
484		2676,076865				yarpdev: ***ERROR*** driver <canmotioncontrol> was found but could not open
485		2676,076996		WARNING		Cannot open device right_arm_mc 
486		2676,077065		WARNING		Cannot open device right_arm_mc 
487		2676,077117		ERROR		AxisName not found
488		2676,077169		ERROR		AxisType not found
489		2676,077210		INFO		Position Pids section found, new format
490		2676,077249		ERROR		maxOutput not found
491		2676,077294		ERROR		Position Pids section: error detected in parameters syntax 
492		2676,077337		ERROR		One or more errors found while parsing device configuration 
493		2676,086959				yarpdev: ***ERROR*** driver <canmotioncontrol> was found but could not open
494		2676,087040		WARNING		Cannot open device right_hand_mc 
495		2676,087084		WARNING		Cannot open device right_hand_mc 
496		2676,087121		ERROR		AxisName not found
497		2676,087154		ERROR		AxisType not found
498		2676,087185		INFO		Position Pids section found, new format
499		2676,087216		INFO		Position Pids successfully loaded
500		2676,087249		WARNING		Torque control parameters not found for part HEAD, skipping...
501		2676,087285		DEBUG		DEBUG parameters section NOT found, skipping...
502		2676,087322		DEBUG		IMPEDANCE parameters section NOT found, skipping...
503		2676,087353		ERROR		motorOverloadCurrents not found
504		2676,087383		ERROR		One or more errors found while parsing device configuration 
505		2676,087417				yarpdev: ***ERROR*** driver <canmotioncontrol> was found but could not open
506		2676,087450		WARNING		Cannot open device head_mc 
507		2676,087480		WARNING		Cannot open device head_mc 
508		2676,097078		ERROR		AxisName not found
509		2676,097184		ERROR		AxisType not found
510		2676,097223		INFO		Position Pids section found, new format
511		2676,097257		ERROR		maxOutput not found
512		2676,097288		ERROR		Position Pids section: error detected in parameters syntax 
513		2676,097320		ERROR		One or more errors found while parsing device configuration 
514		2676,097352				yarpdev: ***ERROR*** driver <canmotioncontrol> was found but could not open

```
##### Detailed conditions when the failure occurred
+ We freshly made a `git pull` the same day and time of: 
 + `yarp`
 + `icub-main`
 + `icub-firmware-shared`
 + `icub-firmware-build`
+ We removed all the build folders
+ We ccmaked `yarp` with the options in [here](http://wiki.icub.org/wiki/Compilation_on_the_pc104) and compiled successfully. 
+ We run the script `updateRobot.sh` without errors
+ We cmaked and compiled `icub-firmware-shared` without errors
+ We cmaked and compiled `icub-main` as detailed [here](http://wiki.icub.org/wiki/Compilation_on_the_pc104), without errors, **but** the following options were not available in our installation, so they were not enabled: 
 + `ENABLE_icubmod_logpolarclient`
 + `ENABLE_icubmod_logpolargrabber`
 + `ENABLE_icubmod_icubheadcalibratorV2`

We still tried to run robotInterface, but it failed with the log attached. 

We also tried different icub-main cmake options that made sense, but they didn't work either

Last time we had a similar problem, I remember @randaz81 solved it, but I'm not sure it was the same. 

@clement-moulin-frier  @sockchinglow @pattacini

randaz81, 2016-05-02:
Hi @jypuigbo, I just gave a look to the files in the repo and they are ok. 
In fact the problem occurs because, according to your log, roboInterface is currently using the configuration stored in:
`/home/icub/.local/share/yarp/robots/iCubBarcelona01/./hardware/motorControl/icub_left_arm.xml` 
and that folder is keeping a **not updated copy** of the robot configuration.

clement-moulin-frier, 2016-05-03:
Thanks @randaz81 , it was indeed a problem of a mismatch between what was is in our `.local` vs in the repository. So we removed all the files of our `.local` that didn't contain local modifications.
However, our `.local` contains a special config file called `icub_no_legs.xml` that replaces the standard `icub_all.xml`, so that the legs are deactivated. When reading this file, it seems that `robotInterface` looks for the files mentioned in `icub_no_legs.xml` (e.g. `left_arm_cartesian.xml `, see error log below) in the `.local` instead of the standard `$YARP_DATA_DIRS$ and therefore doesn't find them. This provides the error copied below.

Is it possible that, when reading our `icub_no_legs.xml`, `robotInterface` looks for files in the standard `$YARP_DATA_DIRS` instead of looking in the `.local`? (i.e. the folder where `icub_no_legs.xml` is located.)
Another solution would be to push our configuration files to the repository, but we guess it is not recommended, is it? A last solution is that we copy all our config files in the `.local`, but this was the source of the first error we had.

Thanks for the help,

LOG:
```
||| configuring
||| default config file specified as robotInterface.ini
||| checking [/home/icub/.local/share/yarp/robots/iCubBarcelona01/robotInterface.ini] (pwd)
||| found /home/icub/.local/share/yarp/robots/iCubBarcelona01/robotInterface.ini
||| finding file [config]
||| checking [/home/icub/.local/share/yarp/robots/iCubBarcelona01/./icub_no_legs.xml] (pwd)
||| found /home/icub/.local/share/yarp/robots/iCubBarcelona01/./icub_no_legs.xml
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubBarcelona01/./icub_no_legs.xml 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubBarcelona01/./icub_no_legs.xml at line 1 . No DTD found. Assuming version robotInterfaceV1.0 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubBarcelona01/./icub_no_legs.xml at line 2 . "robot" element should contain the "build" attribute [unsigned int]. Assuming 0 
[WARNING]Invalid syntax while loading /home/icub/.local/share/yarp/robots/iCubBarcelona01/./icub_no_legs.xml at line 2 . "robot" element should contain the "portprefix" attribute. Using "name" attribute 
[DEBUG]Reading file /home/icub/.local/share/yarp/robots/iCubBarcelona01/./cartesian/left_arm_cartesian.xml 
[FATAL]Syntax error while loading /home/icub/.local/share/yarp/robots/iCubBarcelona01/./cartesian/left_arm_cartesian.xml at line 0 . Failed to open file 
robotInterface(_ZN4yarp2os9LogStreamD1Ev+0x459) [0x4145b9]
robotInterface(_ZN14RobotInterface9XMLReader7Private15readDevicesFileERKSs+0x1e2) [0x430912]
robotInterface(_ZN14RobotInterface9XMLReader7Private14readDevicesTagEP12TiXmlElement+0x1b2) [0x431322]
robotInterface(_ZN14RobotInterface9XMLReader7Private11readDevicesEP12TiXmlElement+0xfe) [0x431b1e]
robotInterface(_ZN14RobotInterface9XMLReader7Private12readRobotTagEP12TiXmlElement+0x371) [0x432021]
robotInterface(_ZN14RobotInterface9XMLReader7Private13readRobotFileERKSs+0x59c) [0x43291c]
robotInterface(_ZN14RobotInterface6Module9configureERN4yarp2os14ResourceFinderE+0x4fe) [0x41c14e]
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_OS.so.1(_ZN4yarp2os8RFModule9runModuleERNS0_14ResourceFinderE+0xa) [0x7f1cdada1eda]
robotInterface(main+0xdc) [0x413c7c]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xfd) [0x7f1cd9b5bead]
robotInterface() [0x413fd9]
```

pattacini, 2016-05-03:
@clement-moulin-frier 

As per this [question](https://github.com/robotology/QA/issues/42#issuecomment-148192928) and its relative [answer](https://github.com/robotology/QA/issues/42#issuecomment-148714718), you cannot launch **`yarprobotinterface`** with mixed robot context, where you have some files in `.local` and the rest in the location pointed by `$YARP_DATA_DIRS`.

You **must** copy the whole set of files in your `.local` and keep it up-to-date with latest changes in the repository. As of now, it's the only way to proceed, given that it's quite likely to tune even a small bit of the content of those robot files.

pattacini, 2016-05-03:
More on this, @clement-moulin-frier 

You ought to always rely on **`yarp-config robot`** utility to copy out robot files in `.local` rather than doing that manually via e.g. `cp` command. This way you could have realized that it is forbidden to copy single files.

pattacini, 2016-05-06:
Is everything fine now?

pattacini, 2016-05-09:
Assuming is all fine now, closing for the time being.
Feel free to reopen it.

closed 2016-05-09 by pattacini

clement-moulin-frier, 2016-05-09:
Yes it's fine now, thanks a lot and sorry for the late reply