robotology/mesh-support #236: right_foot FT sensor (ID 170) has one calibration matrix value set to FFFF

traversaro, 2016-05-05:
##### Description of the failure
A value on the calibration matrix in the FT sensor mounted on the right foot has a wrong value of FFFF . 

##### Detailed conditions when the failure occurred

We noticed that the FT sensor mount in the right ankle of `iCubDarmstdadt01` was returning strange results. In particular, we noticed that the force measurement on the y axis seemed to have the wrong sign. We investigated and apparently there was a wrong parameter in the calibration matrix:
[image]

In particular this value was exactly on the second row, i.e. relative to the measurement of the Y force! 
We then checked this values agains the calibration matrix of this sensor stored in https://svn.icub.iit.local/repos/mecha/ftSensCalib/trunk/ , and we notice that indeed the right value for that element was `4064`, as was possible to see in the `matrix_SN170.txt` file: 
~~~
54
8D
3D63
B9
A
BF11
FF17
4064
E379
FD80
A3
E100
2A05
57
FEE0
29C3
29AB
FEF2
1AE
1B68
F78F
DAF0
242F
EF6B
D65A
FFBA
EA3D
139C
1319
159B
CA
29AB
2565
1CE
FE65
26AD
1
1211
1468
1722
29
30
20
~~~

Changing the calibration matrix element back to `4064` seems to fix the sensor readings (at least the sign seem correct). 

##### Discussion 
What could be the cause of this change in the value of the element of the calibration matrix? 
Background: in our first visit to Darmstadt, we had to change the DAC offset value a lot on this robot to avoid overflow in the `YOGA++`/`TaiChi` demo and unfortunatly it is not possible to change those values on an ETH robot without rebooting the boards (to use the canLoader you need to put the EMS in "discover" mode, and for using them again with the robot you need to reset them). Consequenly we had to write the offset to EEPROM a relativly big number of times. 
I wonder if this `FFFF` are just values that are uncorrectly readed by the EEPROM because it is consuming all its available write cycles. Is this possible? @randaz81 @valegagge

DanielePucci, 2016-05-05:
Here's the `canLoader` screenshot associated with the sensor SN233, which has similar problems.

https://www.dropbox.com/s/cfmb0hv6wdkw1x5/Screenshot%202016-05-05%2018.55.58.png?dl=0

DanielePucci, 2016-05-05:
While here's what the calibration matrix should be, (data taken from `matrix_SN233.txt`)

~~~
10F
FF70
3F6F
FC99
29
C4A8
10
3CD8
DEDC
1A4
FEF2
E0D2
2A31
150
20C
28B5
2915
9F
E2
1B68
EF02
DE26
2228
F314
D7E8
FEEF
E8D0
131D
11D8
1751
FF5D
2783
288A
4B
F
2BDB
1
1505
1668
1954
33
35
21
~~~

traversaro, 2016-05-06:
cc @fjandrad

traversaro, 2016-05-06:
cc @julijenv @nunoguedelha @francesco-romano

traversaro, 2016-06-07:
On `iCubDarmstad01` we fixed this problem by reuploading the right calibration matrix (following http://wiki.icub.org/wiki/Firmware), but the problem could be occuring in random robots and it is quite difficult to debug. We should add a check for it at some level, either the `STRAIN` board, or the `EMS` or the `embObjAnalogSensor` device.

traversaro, 2016-06-07:
Closing for now because the issue was solved, see https://github.com/robotology/icub-firmware/issues/68 for the evolution of this issue.

closed 2016-06-07 by traversaro