robotology/mesh-support #238: Errors after porting to yarprobotinterface, cannot find "embObjStrain" plugin

hu-yue, 2016-05-09:
##### Description of the failure
There are several errors when launching `yarprobotinterface`, the robot cannot strart, for details please see log [here](http://orb.iwr.uni-heidelberg.de/ftp/icub_logs/ypmrprunlog_09_05_2016_01_15_48.log).

##### Detailed conditions when the failure occurred
Everything has been updated, yarp, icub-main, firmware (only EMS, since it seems that the CAN were all on the last version already). After everything have been compiled and new firmware loaded, when launching the `yarprobotinterface` (`robotinterface.ini` has been renamed) I get the errors in the log and the robot won't start.

I've seen that in the latest `icub-main` there were some new options that were set to `OFF` by default, like `ENABLE_icubmod_embObjStrain`, so I did not change them. But from the errors it seems it's complaining about it. Should I do something about this?

hu-yue, 2016-05-09:
Ok, so it seems that the problem is solved by setting the flag `ENABLE_icubmod_embObjStrain` to `ON`. Or at least, the robot has started.

julijenv, 2016-05-09:
Could you please send one of the log.file to see all the error message disappeared completely.
thx in advance.
If yes i will close the issue.
;)

traversaro, 2016-05-09:
Relevant issue : https://github.com/robotology/icub-main/issues/331 .

hu-yue, 2016-05-09:
[Here](http://orb.iwr.uni-heidelberg.de/ftp/icub_logs/log_pc104_yarprobotinterface_3156.txt) is the new log without errors.

traversaro, 2018-01-10:
@hu-yue  I guess you solved the problem long ago (based on https://github.com/robotology/icub-support/issues/504), so I am closing this issue. If necessary please reopen.

closed 2018-01-10 by traversaro