robotology/mesh-support #239: Ethernet, Invalid configuration syntax issues and too many warnings

dogoepp, 2016-05-09:
After #230 was fixed, we faced two kinds of errors and many warnings. Our errors relate to communication with the Ethernet boards and the configuration file apparently in the wrong version. 

Bellow are log excerpts related to the reported errors. The related log files are attached to this issue.

We are confused about the many warnings and the Ethernet issue. Even more, we are concerned that the error about the configuration file might cause the robot to be dangerous for itself if it was to do the calibration.

Thanks for your help

##### 1st attempt
```
659		70,602633		ERROR		EthResource::verifyBoardPresence() DID NOT have replies from BOARD EB1 with IP 10.0.1.1  even after  20  attempts and 10.0116 seconds: CANNOT PROCEED ANY FURTHER 
495		34,784168		ERROR		EthResource::verifyBoardPresence() DID NOT have replies from BOARD EB2 with IP 10.0.1.2  even after  20  attempts and 10.0126 seconds: CANNOT PROCEED ANY FURTHER 
547		46,608674		ERROR		EthResource::verifyEPprotocol() cannot verify BOARD EB3 with IP 10.0.1.3 : cannot proceed any further 
594		58,408800		ERROR		EthResource::verifyBoardPresence() DID NOT have replies from BOARD EB4 with IP 10.0.1.4  even after  20  attempts and 10.0121 seconds: CANNOT PROCEED ANY FURTHER 
396		11,140075		ERROR		EthResource::verifyBoardPresence() DID NOT have replies from BOARD EB5 with IP 10.0.1.5  even after  20  attempts and 10.0129 seconds: CANNOT PROCEED ANY FURTHER 
```
```
50		0,050367		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNancy01/./general.xml at line 1 . No DTD found. Assuming version robotInterfaceV1.0 
48		0,050350		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNancy01/./hardware/motorControl/left_lower_arm-ems2-mc.xml at line 1 . No DTD found. Assuming version robotInterfaceV1.0 
```

##### 2nd attempt
> Related file: [log20160509-2.txt](https://github.com/robotology/icub-support/files/255235/log20160509-2.txt)

```
1804		385,842417		ERROR		embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD EB2 IP 10.0.1.2 for nv ID32 = 0x0100000c -> IND = 0, TAG = eoprot_tag_mc_joint_status_core_modes_controlmodestatus 
1806		385,842519		ERROR		Failure of askRemoteValue() inside embObjMotionControl::doneRaw(axis=0) for BOARD EB2 IP 10.0.1.2
1811		385,852518		ERROR		Received an answer message nobody is waiting for (eoThreadFifo::pop) 
1812		385,852584		ERROR		Received an answer message nobody is waiting for (MCmutex_post) 
1850		392,029563		ERROR		Left_Leg_Calibrator : Timeout while calibrating  0 
1865		392,160851		ERROR		embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD EB2 IP 10.0.1.2 for nv ID32 = 0x0100000c -> IND = 0, TAG = eoprot_tag_mc_joint_status_core_modes_controlmodestatus 
1866		392,160926		ERROR		Failure of askRemoteValue() inside embObjMotionControl::doneRaw(axis=0) for BOARD EB2 IP 10.0.1.2
1867		392,272053		ERROR		embObjMotionControl::askRemoteValue() timed out the wait of reply from BOARD EB4 IP 10.0.1.4 for nv ID32 = 0x0100000c -> IND = 0, TAG = eoprot_tag_mc_joint_status_core_modes_controlmodestatus 
1868		392,272106		ERROR		Failure of askRemoteValue() inside embObjMotionControl::doneRaw(axis=0) for BOARD EB4 IP 10.0.1.4
```

##### 3rd attempt
> Related file: [log20160509-3.txt](https://github.com/robotology/icub-support/files/255234/log20160509-3.txt)

```
2488		628,589291		ERROR		  FATAL: EthResource::verifyEPprotocol() had a timeout of 0.1 secs when asking the endpoint descriptors to BOARD EB1 with IP 10.0.1.1 : cannot proceed any further 
2448		627,740273		ERROR		  FATAL: EthResource::verifyEPprotocol() had a timeout of 0.1 secs when asking the endpoint descriptors to BOARD EB5 with IP 10.0.1.5 : cannot proceed any further 
2586		630,498988		ERROR		  FATAL: EthResource::verifyRemoteValue() DID NOT have replies from BOARD EB2 with IP 10.0.1.2  even after  2  attempts and 0.101136 seconds: CANNOT PROCEED ANY FURTHER 
3550		648,414149		ERROR		         EthResource::verifyEPprotocol() asks: can you ping the board? if so, is the MN protocol version of BOARD equal to ( 2 9 )? if not, perform FW upgrade. if so, was the ropframe transmitted in time? 
4467		653,373655		ERROR		 from BOARD 10.0.1.7 (EB7), src LOCAL, adr 0, time 557s 259m 717u: (code 0x06000003, par16 0x0001 par64 0x000000000000e2d3) -> ETH monitor: the board low level ETH detected a missing ropframe w/ expected sequence number in par64 and number of detected in par16 + .
```

##### 4th attempt
> Related file: [log20160509-4.txt](https://github.com/robotology/icub-support/files/255236/log20160509-4.txt)

This time, we did not have any errors but there are plenty of warnings on the files, as for instance:

```
4638		1059,021034		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubNancy01/./cartesian/left_arm-cartesian.xml at line 1 . "devices" element should contain the "robot" attribute 
```

also some timeouts on two boards

```
5251		1071,240980		WARNING		 from BOARD 10.0.1.2 (EB2), src LOCAL, adr 0, time 98s 111m 241u: (code 0x0000000d, par16 0x0224 par64 0x07000011008c0003) -> SYS: the TX phase of the control loop has last more than wanted. In par16: TX execution time [usec]. In par64: num of tx can2 and can1 frames and latest previous execution times of TX, RX, DO [usec] + .
5288		1072,668286		WARNING		 from BOARD 10.0.1.4 (EB4), src LOCAL, adr 0, time 99s 542m 147u: (code 0x0000000d, par16 0x01c4 par64 0x0700001000340002) -> SYS: the TX phase of the control loop has last more than wanted. In par16: TX execution time [usec]. In par64: num of tx can2 and can1 frames and latest previous execution times of TX, RX, DO [usec] + .
```

And a weird warning about the skin

```
5263		1072,101292		WARNING		CanBusSkin id list contains more than one entry -> devices will be merged.
```

Tobias-Fischer, 2016-05-12:
Regarding the `invalid syntax` warning: The fix for this can be found in https://github.com/robotology/icub-main/commit/f9bb5c091139c1158fcc79652f827e8f1daf7546

Regarding the `CanBusSkin` warning: I don't know how to fix this, but would like to know ;).

Regarding everything else: I have no clue ..

misaki43, 2016-11-10:
In the mean time, we have change a lot of things. We don't know what was the problem but the issue is solved for a long time.

Thanks @Tobias-Fischer

closed 2016-11-10 by misaki43