robotology/mesh-support #240: Frequent HW fault on head

francesco-romano, 2016-05-10:
Very frequently the head goes on hw fault.
This happens when during our balancing demos, while putting the robot in home position, while calibrating the sensors, etc.
At the beginning we tought about a temperature problem, but we have just turned on the robot and the motors are cold.
We also tought about a "real" overcurrent problem, so we performed the demo with the head in idle, and after it, as soon as we put the head back in position, we got the HW fault.

Needless to say that this issue is quite important for all the demos, not only for ours.

Could be related to https://github.com/robotology/icub-support/issues/75 (even if we did not decreased the PWM limit as we already know this did not solve the issue on the other robots)

This is the log when the fault happens.
```
6142		984.298916		DEBUG		cfw2can [0] msg from board 1: PWM DIS CH1 
6144		984.298949		DEBUG		cfw2can [0] msg from board 1: PWM DIS COUPLED:1 
6145		984.298965		DEBUG		cfw2can [0] msg from board 1: PWM DIS CH0 
6146		984.298981		DEBUG		cfw2can [0] msg from board 1: PWM DIS CH1 
6147		984.298996		DEBUG		cfw2can [0] msg from board 1: PWM DIS CH0 
6148		984.308888		ERROR		cfw2can [0] board 1 OVERCURRENT AXIS 0
6151		984.308924		DEBUG		cfw2can [0] msg from board 1: PWM DIS CH1 
6152		984.308933		DEBUG		cfw2can [0] msg from board 1: PWM DIS CH0 
6156		986.947188		DEBUG		cfw2can [0] thread ran 598 times, req.dT:10[ms], av.dT:10.02[ms] av.loopT :0.03[ms]
6159		986.957222		DEBUG		     Device has no ICanBusErr interface
6160		986.957235		DEBUG		cfw2can [0] printing boards infos:
```

I was almost forgetting: when the fault happens in idle, the error is not reported on the logger.

cc @DanielePucci @julijenv @pattacini @randaz81 @maggia80 @marcoaccame @ale-git @valegagge

francesco-romano, 2016-05-10:
I've just decreased the PIDs to 480, and it happened again (I put the head in home..moving of 2 degress)

pattacini, 2016-05-10:
I've heard once in the corridor (:smile: even though there's nothing to smile about) that @randaz81 was on a new piece of software capable in theory of improving this very bad situation that's been hurting us since ages by then.

francesco-romano, 2016-05-10:
@pattacini I heard the same story :smile:

randaz81, 2016-05-10:
Reallly? I've not heard anything about it. I'll ask to @randaz81  :smile:

pattacini, 2016-05-10:
Quoting from #75:

>However, I think @randaz81 is preparing a new special FW release to patch the way the temperature control is done.

:smile:

traversaro, 2016-06-07:
It is interesting to note that on `iCubGenova05` (see https://github.com/robotology/icub-support/issues/253) there was not `cfw2can [0] board 1 OVERCURRENT AXIS 0` error, but just this messages:
~~~
618854      327260,726010       DEBUG       cfw2can [0] msg from board 1: PWM DIS CH1 
618856      327260,726066       DEBUG       cfw2can [0] msg from board 1: PWM DIS COUPLED:1 
618859      327260,726093       DEBUG       cfw2can [0] msg from board 1: PWM DIS CH0 
~~~

traversaro, 2019-10-03:
Since when this issue has opened, the iCub head on the iCubGenova02 has been migrated from using CAN to using the ETH system, the iCubGenova05 does not exist anymore and I am not sure if anyone is currently working on iCubDarmstadt01 . For this reason, I think it make sense to just close this issue, and reopen if we experience similar problems on a robot with a CAN neck, while the rest of the robot is ETH.

closed 2019-10-03 by traversaro