robotology/mesh-support #242: Left hand ring finger always closed

claudiofantacci, 2016-05-24:
I cannot tag the robot name because it has not been assigned yet.
The robot is the one present in the Blue iCub area at IIT. It's black with exposed skin (no covers).

##### Description of the failure
The left hand ring finger is always closed and cannot be opened.

##### Detailed conditions when the failure occurred
After starting yarpmotorinterface, the finger closed and never opened again. After noticing the issue, I tried to open the subject finger with the yarpmotorgui and restarting the yarpmotoringerface, without success.
The finger does not seem to be blocked/stuck, but rather forced by the motors to stay in that particular position; in this respect, I wonder whether or not a problem is also present in the little finger as well.

julijenv, 2016-07-05:
was it solved?

davidetome, 2016-07-21:
the finger moves nicely. closed

closed 2016-07-21 by davidetome