robotology/mesh-support #243: Right leg is always less stable than left leg

hu-yue, 2016-05-24:
##### Description of the situation
During walking experiments we have always had the experience that the robot is **less** stable on the **right** leg in comparison to the left leg.

##### Details and questions
Since the robot always showed unstable behaviour on the right leg, we had adopted the following actions:
- fine calibration
- check possible model and computation mismatch between left and right side (there is actually a small mismatch between the kinematic model of right and left leg of Heidelberg01)
- compare left and right joint trajectories
The robot is mostly unstable on the frontal plane (roll motions).

Since none of the above adopted actions seemed to have given a satisfactory explanation nor a solution to the "weakness" of the right leg of the robot in the last months, I wonder if it is possible that there is an actual mechanical weakness on the right leg, or if it is possible that the motors have a "preferred" rotation direction (assuming they are mounted in the same way left and right but they rotate in opposite direction)?

Thanks!

As a side note I notice the same thing also on iCubGenova01 and iCubGenova02 but since I did not run enough experiments on them I cannot conclude anything on them.

hu-yue, 2016-10-28:
I would like to come back to this issue once again.
Lately I'm trying to control the robot via `iOpenLoop` by computing the values using a PID controller which computes the errors using the `iMotorEncoders` data. 
It seems to work fine, position tracking errors look pretty good on the left leg. 
On the right leg, with the same gains (different sign) and the same reference trajectories, the behaviour is completely different. The joints basically don't move at all and therefore the errors just keep increasing.
I could understand if it was just one joint, but all the joints of the right leg are behaving like this, I just can't figure out why there is this big difference between the two legs.

traversaro, 2019-10-03:
This was probably related to an asymmetry in the URDF model generated for iCubHeidelberg01, see https://github.com/robotology/icub-model-generator/issues/16 . That issue can be tracked there, and if new problem can arise a new issue can be opened.

closed 2019-10-03 by traversaro