robotology/mesh-support #244: Right shoulder stiff while arm in Idle mode on iCubGenova05

nunoguedelha, 2016-05-27:
##### Description of the failure
On `iCubGenova05`, the right shoulder is stiff while all the arm joint motors are set to Idle. We can clearly notice the difference w.r.t. the left arm which is behaving normally. **We suspect that the faulty joint is the right joint 0 (shoulder pitch)**.

##### Detailed conditions when the failure occurred

robot name : `iCubGenova05`
`yarprobotinterface` and `yarpmotorgui` running.
No other application running.
All right arm joints set to Idle.

DanielePucci, 2016-05-31:
After debugging the problem with @iron76, we found out that the stuck motor is the `2B0M1` (see http://wiki.icub.org/wiki/ICub_coupled_joints#Shoulder for further information on motors labelling)

traversaro, 2016-07-22:
Two transmissions were changed in the robot from when this issue was started, and I don't remember to see this behaviour on the robot recently, so I think that this issue can be closed.

closed 2016-07-22 by traversaro