robotology/mesh-support #245: Error on virtual analog time out

hu-yue, 2016-05-27:
##### Description of the failure
An error popped out on virtual analog sensor and I have no idea what this means:
```
[ERROR]Virtual analog sensor timeout!! No new value received for more than  0.080786  secs. 
```
It has appeared several times while the robot was doing nothing but standing.
Also, this error seems to have some printing problems, since it appeared also like this:
```
[[ERROR]Virtual analog sensor timeout!! No new value received for more than  0.0809329  secs. 
```
and this:
```
[ERROR[ERROR]Virtual analog sensor timeout!! No new value received for more than  0.0806031  secs.
```
[Here](http://orb.iwr.uni-heidelberg.de/ftp/icub_logs/log_pc104_yarprobotinterface_3130_27_05_2016.txt). is the full log.

##### Detailed conditions when the failure occurred

The robot was just standing on the floor doing nothing, before this we performed a couple of walking motions on flat ground. 
`wholeBodyDynamicsTree` was running and a calibration of the F/T has been done for standing on floor.
I was reading some data from some ports of `wholeBodyDynamicsTree`.

traversaro, 2016-05-27:
This should happen when the `wholeBodyDynamicsTree` is calibrating, because while calibrating the `wholeBodyDynamicsTree` is not publishing torque estimates (this is an issue that will be solved soon). 

However, if this happens even when the `WBDT` is not calibrating, it may indicates some delays in the network. There was a particular load on the network when you got this errors?

hu-yue, 2016-05-27:
No, I was just reading, nothing was going on. 
Mh maybe it happened while I was calibrating. I did do the calibration a couple of times. But I can't tell if they popped out during the calibration or not.
I will check again on Monday.

hu-yue, 2016-05-30:
@traversaro I verified that they occured while calibrating the `WBDT`.

hu-yue, 2016-05-30:
And it happened when I closed the `WBDT`.

traversaro, 2017-04-20:
Can you try to update your `codyco-superbuild` and launch `yarprobotinterface --config launch-wholebodydynamics.xml` in place of `wholeBodyDynamicsTree`? 

Ref: https://github.com/robotology/codyco-modules/issues/223 .

traversaro, 2017-12-06:
@hu-yue feel free to open again if the issue is still persisting with the latest version of the software.

closed 2017-12-06 by traversaro